ricardoquesada/bluepad32 #3: Gamepad connect but not working.

tomudo, 2021-07-15:
When I run esp32 bluepad32 and connect gamepad, it connect but not work. My gamepad is this model  https://www.dhresource.com/0x0s/f2-albu-g4-M01-12-60-rBVaEVet7y-AClz3AAEwSGBzEK0004.jpg/kosda-bluetooth-wireless-game-controller.jpg

[0;32mI (276) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;32mI (376) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 0[0m
BTstack up and running on 08:3A:F2:AA:47:12.
Stored link keys:
8E:37:82:81:63:25 - type 4, key: .
--> Scanning for new devices...
--> HCI_EVENT_COMMAND_COMPLETE: opcode = 0x200b - status=0
Device found: 8E:37:82:81:63:25 with COD: 0x002540, pageScan 1, clock offset 0x14c0, rssi -33 dBm
STATE_DEVICE_DISCOVERED
--> GAP_EVENT_INQUIRY_COMPLETE
--> Scanning for new devices...
Device found: 8E:37:82:81:63:25 with COD: 0x002540, pageScan 1, clock offset 0x14c0, rssi -33 dBm
--> GAP_EVENT_INQUIRY_COMPLETE
--> Scanning for new devices...
--> HCI_EVENT_CONNECTION_COMPLETE
--> HCI_EVENT_LINK_KEY_REQUEST:
--> HCI_EVENT_COMMAND_COMPLETE: opcode = 0x040b - status=0
--> HCI_EVENT_AUTHENTICATION_COMPLETE_EVENT: status=0, handle=0x0081
--> HCI_EVENT_COMMAND_COMPLETE: opcode = 0x1408 - status=0
L2CAP_EVENT_CHANNEL_OPENED (channel=0x0041)
PSM: 0x0011, local CID=0x0041, remote CID=0x0040, handle=0x0081, incoming=0, local MTU=128, remote MTU=72
HID Control opened, cid 0x41
STATE_L2CAP_CONTROL_CONNECTED
L2CAP_EVENT_CHANNEL_OPENED (channel=0x0042)
PSM: 0x0013, local CID=0x0042, remote CID=0x0041, handle=0x0081, incoming=0, local MTU=128, remote MTU=72
HID Interrupt opened, cid 0x42
STATE_L2CAP_INTERRUPT_CONNECTED
Starting SDP query for HID descriptor for: 8E:37:82:81:63:25
Device without HID descriptor or Product/Vendor ID yet.
Waiting for SDP answer. Ignoring report.
STATE_SDP_HID_DESCRIPTOR_FETCHED
Starting SDP query for product/vendor ID
Vendor ID: 0x1949 - Product ID: 0x0402
Device detected as Android: 0x25
STATE_SDP_HID_VENDOR_FETCHED
unijoysticle: device 8E:37:82:81:63:25 has new gamepad seat: 2
Another SDP query in progress. Disconnect gamepad and try again.
8E:37:82:81:63:25, handle=129, ctrl_cid=0x0041, intr_cid=0x0042, cod=0x00002540, vid=0x1949, pid=0x0402, flags=0x00003902, ctrl_type=0x25, name='', (del:0)

ricardoquesada, 2021-07-15:
yep. 
I think have that gamepad as well, and it is not supported :( ... something is wrong with the gamepad... not sure what... 

In any case, for the future:
- mention which bluepad32 version you are using
- Add link to the gamepad product... (it could be amazon link, aliexpress, or if you have the name/model)
  in this case the photo is enough because I recognize it... but might not be enough for other gamepads.

tomudo, 2021-07-15:
I used latest master Jul 12, 2021 commit. Thank you for clarification.

lionfer51, 2023-09-19:
I tried it with the same gamepad and it works in iOS mode (Y+Home)

ricardoquesada, 2023-09-20:
yep. apparently there is a newer firmware that works Ok. 

I bought it from this link and works:
https://www.aliexpress.us/item/2251832602044731.html?spm=a2g0o.order_list.order_list_main.5.51801802pNXIqZ&gatewayAdapt=glo2usa

Closing bug as fixed.

closed 2023-09-20 by ricardoquesada