ricardoquesada/bluepad32 #16: issue with another library

apvlv, 2022-01-14:
Hello, 

thank you for your work on this library. I have a problem with Fastaccelstepper (https://github.com/gin66/FastAccelStepper) library in conjunction with this on an esp32 wroom32. I understand that it might be not your business to explain why it's not working, but maybe you can help me in the right direction. As soon I initialize the stepper library the esp reboots endlessly with this messages:

`E (451211) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (451211) task_wdt:  - IDLE (CPU 0)
E (451211) task_wdt: Tasks currently running:
E (451211) task_wdt: CPU 0: StepperTask
E (451211) task_wdt: CPU 1: IDLE
E (451211) task_wdt: Print CPU 0 (current core) backtrace

Backtrace:0x40102A12:0x3FFBE7500x40082EBD:0x3FFBE770 0x40085BED:0x3FFDC210 0x40093915:0x3FFDC230 0x400D8356:0x3FFDC250 0x400946A5:0x3FFDC270 
0x40102a12: task_wdt_isr at /Users/xxx/esp/esp-idf/components/esp_system/task_wdt.c:183 (discriminator 3)

0x40082ebd: _xt_lowint1 at /Users/xxx/esp/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1111

0x40085bed: esp_crosscore_int_send_yield at /Users/xxx/esp/esp-idf/components/esp_system/crosscore_int.c:144

0x40093915: vTaskDelay at /Users/xxx/esp/esp-idf/components/freertos/tasks.c:1596

0x400d8356: StepperTask(void*) at /Users/xxx/esp/pan-tilt-head/build/../components/fast_accel_stepper/src/FastAccelStepper.cpp:69 (discriminator 1)

0x400946a5: vPortTaskWrapper at /Users/xxx/esp/esp-idf/components/freertos/port/xtensa/port.c:131

E (451211) task_wdt: Print CPU 1 backtrace

Backtrace:0x40085B81:0x3FFBED500x40082EBD:0x3FFBED70 0x4000BFED:0x3FFD2800 0x40094956:0x3FFD2810 0x40102CD7:0x3FFD2830 0x40102CE3:0x3FFD2860 0x400D3A5D:0x3FFD2880 0x40092C11:0x3FFD28A0 0x400946A5:0x3FFD28C0 
0x40085b81: esp_crosscore_isr at /Users/xxx/esp/esp-idf/components/esp_system/crosscore_int.c:92

0x40082ebd: _xt_lowint1 at /Users/xxx/esp/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1111

0x40094956: vPortClearInterruptMaskFromISR at /Users/xxx/esp/esp-idf/components/freertos/port/xtensa/include/freertos/portmacro.h:571
 (inlined by) vPortExitCritical at /Users/xxx/esp/esp-idf/components/freertos/port/xtensa/port.c:319

0x40102cd7: esp_task_wdt_reset at /Users/xxx/esp/esp-idf/components/esp_system/task_wdt.c:330

0x40102ce3: idle_hook_cb at /Users/xxx/esp/esp-idf/components/esp_system/task_wdt.c:80

0x400d3a5d: esp_vApplicationIdleHook at /Users/xxx/esp/esp-idf/components/esp_system/freertos_hooks.c:51 (discriminator 1)

0x40092c11: prvIdleTask at /Users/xxx/esp/esp-idf/components/freertos/tasks.c:3974 (discriminator 1)

0x400946a5: vPortTaskWrapper at /Users/xxx/esp/esp-idf/components/freertos/port/xtensa/port.c:131`

I installed your library according to the instructions for esp-idf 4.2 and arduino core... Without the stepper library its working fine.

ricardoquesada, 2022-01-14:
Probably easier if you start by cloning this repo:
https://gitlab.com/ricardoquesada/esp-idf-arduino-bluepad32-template

in any case, to use Arduino core you **MUST** use esp-idf v4.4 or newer

apvlv, 2022-01-14:
Sorry, I made a mistake. I was using 4.2 earlier, but now I'm using 4.4 release. I installed it through esp-idf plugin for VSCode

apvlv, 2022-01-14:
And I'm using your template as well

apvlv, 2022-01-14:
The template itself working fine, I can connect my bluetooth gamepad (Steelseries Nimbus) and I see the values changing in the serial console. Though as I start to use the stepper library, the errors come up. Maybe I should ask there for help but I thought you could maybe point me in the right direction because there is some problem with the cpu cores apparently and I saw that your firmware makes use of the cores of the esp32

ricardoquesada, 2022-01-14:
yes, the problem is the stepper.
you traceback shows it: it says that the "watchdog" triggered the abort.

It seems that the stepper is running in CPU0, and it is not returning in time.

I'm not familiar with that library, but try to make it run in CPU1... CPU0 is being used by Bluetooth.

apvlv, 2022-01-14:
Thank you, i will try!

closed 2022-01-14 by apvlv