ricardoquesada/bluepad32 #18: 8Bitdo SFC30 swap button ports

danko79, 2022-02-26:
Hi, updated my unijoysticle 2 with last firmware, the 8bitdo sfc30 controller works, but the key for joy port swap is not mapped and i tried a different combinations of buttons pressed simultaneously but nothing to do, is possible to do a combo buttons like SELECT + START for the swap ports?

https://m.media-amazon.com/images/I/51u0PllAITL._AC_SL1000_.jpg

ricardoquesada, 2022-02-26:
Chances are that the the gamepad is not being recognized correctly.
Please, post the console log. 

If you don't know how to get it, here is the info:
https://gitlab.com/ricardoquesada/unijoysticle2/-/blob/main/docs/user_guide.md#unijoysticle-logs

danko79, 2022-02-26:
I don't know how to compile it, if you can do it for me i will test the new firmware right away!
Regards

danko79, 2022-02-26:
And add this 8bitDo M30 to database

--> HCI_EVENT_CONNECTION_REQUEST: link_type = 1 <--
on_hci_connection_request from: address = E4:17:D8:AD:82:7B, cod=0x2508
--> HCI_EVENT_ROLE_CHANGE
--> HCI_EVENT_CONNECTION_COMPLETE
--> HCI_EVENT_LINK_KEY_REQUEST:
--> HCI_EVENT_LINK_KEY_REQUEST:
--> L2CAP_EVENT_INCOMING_CONNECTION
L2CAP_EVENT_INCOMING_CONNECTION (psm=0x0011, local_cid=0x0041, remote_cid=0x0040, handle=0x0080, channel=0x0041, addr=E4:17:D8:AD:82:7B
--> HCI_EVENT_AUTHENTICATION_COMPLETE_EVENT: status=0, handle=0x0080
--> HCI_EVENT_LINK_KEY_REQUEST:
--> HCI_EVENT_AUTHENTICATION_COMPLETE_EVENT: status=0, handle=0x0080
L2CAP_EVENT_CHANNEL_OPENED (channel=0x0041)
PSM: 0x0011, local CID=0x0041, remote CID=0x0040, handle=0x0080, incoming=1, local MTU=1691, remote MTU=508, addr=E4:17:D8:AD:82:7B
HID Control opened, cid 0x41
uni_bluetooth_process_fsm, bd addr:E4:17:D8:AD:82:7B,  state: 10, incoming:1
--> L2CAP_EVENT_INCOMING_CONNECTION
L2CAP_EVENT_INCOMING_CONNECTION (psm=0x0013, local_cid=0x0042, remote_cid=0x0041, handle=0x0080, channel=0x0042, addr=E4:17:D8:AD:82:7B
L2CAP_EVENT_CHANNEL_OPENED (channel=0x0042)
PSM: 0x0013, local CID=0x0042, remote CID=0x0041, handle=0x0080, incoming=1, local MTU=1691, remote MTU=508, addr=E4:17:D8:AD:82:7B
HID Interrupt opened, cid 0x42
uni_bluetooth_process_fsm, bd addr:E4:17:D8:AD:82:7B,  state: 12, incoming:1
uni_bluetooth_process_fsm: requesting name
--> HCI_EVENT_REMOTE_NAME_REQUEST_COMPLETE
Name: '8BitDo M30 gamepad'
uni_bluetooth_process_fsm, bd addr:E4:17:D8:AD:82:7B,  state: 4, incoming:1
uni_bluetooth_process_fsm: starting SDP query
-----------> sdp_query_start()
Starting SDP VID/PID query for E4:17:D8:AD:82:7B
Vendor ID: 0x045e - Product ID: 0x02e0
Device detected as Xbox One: 0x20
uni_bluetooth_process_fsm, bd addr:E4:17:D8:AD:82:7B,  state: 6, incoming:1
uni_bluetooth_process_fsm: querying HID descriptor
Starting SDP HID-descriptor query for E4:17:D8:AD:82:7B
SDP HID Descriptor (306):
05 01 09 05 A1 01 85 01 09 01 A1 00 09 30 09 31 15 00 27 FF FF 00 00 95 02 75 10 81 02 C0 09 01 A1 00 09 33 09 34 15 00 27 FF FF 00 00 95 02 75 10 81 02 C0 05 01 09 32 15 00 26 FF 03 95 01 75 0A 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09 35 15 00 26 FF 03 95 01 75 0A 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09 39 15 01 25 08 35 00 46 3B 01 66 14 00 75 04 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03 05 09 19 01 29 0A 15 00 25 01 75 01 95 0A 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09 80 85 02 A1 00 09 85 15 00 25 01 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 C0 05 0F 09 21 85 03 A1 02 09 97 15 00 25 01 75 04 95 01 91 02 15 00 25 00 75 04 95 01 91 03 09 70 15 00 25 64 75 08 95 04 91 02 09 50 66 01 10 55 0E 15 00 26 FF 00 75 08 95 01 91 02 09 A7 15 00 26 FF 00 75 08 95 01 91 02 65 00 55 00 09 7C 15 00 26 FF 00 75 08 95 01 91 02 C0 85 04 05 06 09 20 15 00 26 FF 00 75 08 95 01 81 02 C0 
<----------- sdp_query_end()
uni_bluetooth_process_fsm, bd addr:E4:17:D8:AD:82:7B,  state: 8, incoming:1
Xbox one: Assuming it is firmware 3.1
Device setup (E4:17:D8:AD:82:7B) is complete
unijoysticle: device E4:17:D8:AD:82:7B has new gamepad seat: 2

ricardoquesada, 2022-02-27:
@danko79 Here is the new build that includes the change for the SFC30:
https://gitlab.com/ricardoquesada/bluepad32/-/jobs/2140760484/artifacts/file/tools/fw/artifacts/bluepad32-unijoysticle-390b79c.tar.gz

Regarding your second comment, the M30 is "impersonating" the Xbox Wireless controller... I cannot do anything about it.

ricardoquesada, 2022-02-27:
update: Perhaps the M30 has different connection methods... the one you posted is the "xbox mode"... perhaps it has a "swtich" mode, or "mac mode", or "android mode"... try with the other methods and let me know whether it works Ok. thansk!

danko79, 2022-02-27:
Hi Ricardo, the 8BitDo SFC30 controller now works! many thanks!
For the 8BitDo M30 works with the "xbox mode" did you make any changes for this? because in this way it worked even before.
Regards

ricardoquesada, 2022-02-27:
Good to know that it is working Ok. Closing the bug.
I didn't have to do anything for the M30 gamepad... some 8bitdo controllers work by default in xbox or nintendo switch mode... 
And for those controllers I don't have to do anything... besides supporting Xbox and Nintendo Switch controllers.

closed 2022-02-27 by ricardoquesada

danko79, 2022-07-29:
Hi Ricardo,re-open this issue with new fw 3.5 have the same problem as before, the swap button (SELECT) doesn't work anymore, can you take a look?
Regards Juri

ricardoquesada, 2022-07-29:
Yep. Thanks... I know what's happening.... "select" and "start" buttons are buttons are mapped correctly... but the for the Unijoysticle devices, I have to check that "menu" or "select" is pressed.

danko79, 2022-07-29:
Ok ,i'm wait your news.