ricardoquesada/bluepad32 #19: got a it_wdt_reset if call uni_gamepad_dump function

zeroxxx88, 2022-03-19:
hi there!

I got a it_wdt_reset if call uni_gamepad_dump function .It usually appears in a few minutes.I tryed to set Interrupt watchdog timeout bigger but is no efect.even if turn off Interrupt watchdog.

and is hard to conect to xboxone controller some times.

.............................................
(0xb0cfc7f) dpad=0x00, x=-6, y=-4, rx=-15, ry=10, brake=0, accel=0, buttons=0x00000000, misc=0x00
(0xb0cfc7f) dpad=0x00, x=-6, y=-4, rx=-15, ry=5, brake=0, accel=0, buttons=0x00000000, misc=0x00
(0xb0cfc7f) dpad=0x00, x=-6, y=-4, rx=-15, ry=2, brake=0, accel=0, buttons=0x00000000, misc=0x00
(0xb0cfc7f) dpad=0x00, x=-6, y=-4, rx=-15, ry=0, brake=0, accel=0, buttons=0x00000000, misc=0x00
ASSERT_PARAM(2097152 0), in rwbt.c at line 381
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0).

Core  0 register dump:
PC      : 0x400877e3  PS      : 0x00060034  A0      : 0x8008dc40  A1      : 0x3ffbea80
0x400877e3: r_assert_param at ??:?

A2      : 0x00000001  A3      : 0x00000000  A4      : 0x60008048  A5      : 0x00000000
A6      : 0x00000004  A7      : 0x3ffbdbb4  A8      : 0x800877e0  A9      : 0x3ffbea60
A10     : 0x00000000  A11     : 0x0000002f  A12     : 0x00000014  A13     : 0xffffffff
A14     : 0x00000000  A15     : 0xfffffffc  SAR     : 0x00000004  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x40087718  LEND    : 0x4008771f  LCOUNT  : 0x00000000
0x40087718: r_assert_param at ??:?

0x4008771f: r_assert_param at ??:?

Core  0 was running in ISR context:
EPC1    : 0x400d3aef  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x400877e3
0x400d3aef: uart_hal_write_txfifo at D:/Desktop/esp-idf/components/hal/uart_hal_iram.c:35

0x400877e3: r_assert_param at ??:?

Backtrace:0x400877e0:0x3ffbea80 0x4008dc3d:0x3ffbeaa0 0x4008e05b:0x3ffbeac0 0x400829ad:0x3ffbeae0 0x4011c4b3:0x3ffd1720 0x400d98b7:0x3ffd1740 0x40091235:0x3ffd1760 0x4009341d:0x3ffd1780
0x400877e0: r_assert_param at ??:?

0x4008dc3d: r_rwbt_isr at ??:?

0x4008e05b: r_rwbtdm_isr_wrapper at intc.c:?

0x400829ad: _xt_lowint1 at D:/Desktop/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1105

0x4011c4b3: cpu_ll_waiti at d:\desktop\esp-idf\bluepad32\src\build/../../../components/hal/esp32/include/hal/cpu_ll.h:183
 (inlined by) esp_pm_impl_waiti at D:/Desktop/esp-idf/components/esp_pm/pm_impl.c:827

0x400d98b7: esp_vApplicationIdleHook at D:/Desktop/esp-idf/components/esp_common/src/freertos_hooks.c:63

0x40091235: prvIdleTask at D:/Desktop/esp-idf/components/freertos/tasks.c:3839 (discriminator 1)

0x4009341d: vPortTaskWrapper at D:/Desktop/esp-idf/components/freertos/port/xtensa/port.c:168

Core  1 register dump:
PC      : 0x4011c4b6  PS      : 0x00060a34  A0      : 0x800d98ba  A1      : 0x3ffd2180
0x4011c4b6: esp_pm_impl_waiti at D:/Desktop/esp-idf/components/esp_pm/pm_impl.c:829

A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x80000001
A6      : 0x00000003  A7      : 0x00060023  A8      : 0x800f3ada  A9      : 0x3ffd2150
A10     : 0x00000000  A11     : 0x00060c23  A12     : 0x00060c20  A13     : 0x00060c23
A14     : 0x00000001  A15     : 0x00000000  SAR     : 0x0000001e  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace:0x4011c4b3:0x3ffd2180 0x400d98b7:0x3ffd21a0 0x40091235:0x3ffd21c0 0x4009341d:0x3ffd21e0
0x4011c4b3: cpu_ll_waiti at d:\desktop\esp-idf\bluepad32\src\build/../../../components/hal/esp32/include/hal/cpu_ll.h:183
 (inlined by) esp_pm_impl_waiti at D:/Desktop/esp-idf/components/esp_pm/pm_impl.c:827

0x400d98b7: esp_vApplicationIdleHook at D:/Desktop/esp-idf/components/esp_common/src/freertos_hooks.c:63

0x40091235: prvIdleTask at D:/Desktop/esp-idf/components/freertos/tasks.c:3839 (discriminator 1)

0x4009341d: vPortTaskWrapper at D:/Desktop/esp-idf/components/freertos/port/xtensa/port.c:168
[... 130 lines cut ...]
uni_bluetooth_process_fsm, bd addr:5C:BA:37:FF:6E:6B,  state: 4, incoming:1
uni_bluetooth_process_fsm: starting SDP query
-----------> sdp_query_start()
Starting SDP VID/PID query for 5C:BA:37:FF:6E:6B
Vendor ID: 0x045e - Product ID: 0x02fd
Device detected as Xbox One: 0x20
uni_bluetooth_process_fsm, bd addr:5C:BA:37:FF:6E:6B,  state: 6, incoming:1
uni_bluetooth_process_fsm: querying HID descriptor
Starting SDP HID-descriptor query for 5C:BA:37:FF:6E:6B
SDP HID Descriptor (334):
05 01 09 05 A1 01 85 01 09 01 A1 00 09 30 09 31 15 00 27 FF FF 00 00 95 02 75 10 81 02 C0 09 01 A1 00 09 32 09 35 15 00 27 FF FF 00 00 95 02 75 10 81 02 C0 05 02 09 C5 15 00 26 FF 03 95 01 75 0A 81 02 15 00 25 00 75 06 95 01 81 03 05 02 09 C4 15 00 26 FF 03 95 01 75 0A 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09 39 15 01 25 08 35 00 46 3B 01 66 14 00 75 04 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03 05 09 19 01 29 0F 15 00 25 01 75 01 95 0F 81 02 15 00 25 00 75 01 95 01 81 03 05 0C 0A 24 02 15 00 25 01 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 05 0C 09 01 85 02 A1 01 05 0C 0A 23 02 15 00 25 01 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 C0 05 0F 09 21 85 03 A1 02 09 97 15 00 25 01 75 04 95 01 91 02 15 00 25 00 75 04 95 01 91 03 09 70 15 00 25 64 75 08 95 04 91 02 09 50 66 01 10 55 0E 15 00 26 FF 00 75 08 95 01 91 02 09 A7 15 00 26 FF 00 75 08 95 01 91 02 65 00 55 00 09 7C 15 00 26 FF 00 75 08 95 01 91 02 C0 05 06 09 20 85 04 15 00 26 FF 00 75 08 95 01 81 02 C0
Xbox One: Unsupported page: 0x0002, usage: 0x00c5, value=0x0
Xbox One: Unsupported page: 0x0002, usage: 0x00c4, value=0x0
Xbox One: Unsupported page: 0x0009, usage: 0x000b, value=0x0
Xbox One: Unsupported page: 0x0009, usage: 0x000c, value=0x0
Xbox One: Unsupported page: 0x0009, usage: 0x000d, value=0x0
Xbox One: Unsupported page: 0x0009, usage: 0x000e, value=0x0
Xbox one: Firmware 4.8 detected
<----------- sdp_query_end()
uni_bluetooth_process_fsm, bd addr:5C:BA:37:FF:6E:6B,  state: 8, incoming:1
Xbox one: Assuming it is firmware 4.8
Device setup (5C:BA:37:FF:6E:6B) is complete
pc_debug: device ready: 0x3ffc42ac
(0xb0cfc7f) dpad=0x00, x=-5, y=-5, rx=-14, ry=0, brake=0, accel=0, buttons=0x00000000, misc=0x00
(0xb0cfc7f) dpad=0x00, x=-5, y=-5, rx=-14, ry=0, brake=0, accel=0, buttons=0x00000000, misc=0x00

ricardoquesada, 2022-03-19:
The more detailed the context, the better.

Which firmware are  you using? one modified by you?

This indicates that it is running in a ESP32:
> I (27) boot: ESP-IDF v4.3.1 2nd stage bootloader

This indicates that it is running in a PC:
> Platform: PC Debug

So, that's weird...

Also, from where are you calling `uni_gamepad_dump()` ?

zeroxxx88, 2022-03-20:
I set the platform to PC_ Debug,and change some GPIO pin maping, to test xboxone controller.

using the original uni_platform_pc_debug. c file.So far,it work fine if I comment out uni_gamepad_dump funtion.

ricardoquesada, 2022-03-20:
I wouldn't know... testing the `pc_debug` platform in an ESP32 is not tested...
Look at the ISR code... seems that you might be calling `uni_dump_gamepad` from there... and that most probably trigger the watchdog...

closing bug since this is not a Bluepad32 bug.

For future reference, post the entire code... will help me understand better the context

closed 2022-03-20 by ricardoquesada

zeroxxx88, 2022-03-21:
For example, print log in  function "pcdebug_on_gamepad_data"

 I found that the problem was caused by print log.If log length exceeds 92 characters, it will certainly appear.

This is my first time using esp32,I am learning,and can't find the reason.If my usage is wrong, please correct it, thank you!
.............................................
............................................
static void pc_debug_on_gamepad_data(uni_hid_device_t* d, uni_gamepad_t* gp) {
    UNUSED(d);
    static uni_gamepad_t prev = {0};
    if (memcmp(&prev, gp, sizeof(*gp)) == 0) {
        return;
    }
    prev = *gp;	

		logi("(0x%02x),dpad=0x%02x, x=%d, y=%d, rx=%d, ry=%d, brake=%d, accel=%d, buttons=0x%08x, misc=0x%02x\n",
         gp->updated_states,                                          // state
          gp->dpad, gp->axis_x, gp->axis_y, gp->axis_rx, gp->axis_ry,  // axis
         gp->brake, gp->throttle, gp->buttons, gp->misc_buttons       // misc
    );
		
		
.............................................
.............................................

ricardoquesada, 2022-03-21:
How familiar are you with microcontrollers?

From your first comment where you pasted the error:

> Core 0 was running in ISR context:
> EPC1 : 0x400d3aef EPC2 : 0x00000000 EPC3 : 0x00000000 EPC4 : 0x400877e3
> 0x400d3aef: uart_hal_write_txfifo at D:/Desktop/esp-idf/components/hal/uart_hal_iram.c:35

It seems that the ESP32 is sending (printing) to the console inside a ISR (Interrupt Service Routine).
What ever you put in a ISR must be super fast.

Printing a few lines to the console might be Ok... but dumping the gamepad state for each packet received is not...

Long story short: don't dump the gamepad data  to the console

zeroxxx88, 2022-03-21:
I've only used 51 and arm before,with no os.

Now I understand a little,The freeRTOS.Thanks

By the way, Xbox controllers are sometimes very difficult to connect, whether you restart esp32 or the controller or press the sync butn.But sometimes it's very easy to connect. I tried to find out the reason,but it's too difficuilt to me now.