ricardoquesada/bluepad32 #21: [Feature Request] Add enableNewBluetoothConnections to Arduino + Nina Lib

ste2425, 2022-05-31:
Hi

Firstly fantastic library, im using it in my PSP Bluetooth controller project.

However noticed that in one of the components in the Bluepad32 project there is the ability to disable new connections with the method `enableNewBluetoothConnections`. Here:
  https://github.com/ricardoquesada/bluepad32/blob/ee6213b7ca0af17574e3952e8c6f062868df7fab/src/components/bluepad32_arduino/include/ArduinoBluepad32.h#L55

However as I'm using an Arduino nano 33 IOT i used this library based on your readme https://gitlab.com/ricardoquesada/bluepad32-arduino

Which doesn't seem to have the same functionality option.

Either i'm missing something (most likely) or i would like to request the feature added.

Thanks again!

ricardoquesada, 2022-06-03:
fixed.

Requires "bluepad32-arduino" v1.1.2 (or latest git version as of today)

Requires "bluepad32" v3.5 (or latest git version from `develop` branch as of today)

I'll release stable binaries once I finish v3.5.

LMK whether it works for you. ty!

closed 2022-06-03 by ricardoquesada

ste2425, 2022-07-01:
Hi @ricardoquesada  Sorry to re-open an old issue but only just got round to playing with this (family life got in the way!)

I seem to be experiencing some potentually odd behaviour with this feature and would like to confirm its intended behaviour.

I have two controllers. An Xbox one S, and a PS4. The Xbox is already paired, the PS4 is not.

If i set `enableNewBluetoothConnections(false)` do we expect the Xbox controller to still connect but the PS4 not too?

If so then its not behaving as intended. 

`enableNewBluetoothConnections(false)` seems to stop all controllers connecting regardless of if they are already paired or not.

If this is intended behaviour is there a way to block pairing of new controllers but allow already paired controllers to connect?

This in on an Arduino nano 33 IOT.

Thanks again for providing such a brilliant library BTW :)

ricardoquesada, 2022-07-01:
yeah... two things:

a) probably the name of the API is confusing. The intended behavior is that no connections are allowed.  But if there is a connected game, it will continue to be connected. But if that gamepad disconnects, it won't be able to connect again.

b) My intention was to do what a computer/phone does.  If it is in "discover / pair" mode, it will allow new connections. 
But if it is not, then you cannot connect to... even if it has been paired before (???)

If that's not the case..... mmm....  thinking aloud, that's not the case... I can connect my headphones to my phone without opening the "pair/discover" dialog.

I'll fix that....

ste2425, 2022-07-01:
In that case it is behaving as was intended, however as you say that may not be most desired.

The use case i have is I'm building a consolized PSP with this library being at its core. The Arduino is always on awaiting Bluetooth connections so it can turn on the PSP by connecting a gamepad, just like you can with an Xbox etc.

With discovery/pair being the same action this would mean if i was trying to sync a controller to a different console the Arduino could potentially steal that controller. When in reality all i want is for it to listen for an already paired controller.

I think were on the same page with this but just wanted to clarify :)

ricardoquesada, 2022-07-02:
fixed here: https://gitlab.com/ricardoquesada/bluepad32/-/commit/30bc9201d983940ebec813f0c0abff49ed65722e

I'll release a new version (probably) this weekend.