ricardoquesada/bluepad32 #22: Xbox system button no longer works

ste2425, 2022-07-05:
Firstly sorry to report another issue, seems like I'm spamming you recently with changes!

However the following code snippet reports false for a XBOX one s controller even when the Xbox button is pressed. I have a PS4 controller which works fine when the PS button is pressed.

`myGamepad->miscSystem();`

This is on a Arduino nano 33 IOT.

Arduino library:  v1.1.2
Nina firmware: Bluepad32 for NINA v3.5-beta1

ricardoquesada, 2022-07-05:
Thanks for filing bugs! (please, keep filing them!)

(I'll take a look at it later today)

ricardoquesada, 2022-07-05:
I've just tested with Bluepad32-v3.5.0-rc0 + Arduino library v1.1.3  in a RP2040 Connect (which for the purpose of this bug, they are the same), and I can see the "misc" button when I press the "xbox" system button (button 0x01)

what do you see if you print the output of `myGamepad->miscButtons()`, and also the output of `myGamepad->buttons()` ?

closed 2022-07-06 by ste2425