ricardoquesada/bluepad32 #23: Missing Defines

LayerDE, 2022-07-16:
I downloaded the library from the Arduino library manager and the example throws error when i want to verify the sketch

C:\Users\the_g\Documents\Arduino\libraries\Bluepad32\src\utility\spi_drv.cpp: In static member function 'static void SpiDrv::begin()':
C:\Users\the_g\Documents\Arduino\libraries\Bluepad32\src\utility\spi_drv.cpp:91:20: error: 'PINS_COUNT' was not declared in this scope
   if (SLAVERESET > PINS_COUNT) {
                    ^~~~~~~~~~
C:\Users\the_g\Documents\Arduino\libraries\Bluepad32\src\utility\spi_drv.cpp:91:20: note: suggested alternative: 'ICOUNT'
   if (SLAVERESET > PINS_COUNT) {
                    ^~~~~~~~~~
                    ICOUNT
C:\Users\the_g\Documents\Arduino\libraries\Bluepad32\src\utility\spi_drv.cpp:100:11: error: 'NINA_GPIO0' was not declared in this scope
   pinMode(NINA_GPIO0, OUTPUT);
           ^~~~~~~~~~
C:\Users\the_g\Documents\Arduino\libraries\Bluepad32\src\utility\spi_drv.cpp:100:11: note: suggested alternative: 'NINA_GPIOIRQ'
   pinMode(NINA_GPIO0, OUTPUT);
           ^~~~~~~~~~
           NINA_GPIOIRQ
C:\Users\the_g\Documents\Arduino\libraries\Bluepad32\src\utility\spi_drv.cpp: In static member function 'static int SpiDrv::available()':
C:\Users\the_g\Documents\Arduino\libraries\Bluepad32\src\utility\spi_drv.cpp:66:22: error: 'NINA_GPIO0' was not declared in this scope
 #define NINA_GPIOIRQ NINA_GPIO0
                      ^~~~~~~~~~
C:\Users\the_g\Documents\Arduino\libraries\Bluepad32\src\utility\spi_drv.cpp:539:47: note: in expansion of macro 'NINA_GPIOIRQ'
 int SpiDrv::available() { return (digitalRead(NINA_GPIOIRQ) != LOW); }
                                               ^~~~~~~~~~~~
C:\Users\the_g\Documents\Arduino\libraries\Bluepad32\src\utility\spi_drv.cpp:66:22: note: suggested alternative: 'NINA_GPIOIRQ'
 #define NINA_GPIOIRQ NINA_GPIO0
                      ^~~~~~~~~~
C:\Users\the_g\Documents\Arduino\libraries\Bluepad32\src\utility\spi_drv.cpp:539:47: note: in expansion of macro 'NINA_GPIOIRQ'
 int SpiDrv::available() { return (digitalRead(NINA_GPIOIRQ) != LOW); }
                                               ^~~~~~~~~~~~
exit status 1
Fehler beim Kompilieren für das Board WEMOS LOLIN32 Lite.

ricardoquesada, 2022-07-16:
Which board are you using?

LayerDE, 2022-07-16:
Wemos lilon32 lite

LayerDE, 2022-07-16:
same error on the ESP32 Dev board
```
Arduino: 1.8.19 (Windows Store 1.8.57.0) (Windows 10), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, Core 1, Core 1, None"

C:\Users\the_g\Documents\Arduino\libraries\Bluepad32\src\utility\spi_drv.cpp: In static member function 'static void SpiDrv::begin()':

C:\Users\the_g\Documents\Arduino\libraries\Bluepad32\src\utility\spi_drv.cpp:91:20: error: 'PINS_COUNT' was not declared in this scope

   if (SLAVERESET > PINS_COUNT) {

                    ^~~~~~~~~~

C:\Users\the_g\Documents\Arduino\libraries\Bluepad32\src\utility\spi_drv.cpp:91:20: note: suggested alternative: 'ICOUNT'

   if (SLAVERESET > PINS_COUNT) {

                    ^~~~~~~~~~

                    ICOUNT

C:\Users\the_g\Documents\Arduino\libraries\Bluepad32\src\utility\spi_drv.cpp:100:11: error: 'NINA_GPIO0' was not declared in this scope

   pinMode(NINA_GPIO0, OUTPUT);

           ^~~~~~~~~~

C:\Users\the_g\Documents\Arduino\libraries\Bluepad32\src\utility\spi_drv.cpp:100:11: note: suggested alternative: 'NINA_GPIOIRQ'

   pinMode(NINA_GPIO0, OUTPUT);

           ^~~~~~~~~~

           NINA_GPIOIRQ

C:\Users\the_g\Documents\Arduino\libraries\Bluepad32\src\utility\spi_drv.cpp: In static member function 'static int SpiDrv::available()':

C:\Users\the_g\Documents\Arduino\libraries\Bluepad32\src\utility\spi_drv.cpp:66:22: error: 'NINA_GPIO0' was not declared in this scope

 #define NINA_GPIOIRQ NINA_GPIO0

                      ^~~~~~~~~~

C:\Users\the_g\Documents\Arduino\libraries\Bluepad32\src\utility\spi_drv.cpp:539:47: note: in expansion of macro 'NINA_GPIOIRQ'

 int SpiDrv::available() { return (digitalRead(NINA_GPIOIRQ) != LOW); }

                                               ^~~~~~~~~~~~

C:\Users\the_g\Documents\Arduino\libraries\Bluepad32\src\utility\spi_drv.cpp:66:22: note: suggested alternative: 'NINA_GPIOIRQ'

 #define NINA_GPIOIRQ NINA_GPIO0

                      ^~~~~~~~~~

C:\Users\the_g\Documents\Arduino\libraries\Bluepad32\src\utility\spi_drv.cpp:539:47: note: in expansion of macro 'NINA_GPIOIRQ'

 int SpiDrv::available() { return (digitalRead(NINA_GPIOIRQ) != LOW); }

                                               ^~~~~~~~~~~~

exit status 1

Fehler beim Kompilieren für das Board ESP32 Dev Module.

Dieser Bericht wäre detaillierter, wenn die Option
"Ausführliche Ausgabe während der Kompilierung"
in Datei -> Voreinstellungen aktiviert wäre.
```

LayerDE, 2022-07-16:
do i need to create a config somewhere?

ricardoquesada, 2022-07-16:
Please see the supported boards: https://gitlab.com/ricardoquesada/bluepad32-arduino

If you want to use a ESP32 board with Arduino, you have to use this project instead:
https://gitlab.com/ricardoquesada/esp-idf-arduino-bluepad32-template

LayerDE, 2022-07-16:
Is there no solution for using the project in the arduino ide for easy access to the arduino librarys?

ricardoquesada, 2022-07-16:
not yet

closed 2022-07-16 by LayerDE