ricardoquesada/bluepad32 #24: Xbox One X Controller FW version 5.13.3143.0

danko79, 2022-07-30:
Hi, i have a Xbox One X controller 1708 model with the last update firmware 5.13.3143.0, is there any chance updating the bluepad32 firmware?

ricardoquesada, 2022-07-30:
I wasn't aware that a new FW was released for the controller... I'll take a look at it.

ricardoquesada, 2022-09-13:
It seems that FW 5.13.3143.0 migrated from Bluetooth Classic to BLE only.

So, in order to support it I have to add BLE support.... which was already in the todo list

ricardoquesada, 2022-09-13:
More info about 5.13.3143.0 , and how to revert to 4.8 (as a temporary solution):

https://www.reddit.com/r/xboxinsiders/comments/ta7kib/xbox_one_controller_firmware_5133143_available/

danko79, 2022-09-16:
Works with downgrade the firmware.

closed 2022-10-02 by danko79