ricardoquesada/bluepad32 #25: [Feature Request] Battery level/low battery status

ste2425, 2022-09-11:
Iv'e used this project to build a consolized PSP and its fantastic. 

Whilst testing it out playing Spyro half way through a long glide the batteries in my controller died without indication. The controller itself had no low battery indicator. Obviously Spyro died and i was sad.

Would it be possible to query the controller to get a battery level indication, either an overal percentage level or if not a low battery status?

This would allow me to display a low battery warning on my consolizer and prevent Spyro from future suffering.

Thanks!

ste2425, 2022-09-11:
Forgot to add im using the Arduino nano 33 IOT platform. Just incase this functionality exists on another platform.

ricardoquesada, 2022-09-11:
yes, sure. Some controllers are reporting the battery status, so it shouldn't be that difficult to add it.

In particular, what controller are you using? Just to double check that it reports the battery status

ste2425, 2022-09-11:
brilliant thanks.

I primarily use a PS4 pad but also been testing an Xbox One S and Wii controller.

Understandable not all controllers may report a battery status.

ricardoquesada, 2022-12-23:
This feature is going to be included as part of `get_controller_data` (instead of `get_gamepad_data`)... and `controller` will include a union of gampead, mouse, balance_board... and the battery will be part of that.

ste2425, 2023-01-11:
@ricardoquesada  Sorry forgot to reply to this after the XMas holidays.

Yup that plan sounds good. Noticed your recent commits on this work, if i was to pull the BP32 firmware from a latest main build it would have this functionality in right?

Is the Arduino side of the fence implemented to query that data from the Nina chip?

If so i will pull down and have a play, thanks.

ricardoquesada, 2023-01-11:
the NINA part is missing... Planning to work on that this week / weekend.

I split the data in a controller union
  - gamepad data
  - mouse data
  - balance data
  - (...open for other type of controllers in the future)

And the controller properties includes the battery.
The old API will still work... you should be able to do `get_gamepad_data`.
But the new one is more flexible...

Pending: Add the new Controller API on NINA and also on Arduino (non-NINA)... test it... and release it.

ricardoquesada, 2023-01-16:
Fixed.

Please test it with:
*  bluepad32-arduino v1.2.0
* bluepad32 FW v3.6.0-rc0

There is a new method called battery():
0 = Unk
1 = Empty
255 = full

There is a new API to handle Controllers... which is `Controller.h` ... 
And `Gamepad` is just an alias to `Controller`... so you don't have to change anything... but you might want to do some renames in the future to keep it cleaner.

closed 2023-01-16 by ricardoquesada