ricardoquesada/bluepad32 #26: [Feature Request] Adjust LED brightness & programmatic controller disconnect.

ste2425, 2022-09-12:
Sorry to be spamming you with feature requests all of a sudden! However i have two more requests if they are at all possible.

I can split these out into two separate requests if you prefer.

1:) LED Brightness 
My consolized PSP project allows for various controller button mappings to better suite the game being played. For controllers that have LED's (I'm testing on a PS4 pad) i use the LED to indicate the active mapping. Other controllers i use the player number (Wii etc)

Is it possible to control the brightness of the LED? I assume if it is it will depend on individual controller support. Iv'e read that the PS4 can have its LED brightness altered via the PS4 console itself but not sure if this is Sony bespoke stuff or can be done in a generic way.

2) Programmatic controller disconnect
My Consolized PSP has the ability to turn the PSP off via a button, not just disconnecting a controller. It would be great in the situation where the user clicks the button to shut down the console i can also instruct the controller to turn off.

I cannot disable the Bluetooth and let the controller shut itself down after a timeout because i need to listen to new connections to turn the console back on.

I'm using the Arduino nano 33 IOT.

Thanks!

I understand its its allot of work maintaining this project so if you can't implement any of my requests i understand :)

ricardoquesada, 2022-09-12:
for a)

Does `gamepad->SetColorLED(r,g,b);` work ?

https://gitlab.com/ricardoquesada/bluepad32-arduino/-/blob/main/examples/Gamepad/Gamepad.ino#L119

e.g: Let's say you have color red (255, 0, 0)... to change the brightness change it to (200,0,0)... etc.

for b)
sure. 

(for future reference: yes, please file one bug per feature request.... file as many as you want... don't worry about spamming)

ste2425, 2022-09-12:
A:) Of cause! it seems so obvious now you've said it, can't believe i didn't clock on! Just scaled each channel down by the same amount.

B) thats fantastic thanks.

I'll edit this story to scope to just feature be.

Thanks.

ricardoquesada, 2022-10-17:
Fixed.

- Bluepad32: contains the fix in "develop" branch.
- Bluepa32-arduino: I've just released v1.1.4 which contains the fix.

Please LMK if that works Ok.

closed 2022-10-17 by ricardoquesada

ste2425, 2022-10-24:
@ricardoquesada Given up trying to get a local environment setup to compile Bluepad. No idea what I'm doing wrong, but after a couple of days im giving up.

I'll wait until there is a RC for Bluepad and report back. 

Thanks.

ricardoquesada, 2022-10-26:
@ste2425 in case you want to test it before the release, you should know that we have automatic builds.

for example, the latest one is here:
https://gitlab.com/ricardoquesada/bluepad32/-/jobs/3180787721

Just press "download" and the `artifacs.zip` will contain the different images

ste2425, 2022-10-26:
Ohh perfect. I was looking in GitHub actions for this. I forget this is a clone/backup repo and the source is the GitLab one. 

Great thanks,

ste2425, 2022-10-26:
@ricardoquesada 

Flashed the latest and the disconnect method runs. However I'm experiencing potentially odd behaviour.

Using the example code from your repo the on disconnected handler fires, however the PS4 controller remains turned on until i manually turn it off.

I was expecting that disconnecting a controller would turn it off, however this may be my mis-understanding.

ricardoquesada, 2022-10-26:
mmm... good point.

I wonder what happens if you connect the DS4 to Linux (or Android or Windows...) and disconnect it from the host.
Does it turn-off ?

If it turns off, then I might need to send a "disconnect" message  (???)... what I'm doing right now is to close the connection.... and after a few seconds DS4 should "turn off"

ste2425, 2022-10-26:
OKi doki.

Setup:

* Windows 10, DS4
* Connect DS4 to windows, connected but no LED illuminated
* Use https://gamepad-tester.com/to test controller
* Use Steam to make LED illuminate (figured it would be best way to tell if controller turned off or not)
* In Windows settings -> Bluetooth & other devises
* Right click controller and select `Remove`
* Small delay and controller LED turns of
* Pressing home button flashes LED and re-connects

Based on the small delay between clicking the disconnect button and the controller being removed from the list in Windows and the LED turning off i assume there is some comms going on instructing the controller to turn off. 

Back with the Arduino, after executing the disconnect command the controller remains on (lasted 30 min before i got bored) until i power off the Arduino. I wonder if there is still some sort of communication going on keeping the controller on.

ricardoquesada, 2022-10-26:
great. thanks for testing. once I get home I'll fix it.

ricardoquesada, 2022-10-27:
fixed. tested it both in NINA and from the console, and now `disconnect` "shuts down" the gamepad. Tested with DS4.

The FW images should appear here (once the CI finishes):
https://gitlab.com/ricardoquesada/bluepad32/-/jobs/3234232293

ricardoquesada, 2022-10-27:
notice that the binaries generated by CI don't have the "DualShock reconnect" fix... but once v3.5.2 is released, it will contain it (I should patch CI, and tell it to include the patch)

ste2425, 2022-10-27:
Perfect thanks.

DS4 works great, Wii works great.

However Xbox One S behaves slightly differently. When disconnecting the LED flashes for about 60 seconds. I wonder if its trying to re-connect.

Its not an issue, just FYI in case you wanted all controller to behave the same.-

I don't+ have any other controllers to see how they behave.

Thanks for getting a fix out so quick!

ricardoquesada, 2022-10-27:
good to know. thanks!