ricardoquesada/bluepad32 #31: 8bitdo M30 and Amiga button maps

danko79, 2022-12-08:
Hi, on the Amiga 500/600/1200 with 8bitdo m30 controller the map button is:
A: FIRE
B: maps as JUMP (arrow UP)
C: FIRE (same button A)
Request: it's possibile maps button C to FIRE II https://old-computers.com/MUSEUM/connectors/sms_joystick.gif
In some games the second button is required.

ricardoquesada, 2022-12-08:
in which button is mapped Fire II in the 8bitdo m30?

Because fire II and fire III should be mapped...

in any case, I could add an option to disable the "up-is-fire" feature, and map Fire II there

danko79, 2022-12-08:
Fire II is not quite mapped in either case, my request is to map it. I prefer this config

M30
Button C (FIRE II)

SFC30
Button Y (FIRE II)

That would be a complete firmware with these mappings, an example game that requires this configuration is https://mcgeezer.itch.io/kung-fu-remaster

ricardoquesada, 2022-12-08:
I guess I should get an 8bitdo m30... the 3 Amiga buttons are already mapped... but I guess there is a bug with the M30

ricardoquesada, 2022-12-09:
I've just ordered an 8bitdo M30... will arrive in a few days...

ricardoquesada, 2022-12-09:
and to gain some time, could you tell me the Vendor ID (VID) and Product ID (PID) of those gamepads.

You can do it by connecting them, and them from the console type `list_devices` and copy & paste the output here. Thanks!

danko79, 2022-12-09:
Hi, 
8bitdo M30 HID\VID_2DC8&PID_5006

8bitdo SFC30 HID\VID_2DC8&PID_AB21

Grab with UsbTreeView, do you need a full dump?

ricardoquesada, 2022-12-09:
ahh... which firmware are you using?  "unijoysticle" or "mightymiggy" ?

danko79, 2022-12-09:
> ahh... which firmware are you using?  "unijoysticle" or "mightymiggy" ?

Only unijoysticle.

ricardoquesada, 2022-12-09:
Mmm... probably that's the issue... "unijoysticle" is not expected to have the correct mappings for the SukkoPera's board.
Instead, you should use the "mightymiggy" version, which was developed by SukkoPera for his board.

danko79, 2022-12-09:
> Mmm... probably that's the issue... "unijoysticle" is not expected to have the correct mappings for the SukkoPera's board.
> Instead, you should use the "mightymiggy" version, which was developed by SukkoPera for his board.

Unfortunately i connected unijoysticle c64 version to amiga same problem. I can test the sukkopera firmware mightymiggy, I'll let you know.

ricardoquesada, 2022-12-09:
Yep. The original Unijoysticle2 (through-hole version) doesn't support 2nd/3rd buttons on the Amiga.
But the Unijoysticle 2+ / A500, they support it.

danko79, 2022-12-09:
Tested mightymiggy 3.5.2 on Unijoysticle 2 - Amiga Version the controller is mapped.
 
M30 
FIRE I BUTTON A 
FIRE 2 BUTTON X

SFC30 
FIRE 1 Y
FIRE 2 B

Unfortunately the Kensington Mouse (K75351WW)  doesn't work with this FW. 
if you could remap those 2 controllers it would be a perfect firmware for me.

ricardoquesada, 2022-12-09:
it is not as simple remapping... the original Unijoysticle2 (trhough-hole) doesn't have the lines to control buttons 2/3 on joystick 2

ricardoquesada, 2022-12-09:
here is the comparison table: https://gitlab.com/ricardoquesada/unijoysticle2

danko79, 2022-12-09:
Ok, so we can consider this issue as closed, thanks for your support.

ricardoquesada, 2022-12-09:
np.

closed 2022-12-09 by ricardoquesada