ricardoquesada/bluepad32 #33: How  to know which button is being   pressed？

bihailantian655, 2023-04-17:
I am using a gamepad designed by an unknown manufacturer.

 I'm a bit confused about how to set it up in the program to know which button is being 

pressed.

 I'm currently using IDF 4.4. After successfully connecting,

 I can only see information in the serial port when I press the back, start, or menu keys. 

There is no information when I press any other keys. The contents of the serial port output 

are as follows.   

[image]

bihailantian655, 2023-04-17:
[image]

ricardoquesada, 2023-04-17:
which board are you using ? is it a Unijoysticle? 
Please type: `version` on the console and copy & paste the output.

add as much info as possible

bihailantian655, 2023-04-17:
not  any board  just   esp32  wroom

bihailantian655, 2023-04-17:
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6664
load:0x40078000,len:14944
load:0x40080400,len:3804
entry 0x40080698
I (27) boot: ESP-IDF v4.4.4 2nd stage bootloader
I (27) boot: compile time 14:32:53
I (27) boot: chip revision: v1.0
I (30) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (37) boot.esp32: SPI Speed      : 40MHz
I (41) boot.esp32: SPI Mode       : DIO
I (46) boot.esp32: SPI Flash Size : 4MB
I (51) boot: Enabling RNG early entropy source...
I (56) boot: Partition Table:
I (59) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (74) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (82) boot:  2 factory          factory app      00 00 00010000 00100000
I (89) boot:  3 coredump         Unknown data     01 03 00110000 00010000
I (97) boot: End of partition table
I (101) boot_comm: chip revision: 1, min. application chip revision: 0
I (108) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1dab8h (121528) map
I (161) esp_image: segment 1: paddr=0002dae0 vaddr=3ffbdb60 size=02538h (  9528) load
I (165) esp_image: segment 2: paddr=00030020 vaddr=400d0020 size=6dd70h (449904) map
I (329) esp_image: segment 3: paddr=0009dd98 vaddr=3ffc0098 size=031c0h ( 12736) load
I (334) esp_image: segment 4: paddr=000a0f60 vaddr=40080000 size=17eb8h ( 97976) load
I (387) boot: Loaded app from partition at offset 0x10000
I (387) boot: Disabling RNG early entropy source...
I (399) cpu_start: Pro cpu up.
I (399) cpu_start: Starting app cpu, entry point is 0x400812f4
I (0) cpu_start: App cpu up.
I (415) cpu_start: Pro cpu start user code
I (415) cpu_start: cpu freq: 160000000
I (415) cpu_start: Application information:
I (420) cpu_start: Project name:     bluepad32-app
I (425) cpu_start: App version:      release_v3.7.0-rc.0
I (431) cpu_start: Compile time:     Apr 17 2023 14:31:53
I (437) cpu_start: ELF file SHA256:  429f1f7b7be7c0d8...
I (443) cpu_start: ESP-IDF:          v4.4.4-dirty
I (449) heap_init: Initializing. RAM available for dynamic allocation:
I (456) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (462) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (468) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (474) heap_init: At 3FFD2558 len 0000DAA8 (54 KiB): DRAM
I (481) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (487) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (493) heap_init: At 40097EB8 len 00008148 (32 KiB): IRAM
I (501) esp_core_dump_flash: Init core dump to flash
I (506) esp_core_dump_flash: Found partition 'coredump' @ 110000 65536 bytes
E (513) esp_core_dump_flash: Incorrect size of core dump image: 270737568
luepad32 (C) 2016-2023 Ricardo Quesada and contributors.
Version: v3.7.0-rc0
BTStack: Copyright (C) 2017 BlueKitchen GmbH.
Platform: unijoysticle2
Unijoysticle: Board ID values: 1,1,1
Hardware detected: Unijoysticle2
I (595) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (605) gpio: GPIO[12]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (615) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (625) gpio: GPIO[14]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (635) gpio: GPIO[16]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (645) gpio: GPIO[17]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (655) gpio: GPIO[18]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (665) gpio: GPIO[19]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (675) gpio: GPIO[23]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (685) gpio: GPIO[25]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (695) gpio: GPIO[26]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
[... 107 lines cut ...]
mice timers enabled/disabled: port A=0, port B=0
Connected devices:
idx=0:
        btaddr: 00:90:E1:F0:DB:08
        bt: handle=129 (ACL), ctrl_cid=0x0041, intr_cid=0x0042, cod=0x00002504, flags=0x00003f00, incoming=0
        model: vid=0x1949, pid=0x0402, model='Android', name='REMAX RY-01'
        unijoysticle: type=gamepad, mode=enhanced, seat=0x03

--> Stop scanning for new gamepads
BR/EDR scan -> 0
BLE scan -> 0
unijoysticle: Bluetooth discovery mode is disabled
unijoysticle: device 00:90:E1:F0:DB:08 has new gamepad seat: 1
unijoysticle: Gamepad mode = normal
mice timers enabled/disabled: port A=0, port B=0
Connected devices:
idx=0:
        btaddr: 00:90:E1:F0:DB:08
        bt: handle=129 (ACL), ctrl_cid=0x0041, intr_cid=0x0042, cod=0x00002504, flags=0x00003f00, incoming=0
        model: vid=0x1949, pid=0x0402, model='Android', name='REMAX RY-01'
        unijoysticle: type=gamepad, mode=normal, seat=0x01

BR/EDR scan -> 1
BLE scan -> 1
unijoysticle: Bluetooth discovery mode is enabled

bihailantian655, 2023-04-17:
[image]

bihailantian655, 2023-04-17:
[image]

ricardoquesada, 2023-04-17:
you are using the "unijoysticle" platform. that's meat to be used for Unijjoysticle boards:
https://retro.moe/unijoysticle2/
is that what you want? 

my guess is that you need the arduino platform. See:
https://gitlab.com/ricardoquesada/bluepad32/-/blob/main/docs/architecture.md

If so, you have to use the Arduino builds. See:
https://gitlab.com/ricardoquesada/bluepad32/-/blob/main/docs/plat_arduino.md

And the Arduino example prints the pressed buttons

bihailantian655, 2023-04-17:
can i  build  the project 

[image]

bihailantian655, 2023-04-17:
Copy the content of  sketch.cpp  to   main.c  then  build

ricardoquesada, 2023-04-17:
> Copy the content of sketch.cpp to main.c then build

ohhhh... good to know. I'll fix it tonight.

ricardoquesada, 2023-04-17:
just double-checking, which ESP-IDF version are you using?
You should be using v4.4.x (not v5.x, not v4.3...)

bihailantian655, 2023-04-17:
4.4

bihailantian655, 2023-04-17:
[image]

ricardoquesada, 2023-04-17:
which OS are you using? Linux? mac?... which version?

bihailantian655, 2023-04-17:
it  is   github  work flow ,  i  just   delete  main.c then   rename sketch.cpp   to main.c

ricardoquesada, 2023-04-18:
mmm... seems to be compiling ok for me. using Github actions as well:

https://github.com/ricardoquesada/esp-idf-arduino-bluepad32-template/actions/runs/4728529709/jobs/8390158992

compare my repo with yours, and see whether there are any differences

bihailantian655, 2023-04-20:
https://github.com/bihailantian655/esp-idf-arduino-bluepad32-template/actions/runs/4750290071/jobs/8438332288   .

just  a small question  about  github action

ricardoquesada, 2023-08-10:
closing it.

closed 2023-08-10 by ricardoquesada