ricardoquesada/bluepad32 #34: Conflict with SerialBluetooth.h?

tbo711, 2023-05-02:
Hello.  Great work, this is amazing!  I am trying to add the use of SerialBluetooth to my sketch but it won't work.  I'm new to this but after hours of looking, my best guess is that it might be because Bluedroid mode is needed?

My end goal is to be able to connect my controller with Bluepad32 but also be able to connect to an Android phone to use something like the MIT App Inventor.

THANK YOU!
Blake

ricardoquesada, 2023-05-02:
Yes. BluetoothSerial is disabled.

See:
https://gitlab.com/ricardoquesada/esp-idf-arduino-bluepad32-template/-/blob/main/sdkconfig.defaults#L56

this is because it doesn't use Bluedroid. Instead it uses the BTstack stack (different one).

I would add Arduino APIs for BTstack, but you will need to implement the "serial" part yourself.

ricardoquesada, 2023-05-02:
not sure... just try with regular ESP32 Arduino core.... create a very simple example that uses BT and WiFi... if that works, then it should work with ESP32-Bluepad32 Arduino core

ste2425, 2023-09-28:
@tbo711 

This may not be of any help but i created an example using BTStack (the Bluetooth lib BluePad32 uses) to fire up a ATT Server and communicate with a web app. This was for me to build a web app that could perform OTA Updates on the EPS32 whilst using BluePad32.

It is possible to communicate with the ATT server via a mobile app or dekstop app if cofigured for BLE.

Obviously depends on what your are wanting to do but thought i would share as it may provide you an option.

https://github.com/ricardoquesada/bluepad32/issues/37#issuecomment-1720803083