ricardoquesada/bluepad32 #36: How to disconnect controller?

tapererwaj, 2023-05-20:
Is there a built-in method in this library that I can use to disconnect a connected controller after pressing any button on the controller , if not, can you suggest a way to achieve this.

ricardoquesada, 2023-05-20:
fixed, I'm assuming you are using the Arduino API.

will be part of the next stable release

https://gitlab.com/ricardoquesada/bluepad32/-/commit/227db920471c5fe78f40820e090124748df2cbcc
https://gitlab.com/ricardoquesada/bluepad32/-/commit/a702cc6c4b210c3df32f5ff86fcc5ce8b98c0a37

closed 2023-05-20 by ricardoquesada

tapererwaj, 2023-05-20:
Hello, thank you for your reply.

When will the next stable release be available?

ricardoquesada, 2023-05-20:
I don't have a date, but soon :)

perhaps in one week or less

tapererwaj, 2023-05-20:
Okay, much appreciated

ricardoquesada, 2023-05-21:
v3.7.2 released

tapererwaj, 2023-05-21:
> fixed, I'm assuming you are using the Arduino API.
> 
> will be part of the next stable release
> 
> https://gitlab.com/ricardoquesada/bluepad32/-/commit/227db920471c5fe78f40820e090124748df2cbcc https://gitlab.com/ricardoquesada/bluepad32/-/commit/a702cc6c4b210c3df32f5ff86fcc5ce8b98c0a37

Can I also use this for the ESP32 DEVKIT?

ricardoquesada, 2023-05-21:
yep. if it works with regular ESP32 Arduino, then it should work with ESP32-Bluepad32 Ardunio.

tapererwaj, 2023-05-21:
I have tried using the disconnect class method through the Controller example as shown in the image below:

[image]

But the ESP32 just reboots and gives me the following error:
[image]

How can I resolve this issue?

ricardoquesada, 2023-05-21:
mmm.... let me double check it.

ricardoquesada, 2023-05-21:
I cannot reproduce it... seems to be working ok with me.

Please provide:

- Platform you are using:  Arduino IDE (which version), Arduino ESP-IDF (which ESP 4.4 version), etc..
- OS: Linux, Windows, etc... (which version)
- Gamepad and the FW used by the gamepad
- somewhere post your  entire sketch (not a screenshot... the code so that I can copy paste it)
- the more info you add, the easier for me

tapererwaj, 2023-05-22:
Platform: Arduino IDE v2.1.0
ESP32 Chip - ESP32-D0WDQ6 (revision v1.0)
esptool.py v4.5.1
OS - Windows 11
Controller - PS5 Dual Sense

Arduino IDE Sketch:
[Controller.zip](https://github.com/ricardoquesada/bluepad32/files/11531236/Controller.zip)

ricardoquesada, 2023-05-23:
In you code you have:

```
      if (myGamepad->x()) {
        // try to disconnect the controller
        myGamepad->disconnect();
        return;  // <--- DO NOT RETURN FROM update();
      }
```

just remove the `return;`

tapererwaj, 2023-05-23:
It works, thank you

ricardoquesada, 2023-05-23:
look at the NINA code, which is similar as what you want but using SPI:
https://gitlab.com/ricardoquesada/bluepad32/-/blob/main/docs/plat_nina.md