ricardoquesada/bluepad32 #37: Add BTStack examples for Arduino IDE

ste2425, 2023-06-02:
Please provide examples on the use of BTStack for communicating with the ESP32.

**Additional context**
The conversation began trying to compile OTA examples to implement OTA firmware uploads whilst using Bluepad32. The gist describing these issues can be found [here](https://gist.github.com/ste2425/c601fb16df8ece2c91c7a65b2f29344a) for reference.

It would be fantastic if the examples could include OTA examples however that may be a stretch so including BTStack usage examples which would allow a developer to implement their own OTA implantation would also be great.

ste2425, 2023-09-04:
As discussed on the Discord i have a little example repo of a web app communicating the the ESP32 running BluePad 32.

Here is a short video of it in action: https://youtu.be/PmGVCzsbee8

Here is the repo: https://github.com/ste2425/BluePad32BLEDemo/tree/main

Hopefully this can be of some use in writing some demo's into BluePad32. Please feel free to use directly or modify any of this code if it is at all usefull.

Going to extend it next to include OTA functionality.

ricardoquesada, 2023-09-08:
How do you envision adding it ?

For example, what should be the name: "WebAppService"  ? Do you have a better name ?

And the WebApp application, should it be part of the example ? or perhaps a link to your site ?

I also think I'd rename some files/functions... for example the file `ble.h` could be confusing... perhaps `ble_service.h` or something like that to avoid confusion.

What do you think?

ste2425, 2023-09-11:
As they say naming things and caching are the hardest problems to solve so not too sure on the web app name!

It could be called `WebAppBLEService` to match the proposed name change in the ESP (`ble_service.h`).

IMHO i think having the source code for both the WebApp and the ESP firmware in one location is rather important especially for the OTA Update demo i working on now. The webapp and the ESP firmware form two halves of the same coin so having them easily referenceable (especially for newcomers) i think will be useful.

Now where that source code lives (a repo i own or part of BluePad32) i'm happy to be your call. I do not know if you desired the EPS32 firmware to be accessible from Arduino IDE examples like the controller example its based of. If so that may influence the code structure and where it lives (would the web app code break the Arduino IDE integration for example?).

As for hosting the Web App, I'm happy to deploy the latest main to my GitHub pages.

Finally naming of functions again I'm happy to make any changes you feel are needed, just suggest and make them :)

I have a working OTA update example which extends the example linked above. It may be worth just having one example that has everything in rather than two (one with OTA and one without).

However the OTA example does have a problem which i have logged [here](https://groups.google.com/g/btstack-dev/c/tyDfoxOO1Hw) looking for support (i believe it is a me issue using BTStack and not a BluePad32 issue, however if you could offer any advise i would greatly appreciate it.)

ste2425, 2023-09-15:
Right so as mentioned int he Discord chat i now have a working end to end OTA example combined with BluePad32. In my testing not had a single disconnect or error.

The final example ESP code is here: https://github.com/ste2425/BluePad32OTAExample
The corresponding web app to perform the OTA is here: https://github.com/ste2425/OTAWebapp
It is also hosted on my GitHub pages here: https://ste2425.github.io/OTAWebapp/

This is the implementation I'm going to build upon for my PSP consolizer (i will fork the linked repo).

The repo can be used for reference if you wish to provide your own official BluePad32 example or i can clean up and make any suggested changes if you wish the linked repos to be the official examples. Thanks for your help on getting to this point :)

ricardoquesada, 2023-10-12:
Added example to esp-idf-bluepad32-template project:

https://github.com/ricardoquesada/esp-idf-arduino-bluepad32-template/commit/dbcc0d176f69a1f4515c27be981bdd98c3016cef

ricardoquesada, 2023-10-12:
Thanks for the patch!
Finally I released 3.9.0 for ARduino IDE.

If you happen to update the example, could you please update this folder ?
https://gitlab.com/ricardoquesada/esp-idf-arduino-bluepad32-template/-/tree/develop/examples/ota?ref_type=heads

I'm copy+pasting files from there to the Arduino IDE. ty!

closed 2023-10-12 by ricardoquesada