ricardoquesada/bluepad32 #38: [Bug]:

davidxgames, 2023-06-30:
### What happened?

A bug happened! thumbR and thumbL buttons not working
 working:
```
            if (myGamepad->l2()) {
                myGamepad->setRumble(0xc0 /* force */, 0xc0 /* duration */);
            }
```
           not working:
```
            if (myGamepad->thumbR()) {
                myGamepad->setRumble(0xc0 /* force */, 0xc0 /* duration */);
            }
            if (myGamepad->thumbL()) {
                myGamepad->setRumble(0xc0 /* force */, 0xc0 /* duration */);
            }
```

### Bluepad32 Version

3.7.2 (latest stable)

### Bluepad32 version custom

Example: Using Git develop branch commit hash #xxxxxxx

### Bluepad32 Platform

Arduino ESP-IDF

### Platform version

ESP-IDF v4.4.4

### ESP32 chip

ESP32

### ESP32 board

ESP32 WROOM DevKit 

### OS

Windows

ricardoquesada, 2023-06-30:
which gamepad are you using?

davidxgames, 2023-06-30:
> which gamepad are you using?

DS5

davidxgames, 2023-06-30:
The button works in serial shows this, buttons: 0x0200, but it does not activate the rumble.

ricardoquesada, 2023-06-30:
so the bug is that rumble doesn't work on DualSense ?

davidxgames, 2023-06-30:
I already saw the problem, when the rumble function is like this it does not work

`            if (rumble) {
                if (myGamepad->r1()) {
                    digitalWrite(RPin, LOW);
                    myGamepad->setRumble(0xc0 /* force */, 0xc0 /* duration */);
                } else {
                    digitalWrite(RPin, HIGH);
                }
            }`

davidxgames, 2023-06-30:
> so the bug is that rumble doesn't work on DualSense ?

A thousand apologies, I already checked and the problem is how I am writing the code, thanks for answering, sorry for the wasted time.

closed 2023-06-30 by davidxgames