ricardoquesada/bluepad32 #39: Missing XIAO ESP32S3 board

Teneppa, 2023-07-04:
The board list of esp32_bluepad (version 3.7.3) is missing a XIAO ESP32S3 which is present on the default esp32 board list. (version  2.0.9)

Here's also a link to the board in question: https://www.seeedstudio.com/XIAO-ESP32S3-p-5627.html

Tested on Arduino IDE 2.1.1

---
Here's the board on the default esp32 board list:
[image]

Teneppa, 2023-07-12:
Found the source of this issue, the boards.txt is missing the XIAO_ESP32S3 configurations. Seems like this has been forgotten, because the board can be found already in the variants folder.

ricardoquesada, 2023-08-10:
fixed 
https://github.com/ricardoquesada/esp32-arduino-lib-builder/releases/tag/3.8.3

closed 2023-08-10 by ricardoquesada