ricardoquesada/bluepad32 #41: PS Move Controller connects, but returns 0 for all values and doesn't process correctly [Bug]:

holden-as-published, 2023-08-15:
### What happened?

A bug happened! Using Bluepad32 1.3.0 + NINA 3.7.0 fw on Arduino RP2040Connect, a previously known good configuration.  PS4 Move controller connects but returns 0s (sometimes a really big number but 99% of the time it's 0s) on all fields with the default controller class. Also, processGamepad may not be running correctly as hardcoded output values for leds or rumble do nothing. 

### Bluepad32 Version

3.7.0

### Bluepad32 version custom

Example: Using Git develop branch commit hash #xxxxxxx

### Bluepad32 Platform

Arduino + NINA

### Platform version

 Arduino IDE 2.1.0

### Controller

PsMove

### ESP32 chip

ESP32

### ESP32 board

Unsure

### OS

Windows

JuPrgn, 2023-11-29:
I also noticed that the LED value and rumble strength don't seem to have any effect it seems to be all or nothing (0 or 255).
I'm going to do some more tests to try to figure it out

JuPrgn, 2023-11-30:
Looks like some controllers do a calibration I got some clean accel datas as you can see on #52 with Data frog clone but non zero datas with official DS4

JuPrgn, 2023-11-30:
Is there a way to access Arduino installed library to modify Bluepad src ? I would like to try the changes you did in https://github.com/ricardoquesada/bluepad32/commit/49497c39266eec4201ae5bb1ab945fef957017e0

ricardoquesada, 2023-11-30:
@JuPrgn not easy what you want... the "Bluepad32 Arduino Core" is a big pre compiled library.

Although, thinking about it... for v4 probably it will make sense to split it in two:
 
- Arduino Core with needed changes for Bluepad32
- Bluepad32 library (although it might be confusing for the existing Bluepad32 library for NINA chips)

ricardoquesada, 2023-11-30:
PSMove is "PS Move Controller"... not the a DualShock controller... not sure what you mean by "fixed LED values"   "rumble force"... please file a bug for those.