ricardoquesada/bluepad32 #42: I want to get rssi

potaka001, 2023-08-31:
Is it possible to get rssi?

ricardoquesada, 2023-08-31:
could you be more specific ? which platform ? use case ? etc ( https://tinyurl.com/bluepad32-help )

potaka001, 2023-08-31:
I'm using an ESP32 Dev kit and an Android TV Game controller.
BTstack can get the RSSI value, but how do I get the RSSI value when using Bluepad32?
I would like to check the Bluetooth signal quality by checking the Received Signal Strength Indicator.

potaka001, 2023-08-31:
I would like to record the rssi when input lag is measured.

apoll2000, 2024-01-10:
Hi, this feature would also be really helpful for me. I am using a PS5 controller as a remote controller with an ESP32 running the bluepad32 arduino library, and reading out the RSSI would be useful as a safety feature so that a drop in signal strength can be detected.
Is it possible to access the RSSI somehow? Thank you very much in advance!

ricardoquesada, 2024-01-10:
@apoll2000 as a safety measure you can do something similar to what I described here:
https://github.com/ricardoquesada/bluepad32/issues/45#issuecomment-1858839758

Regarding RSSI, the entire BTstack API is avaiable for Arduino users... you can call any API.
Wondering what is the API that returns the RSSI... I know the RSSI appears for BR/EDR connections when they advertise... 

Not sure how to get it runtime... but if BTstack has that API, then you should be able to  call it.
You might need the `connection handle`, which is not available to the Arduino API, but I can expose it so you can  use.... if you can tell me which BTstack API you need to call, I can provide the needed arguments for that API

apoll2000, 2024-01-25:
Thank you very much for your quick response, I had to read into everything a bit to understand.

The RSSI readout is implemented in the BTStack GAP API, with the function `int gap_read_rssi(hci_con_handle_t con_handle)`, and it indeed requires the connection handle (function listed [here](https://bluekitchen-gmbh.com/btstack/#appendix/gap/) in the documentation).

It would be great if you could help me access that from within Arduino.

Also just wanted to say thank you for this great library, it's already helping me a lot!

ricardoquesada, 2024-02-08:
I've just some changes so that this issue can be mitigated:

- `BP32.update()` returns whether data has been received in any of the controllers.
- `Controller.hasData()` returns whether data has been received since `BP32.update()` was called.

Sketch updated:
https://github.com/ricardoquesada/esp-idf-arduino-bluepad32-template/blob/develop/main/sketch.cpp#L274

and here:
https://github.com/ricardoquesada/esp-idf-arduino-bluepad32-template/blob/develop/main/sketch.cpp#L229

apoll2000, 2024-02-22:
Thank you for this, the `update()` method seems to solve exactly the issue I was having when data did not come through.

TheRob87, 2024-07-22:
> I've just added some changes so that this issue can be mitigated:
> 
> * `BP32.update()` returns whether data has been received in any of the controllers.
> * `Controller.hasData()` returns whether data has been received since `BP32.update()` was called.
> 
> Sketch updated: https://github.com/ricardoquesada/esp-idf-arduino-bluepad32-template/blob/develop/main/sketch.cpp#L274
> 
> and here: https://github.com/ricardoquesada/esp-idf-arduino-bluepad32-template/blob/develop/main/sketch.cpp#L229

Hi,

it seems `.update()` and `.hasData()` has the same State so the ELSE condition will never be called.

I tried that like the FAQ says, but the print output (2 and 3) come never. 

My Code (connected with XBox Controller)

```

void processControllers() {
    for (auto myController : myControllers) {
        //Serial.printf("Call Update %d\n",(millis() - lastUpdate));
        if (myController && myController->isConnected() ){
        Serial.printf("Call Update  1 %d\n",(millis() - lastUpdate));
          if( myController->hasData()) {
            lastUpdate = millis();
            if (myController->isGamepad()) {
              processGamepad(myController);
            } 
          }
          else {
            Serial.printf("Call Update 2 %d\n",(millis() - lastUpdate));
            if(speed != 0){
              
              if((millis() - lastUpdate) > maxUpdate ){
                maxUpdate = (millis() - lastUpdate);
                Serial.printf("Call Update 3 %d\n",maxUpdate);
              }
            }
          }
        }
    }
}
```

ricardoquesada, 2024-07-24:
> it seems `.update()` and `.hasData()` has the same State so the ELSE condition will never be called.

What `update()` does is to fetch data from Bluepad32 process to the Arduino process.
Data might be available, or not.
So, `hasData()` is exactly for that

MikeyC3040, 2024-12-20:
I'm finding the same issue as https://github.com/ricardoquesada/bluepad32/issues/42#issuecomment-2241860641

If I have 1 controller connected, and manually remove controller power or turn it off (in my case pushing the button on the joycon) to simulate it being too far away. BP32.Update() returns false [as it's designed to do](https://github.com/ricardoquesada/esp-idf-arduino-bluepad32-template/blob/1c4cb01c261961fe6dce3d323b09ed774f9511c8/components/bluepad32_arduino/ArduinoBluepad32.cpp#L29C1-L29C64) when no controllers have updates, and therefore never would actually run what is in the check loop including the timeout, yet the disconnect callback still doesn't run for that 20 seconds or so negating the use of a timeout.

I'd recommend updating the [FAQ,](https://bluepad32.readthedocs.io/en/latest/FAQ/#how-to-detect-when-a-gamepad-is-out-of-range) in my case I pulled the processControllers loop out of BP32.Update() check.