ricardoquesada/bluepad32 #43: Moving PS5 controller away from ESP32 cause unhandled Exception

gunnicus, 2023-09-05:
Hello sir, I'm not sure where should I post this issue, I also posted this issue at :
https://github.com/ricardoquesada/bluepad32-arduino/issues
If it creates a duplicate, I will delete that post.

I connected a PS5 controller to ESP32, it is working fine unless I walk away with the PS5 controller from the ESP32. The issue starts to happen when the PS5 controller 6 meters away from the ESP32.

What happened:
I'm using 'Controller' from example. Upload Controller to ESP32.
[image]

Pair PS5 controller with ESP32
Take PS5 controller about 6 memters away from ESP32, turn myself around make sure my body is between the PS5 controller and the ESP32's direction. Keep walking untill abount 10 meters away from ESP32, the PS5 controller turned off itself.

Below error happened in ESP32

FEX 0 0
FEX 0 0
FEX 7 0
ASSERT_PARAM(0 7), in ld_acl.c at line 1772
Guru Meditation Error: Core 0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4008fed4: 00f01d00 00004136 f01d0000
Core 0 register dump:
PC : 0x4008fedb PS : 0x00060f34 A0 : 0x8008573e A1 : 0x3ffc1f50
A2 : 0x00000000 A3 : 0x00000000 A4 : 0x00000007 A5 : 0x3ff9cd42
A6 : 0x000006ec A7 : 0xfffffffb A8 : 0x8000814b A9 : 0x3ffc1ec0
A10 : 0x00000000 A11 : 0x3ffc1ee4 A12 : 0x3ffc1e8f A13 : 0x00000032
A14 : 0x00000000 A15 : 0x3ffc1e95 SAR : 0x00000004 EXCCAUSE: 0x00000000
EXCVADDR: 0x00000000 LBEG : 0x40090f09 LEND : 0x40090f19 LCOUNT : 0xfffffffe

Backtrace: 0x4008fed8:0x3ffc1f50 0x4008573b:0x3ffc1f70 0x400327cb:0x3ffc1f90 0x40087bb5:0x3ffc1fd0 0x40087e89:0x3ffc1ff0 0x4008bde6:0x3ffc2020 0x4008c8af:0x3ffc2040 0x400850be:0x3ffc2060 0x400853dd:0x3ffc2080 0x400853bf:0x00000000 |<-CORRUPTED

ELF file SHA256: 60dedc08e98be6b0

Rebooting...

ricardoquesada, 2023-09-05:
mmm.... Interesting

gunnicus, 2023-09-05:
It is a bit odd, I suppose the disConnected event should be triggered, but actually unhandled Exception been thrown. 
I noticed FEX 0 0
FEX 0 0
FEX 7 0 showed many times before the exception, what does this FEX mean.

ricardoquesada, 2023-09-08:
Any chance that you can symbolize the stacktrace (e.g: convert the addresses to names) ?

I've just tried to reproduce it on a Sparkfun ESP32 Thing + Controller.ino and it disconnected without any problem.

gunnicus, 2023-09-08:
Hello sir, thanks for your reply, sorry I'm new to esp32, not too sure how can I symbolize the stacktrace, this issue doesn't happen every time but it happens pretty often, I can replicate this issue after a few attempts. I do turn the ESP32 power off and on again everytime I tried. Also while I was walking out of range, I was pressing down X button, pressing down L1,L2,R1,R1, rolling around left joysticks none stop. 

Please see below log output is I replicated this issue tonight. You may notcie before the  21:11:28.197 -> scan_evt timeout happened the controller already stuck with same input for about 10 seconds even I was moving the left joystick none stop the whole time.

21:11:00.494 -> CALLBACK: Controller is connected, index=0
21:11:00.529 -> BTAddr: 88:03:4c:73:18:22, VID/PID: 054c:0ce6, flags: 0x07
21:11:00.596 -> idx=0, dpad: 0x00, buttons: 0x0000, axis L:    4,   12, axis R:    4,    0, brake:    0, throttle:    0, misc: 0x00, gyro x:   -62 y:   374 z:  -124, accel x:   192 y:  8304 z:  -910, battery: 201
21:11:00.870 -> idx=0, dpad: 0x00, buttons: 0x0000, axis L:    4,   12, axis R:    4,    0, brake:    0, throttle:    0, misc: 0x00, gyro x:  -496 y:   124 z:   187, accel x:   235 y:  8353 z:  -876, battery: 201
21:11:01.144 -> idx=0, dpad: 0x00, buttons: 0x0000, axis L:    4,   12, axis R:    4,    0, brake:    0, throttle:    0, misc: 0x00, gyro x: -1490 y:  -436 z:   374, accel x:   179 y:  8336 z:  -801, battery: 201
21:11:01.416 -> idx=0, dpad: 0x00, buttons: 0x0000, axis L:    4,   12, axis R:    4,    0, brake:    0, throttle:    0, misc: 0x00, gyro x:-22909 y:  4494 z: -8356, accel x:    12 y:  8406 z:  -425, battery: 201
21:11:01.690 -> idx=0, dpad: 0x00, buttons: 0x0000, axis L:    4,   12, axis R:    4,    0, brake:    0, throttle:    0, misc: 0x00, gyro x:-37623 y:  4120 z: -5862, accel x:   253 y: 10003 z:  1758, battery: 201
21:11:01.964 -> idx=0, dpad: 0x00, buttons: 0x0000, axis L:    4,   12, axis R:    4,    4, brake:    0, throttle:    0, misc: 0x00, gyro x: 11485 y: 12484 z: 11787, accel x:  -181 y:  8641 z:   716, battery: 201
21:11:02.237 -> idx=0, dpad: 0x00, buttons: 0x0000, axis L:    4,   12, axis R:    4,    0, brake:    0, throttle:    0, misc: 0x00, gyro x: 29428 y: 70727 z:  5176, accel x:  -950 y:  8197 z:   123, battery: 201
21:11:02.512 -> idx=0, dpad: 0x00, buttons: 0x0000, axis L:    4,   12, axis R:    4,    0, brake:    0, throttle:    0, misc: 0x00, gyro x:  9312 y: 63236 z: -3617, accel x:  -774 y:  8967 z:  -345, battery: 201
21:11:02.786 -> idx=0, dpad: 0x00, buttons: 0x0000, axis L:   12,    8, axis R:    4,    0, brake:    0, throttle:    0, misc: 0x00, gyro x: 16514 y: 37517 z:  -249, accel x: -1207 y:  8200 z: -1304, battery: 201
21:11:03.062 -> idx=0, dpad: 0x00, buttons: 0x0000, axis L:    8,    8, axis R:    4,    0, brake:    0, throttle:    0, misc: 0x00, gyro x:  7139 y: 46444 z:-15154, accel x:   888 y:  7236 z:  -727, battery: 201
21:11:03.302 -> idx=0, dpad: 0x00, buttons: 0x0000, axis L:   12,    8, axis R:    4,    0, brake:    0, throttle:    0, misc: 0x00, gyro x:-11671 y: 75659 z:  9604, accel x:  1239 y:  8356 z:  -945, battery: 201
21:11:03.577 -> idx=0, dpad: 0x00, buttons: 0x0000, axis L:   12,    8, axis R:    4,    0, brake:    0, throttle:    0, misc: 0x00, gyro x:  7077 y: 42011 z:-27315, accel x:   723 y:  8704 z: -1245, battery: 201
21:11:03.852 -> idx=0, dpad: 0x00, buttons: 0x0000, axis L:   12,    8, axis R:    4,    0, brake:    0, throttle:    0, misc: 0x00, gyro x:  4408 y: 13483 z:  -686, accel x:  -873 y:  7930 z:  -157, battery: 201
21:11:04.126 -> idx=0, dpad: 0x00, buttons: 0x0001, axis L:   12,    8, axis R:    4,    0, brake:    0, throttle:    0, misc: 0x00, gyro x:-13099 y:-13046 z: -8294, accel x:    80 y:  8503 z:  -861, battery: 201
21:11:04.401 -> idx=0, dpad: 0x00, buttons: 0x0001, axis L:   12,    8, axis R:    4,    0, brake:    0, throttle:    0, misc: 0x00, gyro x: -8319 y:-45133 z: -8980, accel x:   200 y:  9947 z:  -531, battery: 201
21:11:04.675 -> idx=0, dpad: 0x00, buttons: 0x0001, axis L:   12,    8, axis R:    4,    0, brake:    0, throttle:    0, misc: 0x00, gyro x:  7512 y:-13234 z: -7296, accel x:    92 y:  7145 z:  -846, battery: 201
21:11:04.949 -> idx=0, dpad: 0x00, buttons: 0x0001, axis L:   12,    8, axis R:    4,    0, brake:    0, throttle:    0, misc: 0x00, gyro x:  2235 y: 13483 z: -2120, accel x:   577 y:  7921 z: -1963, battery: 201
21:11:05.222 -> idx=0, dpad: 0x00, buttons: 0x0001, axis L:   12,    8, axis R:    4,    0, brake:    0, throttle:    0, misc: 0x00, gyro x: 17694 y: 12921 z: -4365, accel x:   321 y:  6865 z:  -798, battery: 201
21:11:05.495 -> idx=0, dpad: 0x00, buttons: 0x00c1, axis L:   12,    8, axis R:    4,    0, brake:  856, throttle:  740, misc: 0x00, gyro x: -3290 y:-14919 z: -4178, accel x:   354 y:  8069 z:  -994, battery: 201
21:11:05.770 -> idx=0, dpad: 0x00, buttons: 0x00c1, axis L:   16,    8, axis R:    4,    0, brake: 1020, throttle: 1020, misc: 0x00, gyro x: 10554 y:-19663 z: 10040, accel x:   313 y:  8835 z: -1909, battery: 201
21:11:06.045 -> idx=0, dpad: 0x00, buttons: 0x00c1, axis L:    8,   12, axis R:    4,    0, brake: 1020, throttle: 1020, misc: 0x00, gyro x:-10740 y:  -936 z:-13221, accel x:    68 y:  8157 z:  -124, battery: 201
21:11:06.286 -> idx=0, dpad: 0x00, buttons: 0x00d1, axis L:   16,  -24, axis R:    4,    4, brake: 1020, throttle: 1020, misc: 0x00, gyro x:-10119 y: 38328 z:-10352, accel x:   657 y:  8039 z: -1382, battery: [... cut ...]
21:11:41.186 -> Backtrace: 0x4008fed8:0x3ffc1f50 0x4008573b:0x3ffc1f70 0x400327cb:0x3ffc1f90 0x40087bb5:0x3ffc1fd0 0x40087e89:0x3ffc1ff0 0x4008bde6:0x3ffc2020 0x4008c8af:0x3ffc2040 0x400850be:0x3ffc2060 0x400853dd:0x3ffc2080 0x400853bf:0x00000000 |<-CORRUPTED
21:11:41.424 -> 
21:11:41.424 -> 
21:11:41.424 -> 
21:11:41.424 -> 
21:11:41.424 -> ELF file SHA256: b43d13d264ec5680
21:11:41.459 -> 
21:11:41.459 -> Rebooting...
21:11:41.493 -> ⸮⸮zY⸮bkbb⸮H⸮	⸮ۤ⸮Firmware version installed: Bluepad32 for Arduino v3.8.3
21:11:42.878 -> BD Address: D4:D4:DA:57:B7:CE

ricardoquesada, 2023-09-08:
yep. not sure what could be causing it... unfortunately I cannot reproduce it.

The backtrace could be useful if you somewhat can get the symbols of it. I cannot do that for you compiled the sketch.

ricardoquesada, 2023-09-08:
try with this site: https://maximeborges.github.io/esp-stacktrace-decoder/

gunnicus, 2023-09-09:
Thanks a lot, I tried again, please note I commented out  BP32.forgetBluetoothKeys() in setup(), 
below is Stack trace output : 
0x4008fed8: r_assert at /home/riq/bin/esp/esp-idf-v4.4/components/bt/controller/esp32/bt.c:1892
0x400850be: hli_c_handler at /home/riq/bin/esp/esp-idf-v4.4/components/bt/controller/esp32/hli_api.c:101

15:57:01.296 -> FEX 0 0
15:57:01.329 -> FEX 0 0
15:57:01.364 -> idx=0, dpad: 0x00, buttons: 0x0001, axis L:   16,   24, axis R:  -76, -508, brake:    0, throttle:    0, misc: 0x00, gyro x: -23FEX 0 0
15:57:01.498 -> FEX 7 0
15:57:01.498 -> ASSERT_PARAM(0 7), in ld_acl.c at line 1772
15:57:01.566 -> Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
15:57:01.668 -> Memory dump at 0x4008fed4: 00f01d00 00004136 f01d0000
15:57:01.701 -> Core  0 register dump:
15:57:01.736 -> PC      : 0x4008fedb  PS      : 0x00060f34  A0      : 0x8008573e  A1      : 0x3ffc1f50  
15:57:01.836 -> A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000007  A5      : 0x3ff9cd42  
15:57:01.905 -> A6      : 0x000006ec  A7      : 0xfffffffb  A8      : 0x8000814b  A9      : 0x3ffc1ec0  
15:57:02.008 -> A10     : 0x00000000  A11     : 0x3ffc1ee4  A12     : 0x3ffc1e8f  A13     : 0x00000032  
15:57:02.111 -> A14     : 0x00000000  A15     : 0x3ffc1e95  SAR     : 0x00000004  EXCCAUSE: 0x00000000  
15:57:02.214 -> EXCVADDR: 0x00000000  LBEG    : 0x40090f09  LEND    : 0x40090f19  LCOUNT  : 0xfffffffe  
15:57:02.281 -> 
15:57:02.316 -> 
15:57:02.316 -> Backtrace: 0x4008fed8:0x3ffc1f50 0x4008573b:0x3ffc1f70 0x400327cb:0x3ffc1f90 0x40087bb5:0x3ffc1fd0 0x40087e89:0x3ffc1ff0 0x4008bde6:0x3ffc2020 0x4008c8af:0x3ffc2040 0x400850be:0x3ffc2060 0x400853dd:0x3ffc2080 0x400853bf:0x00000000 |<-CORRUPTED
15:57:02.556 -> 
15:57:02.556 -> 
15:57:02.556 -> 
15:57:02.556 -> 
15:57:02.556 -> ELF file SHA256: 72aa410f4eb851d0
15:57:02.590 -> 
15:57:02.625 -> Rebooting...
15:57:02.625 -> 	1⸮⸮(Y⸮b[⸮V7|⸮$q⸮⸮⸮Firmware version installed: Bluepad32 for Arduino v3.8.3
15:57:03.646 -> BD Address: D4:D4:DA:57:B7:CE

[image]
[image]

gunnicus, 2023-09-26:
Hello sir @ricardoquesada , I figured out a workaround. I notcied this issue happen after scan_evt timeout, so I added BP32.enableNewBluetoothConnections(false); inside onConnectedController. I belive enableNewBluetoothConnections(false) makes scan_evt stop. 
Basically, I tried to stop scanning after my controller connected to ESP32, that seems worked, I haven't encountered the issue since. 
I will do more testing.

ricardoquesada, 2023-09-26:
mmm... very interesting. I wonder if BLE is causing that.