ricardoquesada/bluepad32 #45: Possible to set Disconnect Timeout or get Last Bluetooth Update Time?

JoeSc, 2023-10-04:
**Is your feature request related to a problem? Please describe.**
I have 3 controllers, xbox one, stadia, and 8bitdo SN30pro+.  

The issue I am trying to avoid is if the controller goes out of range or shuts off the last joystick values will be preserved until the bluetooth stack disconnects, I have seen this take a long time.

On the xbox one controller it takes 0.6s from when I power off the controller(last value change when I am moving the joystick in circles) until bluepad32 disconnects the controller.  This is not bad, only an out of control robot/thing for 600ms
On the stadia controller it takes ~4 seconds from power off until blluepad32 disconnects, this is borderline bad.
On the 8bitdo SN30pro+ it takes ~18 seconds from power off until bluepad32 disconnects, this is bad.

**Describe the solution you'd like**
A method or field that showed if new controller data has been received since last calling of `BP32.update()`.
Or a snapshot of CPU `millis()` each time the controller data is received.

**Describe alternatives you've considered**
I have thought about adding some kind of "are all the inputs identical for too long" but that doesn't work if you are just driving straight for a long time.

**Additional context**
I'm going to start digging into the code, but I figured I'd ask here first since maybe I'm missing something

ricardoquesada, 2023-10-04:
Perhaps there is a way to ping the "db" (how strong the signal is)... and report that feature.

A "ping" or "keep alive" also, but I don't know whether Bluetooth has that feature

RC0D3, 2023-12-14:
Yeah... I have a car with xbox controller, works as well, but when disconnect, by code of Bluepad32 have a timeout of 20s, then if I'm pressing to accelerate and lose connection, my car continue forward movement and crash in any obstacle :/
I try to find some way to do that manually but don't have how, no true feedback from bluepad.

ricardoquesada, 2023-12-15:
yes, I should provide a way (assuming it is possible) that the controller might be out-of-range soon... I need to investigate it a bit more

ricardoquesada, 2023-12-16:
@JoeSc @RC0D3 

I recommend using something like this:

in your main loop, you should receive a "gampad event" almost everyframe.
if you don't receive an event in 250ms, then either:

- no new events were received (buttons/axis are not being pressed)
- or the gamepad was disconnected/outside the range

in any of those cases, just turn off the motors, etc.

JochenKr, 2024-02-16:
Hi,
I'm facing a similar issue using 8bitdo Controller.
Can you explain more specific how to implement such a "gamepad event" flag which could be used to react on?
thx

ricardoquesada, 2024-02-16:
@JochenKr See this comment. It explains how to use it: https://github.com/ricardoquesada/bluepad32/issues/42#issuecomment-1933270864

JochenKr, 2024-02-17:
Thanks. Tried it and it improves the situation a lot. Still I need to add some delay like 2s, because the response from update() seems to be often false. But this should be good enough for my case

ricardoquesada, 2024-02-17:
I didn't understand the delay that you need to add.

Take a look at the updated Sketch to learn how to use correctly:

https://github.com/ricardoquesada/esp32-arduino-lib-builder/blob/release/v4.4/bluepad32_files/libraries/Bluepad32_ESP32/examples/Controller/Controller.ino

search for `hasData` and `update` in that file... those are the two new functions added.