ricardoquesada/bluepad32 #50: Two idx are displayed with one connection.(PS4 controller esp32WROVER-IE)

8Kei, 2023-10-29:
Hello sir , I am tried the sample as is, the following is displayed when a ps4 controller is connected.
When second ps4 controllers are connected, idx = 2 and idx=3 are shown, but data is not updated for idx=3.

Also,When the disconnect() function is called on the second ps4 controller, only the first ps4 controller will be disconnected.

(I'm using 'Controller.ino' from example in Arduino IDE . Upload Controller.ino to ESP32.)

**Display on serial monitor:**
・CALLBACK: Gamepad is connected, index=0
・Gamepad model: DualShock 4, VID=0x054c, PID=0x09cc
・CALLBACK: Gamepad is connected, index=1
・Gamepad model: DualShock 4, VID=0x054c, PID=0x09cc
・idx=0, dpad: 0x00, buttons: 0x0000, axis L:   -4,   -4, axis R:    4,    8, brake:    0, throttle:    0, misc: 0x00, gyro x:   752 y:  -874 z:  -311, accel x:   -96 y:  8265 z:  1740
・idx=1, dpad: 0x00, buttons: 0x0000, axis L:    0,    0, axis R:    0,    0, brake:    0, throttle:    0, misc: 0x00, gyro x:     0 y:     0 z:     0, accel x:     0 y:     0 z:     0
・CALLBACK: Gamepad is disconnected from index=0
・CALLBACK: Gamepad is disconnected from index=1

8Kei, 2023-10-29:
Platform:Arduino IDE for ESP32 modules
ESP32 Module:esp32 wrover ie
Controller:ps4 controller (MODEL CUH-ZCT2J)
Sketch :https://gist.github.com/8Kei/555e1bcd2e3acb1382e58c518f0c3ec3
OS:windows 11
Debug log(Serial monitor):
Pressing a button on the second controller will also display this statement.

22:24:02.287 -> CALLBACK: Gamepad is disconnected from index=0
22:24:02.287 -> CALLBACK: Gamepad is disconnected from index=1

ricardoquesada, 2023-10-29:
what Bluepad32 version?

8Kei, 2023-10-29:
version 3.9.0
[image]

ricardoquesada, 2023-10-29:
v3.9.0 and newer have "virtual devices" support. Meaning that controllers like the DualShock4 and DualSense will have two "devices" when one is connected.
The first one is the gamepad.
The second one is a mouse... you can use touchpad as the mouse.
If you disconnect the real device (gamepad), the mouse will be disconnected as well

8Kei, 2023-10-29:
I'm so Sorry, I typed that out and didn't realize how vague it ended up being.
When a real second controller is connected, the connection of the first controller is disconnected.

ricardoquesada, 2023-10-29:
I really don't understand the problem...

add more details please... otherwise I wouldn't know how to help you.

8Kei, 2023-10-29:
When I press the button on the real second controller and try to disconnect by calling disconnect as in the source code, only those two lines are called back and displayed.

8Kei, 2023-10-29:
I want to disconnect only the second connected controller with different buttons assigned. However, when I press the button on the second controller, the first controller is disconnected and I do not know how to disconnect the second controller.

ricardoquesada, 2024-01-11:
got it. you can use the allowlist API.

https://github.com/ricardoquesada/bluepad32/blob/main/src/components/bluepad32/include/uni_bt_allowlist.h

microbmen, 2024-02-01:
I am seeing this bug too. 

if I call ctl->disconnect(); on any index ... i.e. index 1 ... only the controller on index 0 is disconnected.  

I'm on Bluepad32  3.10.2

to reproduce, use your example .ino ...  make ctl->y() perform ctl->disconnect()  ... connect 2 or more controllers... pressing ctl->y() on any of the controllers only disconnects controller index 0;

ricardoquesada, 2024-02-01:
if I have the time, I'll release 4.0-beta0 this weekend, which will have the bug fixed

ricardoquesada, 2024-02-01:
just to clarify, the bug appears only on the NINA platform... if you are using "Arduino IDE" + regular ESP32 boards, then it shouldn't happen (I guess).

could you confirm the board that you are using ?

ricardoquesada, 2024-02-01:
got it. I'll double check it on the non-nina then

ricardoquesada, 2024-02-03:
I confirm that this was fixed on `develop` branch as well.

closed 2024-02-03 by ricardoquesada

microbmen, 2024-02-05:
Yes fixed in 4.0. Thx.