ricardoquesada/bluepad32 #51: Trying out Pico W support in develop

lucaspcamargo, 2023-11-23:
Hello,

I'm trying out the new Pico W support in the develop branch.
I setup [my project](https://github.com/lucaspcamargo/megapaladin/tree/v2) following the provided example as closely as possible. Without touching core 1, bluepad32 seems to start and print stuff to the console. Nice.

I have a few questions about the current state of things. I'm using `pico-sdk` master, as a point of reference.

I need to use core 1 for other tasks. Right now, the program seems to crash during `cyw43_arch_init`, if I initialize core 1 beforehand using `multicore_launch_core1` as soon as the program starts. If I try to start core 1 afterwards, then it crashes there. Do you know how this could be related, though? I really need to use the second core.

Moreover, can I use USB stdio after starting the main loop? From what I gather bluepad32 is supposed to have its own console, but I've read somewhere it does not work as of now. 

Finally, if I need a periodic task to check incoming commands from core 1, and react, can I register a timer like this?

```
// global:    
btstack_timer_source_t main_timer;

int main() {

    [...]

    // Initialize BP32
    uni_init(0, NULL);

    // add application process in run loop as a timer
    memset(&main_timer, 0x00, sizeof(btstack_timer_source_t));

    // setup my own callback
    main_timer.process = &core0_btstack_timer;
    btstack_run_loop_set_timer(&main_timer, CORE0_MAIN_TIMER_INTERVAL);
    btstack_run_loop_add_timer(&main_timer);

    // now run
    btstack_run_loop_execute();

    return 0;
}

// timer will call set_timer and add_timer again at the end of handler func
```

I've done that but the timer does not seem to run.

I also ran the SPP streamer from pico-examples to ensure my board is working as it should, it seemed ok.
In case the example is not expected to be working for now, for sure I can wait everything becoming stable. 

Thank you so much for this amazing project.

lucaspcamargo, 2023-11-23:
Well, I found out the issue. It was missing a call to flash_safe_execute_core_init, on both cores. That was causing problems in other parts of the stack that need to access flash. An assert triggered by BLE link key deletion gave me the clue.

[image]

ricardoquesada, 2023-11-23:
good to know that it is working ok.

closed 2023-11-23 by ricardoquesada