ricardoquesada/bluepad32 #52: [Bug]: DS4 Gyro and Accel

JuPrgn, 2023-11-29:
### What happened?

I am playing with Bluepad and this is working really well even with a DS4 clone that I was not able to pair with another library. Thank you very much for your work !

Using your arduino Controller example it looks like Gyro and Accel datas are reversed because the printed Accel is stable and depending on the orientation of the controller and Gyro datas change a lot with rapid movements. I tried with both controllers : clone (Data frog CFKJ-P02-DEC-16) and official (CUH-ZCT1).

### Bluepad32 Version

Other

### Bluepad32 version custom

Not sure about my version but in Arduino board manager : esp32_bluepad 3.10.2

### Bluepad32 Platform

Arduino IDE

### Platform version

Arduino IDE 2.2.1

### Controller

- clone (Data frog CFKJ-P02-DEC-16)
- official (CUH-ZCT1)

### ESP32 chip

ESP32

### ESP32 board

ESP32 DevKit WROOM32

### OS

Windows

### Relevant log output

```shell
gyro x:     0 y:     0 z:     0, accel x:  -223 y:  8300 z:  1393
gyro x:     0 y:     0 z:     0, accel x:  -240 y:  8301 z:  1393
gyro x:     0 y:     0 z:     0, accel x:  -220 y:  8257 z:  1439
gyro x:     0 y:     0 z:     0, accel x:  -205 y:  8269 z:  1399
gyro x:     0 y:     0 z:     0, accel x:  -197 y:  8288 z:  1432
gyro x:     0 y:     0 z:     0, accel x:  -193 y:  8295 z:  1415
gyro x:     0 y:     0 z:     0, accel x:  -211 y:  8279 z:  1431
gyro x:     0 y:     0 z:     0, accel x:  -183 y:  8307 z:  1399
gyro x:     0 y:     0 z:     0, accel x:  -251 y:  8285 z:  1401
gyro x:     0 y:     0 z:     0, accel x:  -183 y:  8271 z:  1445
gyro x:     0 y:     0 z:     0, accel x:  -222 y:  8245 z:  1462
gyro x: 66626 y:     0 z: 92354, accel x:   524 y:  9338 z:  1299
gyro x: 78146 y: 11456 z: 72194, accel x:  3031 y:  6743 z:  -449
gyro x:-42689 y:-19328 z:-84034, accel x:  2085 y:  7572 z: -1374
gyro x:-245191 y:-38017 z:-59137, accel x:  -419 y:  7459 z:  1207
gyro x:     0 y:     0 z: -7424, accel x:  -279 y:  8412 z:  1053
gyro x:     0 y:     0 z:     0, accel x:  -120 y:  8274 z:  1442
gyro x:     0 y:     0 z:     0, accel x:  -147 y:  8305 z:  1436
gyro x:     0 y:     0 z:     0, accel x:  -119 y:  8309 z:  1439
gyro x:     0 y:     0 z:     0, accel x:  -146 y:  8320 z:  1425
gyro x:     0 y:     0 z:     0, accel x:  -140 y:  8281 z:  1407
gyro x:     0 y:     0 z:     0, accel x:  -151 y:  8277 z:  1425
gyro x:     0 y:     0 z:     0, accel x:  -174 y:  8326 z:  1392
```

### Relevant sketch

```shell
Bluepad example : Controller
```

lovelytwo, 2024-02-10:
My guess is your not understanding what gyro and acceleration data is.

The dualshock 4 has rate gyros. The gyro will have base drift, but will also detect the rate of rotation, rate is different to position. The accelerometers on the other hand are mainly detecting gravity as it's applying a constant acceleration.

JuPrgn, 2024-02-10:
@lovelytwo Thanks for the explanation, I actually expected the gyro to give an angular position and not a rotation rate.

closed 2024-02-10 by JuPrgn