ricardoquesada/bluepad32 #54: DS4 setRumble

JuPrgn, 2023-12-02:
Using Bluepad example : Controller and changing setRumble values

I don't see differences when changing rumble force value, it seems this value may be hardcoded somewhere ? Changing rumble duration works well.

Bluepad32 version
esp32_bluepad 3.10.2

Bluepad32 Platform
Arduino IDE 2.2.1

```
void processGamepad(ControllerPtr ctl) {
    if (ctl->x()) {
        // Duration: 255 is ~2 seconds
        // force: intensity
        // Some gamepads like DS3, DS4, DualSense, Switch, Xbox One S support
        // rumble.
        // It is possible to set it by calling:
        static int rblForce = 0;
        rblForce+=10;
        if(rblForce > 200)
          rblForce = 0;
        ctl->setRumble(rblForce /* force */, 0xc0 /* duration */);
    }
```

ricardoquesada, 2023-12-03:
rumble API is kind of "hacky"... it is very difficult to map the different "rumble" options to one API.

Take into account that Genuine vs. Clone has a big difference.

Probably I'll implement an API  similar to Android for Rumble for v4.

JuPrgn, 2023-12-03:
Same problem as #53 this is relative to clone controller sorry for not having made the comparison with genuine part before

closed 2023-12-03 by JuPrgn