ricardoquesada/bluepad32 #58: [Bug]: Not reporting Y-axis data on ESP32 C3 from Xbox One Controller

darcoza, 2024-01-11:
### What happened?

I'm using an ESP 32 C3 with an Xbox One Controller, I've loaded up the example code from the Arduino IDE and it connects right up and reports all the button presses and the X-axis of both joysticks however it always reports 0 for the Y-axis of both joysticks no matter what. 

### Bluepad32 Version

3.8.3

### Bluepad32 version custom

Example: Using Git develop branch commit hash #xxxxxxx

### Bluepad32 Platform

Arduino IDE

### Platform version

E.g: Arduino IDE 2.1.0, or ESP-IDF v4.4.4, or CircuitPython 8.0.0, etc.

### Controller

issues

### ESP32 chip

ESP32

### ESP32 board

ESP32-C3 Super mini ex: https://www.aliexpress.us/item/3256805781327184.html?spm=a2g0o.order_list.order_list_main.29.b2811802zmLh3P&gatewayAdapt=glo2usa

### OS

Windows

### Relevant sketch

```shell
The default example sketch.
```

ricardoquesada, 2024-01-11:
are you using v3.8.3 ? please upgrade to latest Bluepad32 version which is 3.10.2

darcoza, 2024-01-11:
Yup I updated and everything started working... Thanks! Sorry for the false alarm!

closed 2024-01-12 by ricardoquesada