ricardoquesada/bluepad32 #59: [Bug]: Stadia Controller error: 'NINA_GPIO0' was not declared in this scope

jwilleke, 2024-01-12:
### What happened?

Would not compile.

### Bluepad32 Version

3.10.3

### Bluepad32 version custom

PACKAGES: 
 - framework-arduinoespressif32 @ 3.20011.230801 (2.0.11) 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5

Dependency Graph
|-- Bluepad32 @ 1.3.1

### Bluepad32 Platform

Other

### Platform version

Processing  PlatformIO 
using dfrobot_firebeetle2_esp32e (platform: espressif32; board: dfrobot_firebeetle2_esp32e; framework: arduino)

### Controller

issues 

### ESP32 chip

ESP32

### ESP32 board

We used: [FireBeetle 2 ESP32-E IoT Microcontroller (Supports Wi-Fi & Bluetooth)](https://www.dfrobot.com/product-2195.html)

### OS

macOS

### Relevant log output

```shell
Compiling .pio/build/dfrobot_firebeetle2_esp32e/lib3f3/Bluepad32/utility/spi_drv.cpp.o
src/main.cpp: In function 'void setup()':
src/main.cpp:50:15: error: 'onConnectedController' was not declared in this scope
   BP32.setup(&onConnectedController, &onDisconnectedController);
               ^~~~~~~~~~~~~~~~~~~~~
src/main.cpp:50:39: error: 'onDisconnectedController' was not declared in this scope
   BP32.setup(&onConnectedController, &onDisconnectedController);
                                       ^~~~~~~~~~~~~~~~~~~~~~~~
Compiling .pio/build/dfrobot_firebeetle2_esp32e/FrameworkArduino/Esp.cpp.o
Compiling .pio/build/dfrobot_firebeetle2_esp32e/FrameworkArduino/FirmwareMSC.cpp.o
Compiling .pio/build/dfrobot_firebeetle2_esp32e/FrameworkArduino/FunctionalInterrupt.cpp.o
Compiling .pio/build/dfrobot_firebeetle2_esp32e/FrameworkArduino/HWCDC.cpp.o
Compiling .pio/build/dfrobot_firebeetle2_esp32e/FrameworkArduino/HardwareSerial.cpp.o
Compiling .pio/build/dfrobot_firebeetle2_esp32e/FrameworkArduino/IPAddress.cpp.o
Compiling .pio/build/dfrobot_firebeetle2_esp32e/FrameworkArduino/IPv6Address.cpp.o
Compiling .pio/build/dfrobot_firebeetle2_esp32e/FrameworkArduino/MD5Builder.cpp.o
*** [.pio/build/dfrobot_firebeetle2_esp32e/src/main.cpp.o] Error 1
.pio/libdeps/dfrobot_firebeetle2_esp32e/Bluepad32/src/utility/spi_drv.cpp: In static member function 'static void SpiDrv::begin()':
.pio/libdeps/dfrobot_firebeetle2_esp32e/Bluepad32/src/utility/spi_drv.cpp:91:20: error: 'PINS_COUNT' was not declared in this scope
   if (SLAVERESET > PINS_COUNT) {
                    ^~~~~~~~~~
.pio/libdeps/dfrobot_firebeetle2_esp32e/Bluepad32/src/utility/spi_drv.cpp:91:20: note: suggested alternative: 'ICOUNT'
   if (SLAVERESET > PINS_COUNT) {
                    ^~~~~~~~~~
[... 245 lines cut ...]
// Arduino loop function. Runs in CPU 1
void loop() {
  // This call fetches all the controller info from the NINA (ESP32) module.
  // Call this function in your main loop.
  // The controllers' pointer (the ones received in the callbacks) gets updated
  // automatically.
  BP32.update();

  // It is safe to always do this before using the controller API.
  // This guarantees that the controller is valid and connected.
  for (int i = 0; i < BP32_MAX_CONTROLLERS; i++) {
    ControllerPtr myController = myControllers[i];

    if (myController && myController->isConnected()) {
      if (myController->isGamepad())
        processGamepad(myController);
      else if (myController->isMouse())
        processMouse(myController);
      else if (myController->isBalanceBoard())
        processBalanceBoard(myController);
    }
  }
  delay(150);
}
```

ricardoquesada, 2024-01-12:
you should not use "Bluepad32 @ 1.3.1" for ESP32 boards.

As the example says at the top:

```
/*
 * This example shows how to use the Controller API.
 *
 * Supported on boards with NINA W10x. In particular, these boards:
 *  - Arduino MKR WiFi 1010,
 *  - UNO WiFi Rev.2,
 *  - Nano RP2040 Connect,
 *  - Nano 33 IoT,
 *  - Arduino MKR Vidor 4000
 */
```

I wouldn't know how to use PlatformIO with Bluepad32, but definitely you should not use that library.

jwilleke, 2024-01-12:
First, Thanks for the prompt reply which seems so uncommon these days.

I guess I do not know what "Platform version" would imply?

Where might I find an example using the "Bluepad32" with an Stadia Controller using a ESP32 board
We used: [FireBeetle 2 ESP32-E IoT Microcontroller (Supports Wi-Fi & Bluetooth)](https://www.dfrobot.com/product-2195.html)

Thanks!

ricardoquesada, 2024-01-12:
Two options:

- Using regular Arduino IDE:  https://www.youtube.com/watch?v=0jnY-XXiD8Q
- Using ESP-IDF toolchain + Arduino Core:  https://www.youtube.com/watch?v=0jnY-XXiD8Q

closed 2024-02-19 by ricardoquesada