ricardoquesada/bluepad32 #62: [Bug]: `uni_hid_device_get_idx_for_instance()` reports index 0 for all devices

juan518munoz, 2024-01-17:
### What happened?

Running example program, inside `static void my_platform_on_controller_data(uni_hid_device_t* d, uni_controller_t* ctl)` I've set a variable to get the index of each connected gamepad, all report to zero.

```c
static void my_platform_on_controller_data(uni_hid_device_t* d, uni_controller_t* ctl)
{
	uni_gamepad_t *gp;
	uint8_t idx = uni_hid_device_get_idx_for_instance(d);
        ...
}
```

### Bluepad32 Version

I don't know

### Bluepad32 version custom

Example: Using Git develop branch, latest commit

### Bluepad32 Platform

Other

### Platform version

Pico W

### Controller

Joycons, Xbox Series controller

### ESP32 chip

ESP32

### OS

Linux

### Relevant log output

```shell
(0x20006644) dpad=0x00, x= -54, y=  79, rx=   0, ry=   0, brake=   0, throttle=   0, buttons=0x0000, misc=0x00, gyro= 176000, -39000, 120000 accel=    535,  -3105,  -2859, batte5
id: 0
(0x20006644) dpad=0x00, x= -54, y=  78, rx=   0, ry=   0, brake=   0, throttle=   0, buttons=0x0000, misc=0x00, gyro=-417000,-112000, -48000 accel=    611,  -3131,  -2622, batte5
id: 0
(0x200039bc) dpad=0x02, x= -15, y= -23, rx=  -4, ry=   7, brake=   0, throttle=   0, buttons=0x0000, misc=0x00, gyro=      0,      0,      0 accel=      0,      0,      0, batte0
id: 0
(0x200039bc) dpad=0x00, x= -15, y= -23, rx=  -4, ry=   7, brake=   0, throttle=   0, buttons=0x0000, misc=0x00, gyro=      0,      0,      0 accel=      0,      0,      0, batte0
id: 0
```

### Relevant sketch

```shell
static void my_platform_on_controller_data(uni_hid_device_t* d, uni_controller_t* ctl)
{
	uni_gamepad_t *gp;
	uint8_t idx = uni_hid_device_get_idx_for_instance(d);

	// Print device Id before dumping gamepad.
	logi("(%p) ", d);
	uni_controller_dump(ctl);

	switch (ctl->klass) {
	case UNI_CONTROLLER_CLASS_GAMEPAD:
		gp = &ctl->gamepad;
		fill_gamepad_report(idx, gp);
		idx_r.idx = idx;
		idx_r.report = report[idx];

		serialized = serialize_report(idx_r);
		if (multicore_fifo_push_timeout_us(serialized.low, 10)) {
			if (!multicore_fifo_push_timeout_us(serialized.high, 10)) {
				// if we fail to send the second part, clear poisoned FIFO
				multicore_fifo_drain();
			}
		}
		break;
	case UNI_CONTROLLER_CLASS_BALANCE_BOARD:
		// Do something
		uni_balance_board_dump(&ctl->balance_board);
		break;
	case UNI_CONTROLLER_CLASS_MOUSE:
		// Do something
		uni_mouse_dump(&ctl->mouse);
		break;
	case UNI_CONTROLLER_CLASS_KEYBOARD:
		// Do something
		uni_keyboard_dump(&ctl->keyboard);
		break;
	default:
		loge("Unsupported controller class: %d\n", ctl->klass);
		break;
	}
}
```

ricardoquesada, 2024-01-18:
your sketch is incomplete... how are you printing `idx` ?

ricardoquesada, 2024-01-18:
also, you should check return value... if it `idx < 0` it is an error

juan518munoz, 2024-01-18:
Sorry, I was working with two code editors side by side and passed the wrong one, it's now updated.

I never got a `-1` signaling an error.

ricardoquesada, 2024-01-18:
again, your sketch is incomplete.
I strongly suggest that you paste a complete sketch... something that compiles and reproduce the error. don't just copy & paste parts of it. 

for example, I don't know what  `get_connected_device_idx()`  is.

juan518munoz, 2024-01-18:
Sent the wrong snipped again, sorry, It was late and instead of focusing on what I was doing, I wanted to get it done. I've updated the sketch. Please let me know if something else is needed.

closed 2024-01-18 by ricardoquesada