ricardoquesada/bluepad32 #65: Support Arduino NANO ESP32 in Arduino IDE

ricardoquesada, 2024-02-02:
Bluepad32 + Arduino IDE currently does not work as expected using Arduino NANO ESP32.

https://store-usa.arduino.cc/products/nano-esp32

ricardoquesada, 2024-02-23:
fixed. confirmed by jjs357 user. It is working Ok now.

Freihaendich, 2024-02-29:
I tried to follow the guide, but it is not working for me right now. I have an Arduino Nano ESP32 and a PS4 Controller.

ricardoquesada, 2024-02-29:
@Freihaendich wait until v4.0-beta2 is released.... alternative use `develop` branch of https://github.com/ricardoquesada/esp-idf-arduino-bluepad32-template

ricardoquesada, 2024-03-03:
re-opening... using "print" works Ok, but not with `Serial.printf()`... WTF...  something is wrong with UART... not sure what.

ricardoquesada, 2024-03-03:
Same issue with the Lolin S3 Mini.
What these two boards have in common is that the share the UART and USB OTG ports... they have just one port.

And `Serial`, instead of being redirected to "right" port, it gets redirected somewhere else (?).

ricardoquesada, 2024-03-03:
Having "USB CDC on Boot : Enabled" fixes the issue...

But the bug seems  to be that "boards.txt" is not being used correctly, because `nano_nora` entry (the Arduino Nano ESP32) has `nano_nora.build.cdc_on_boot=1`.

And that would explain why the Arduino Nano ESP32 gets recognized as a different board

ricardoquesada, 2024-03-04:
Something seems to be happening when BTstack is initialized. I'm not calling `btstack_stdio_init()`... so something else, not clear what yet

ricardoquesada, 2024-03-10:
The solution for "Arduino Nano ESP32" is to `Tools` -> `USB Mode` -> `Debug Mode (Hardware CDC)`

[image]

ricardoquesada, 2024-03-10:
For other boards, like "Lolin S3 Mini" do:

- Make sure the 'Lolin S3 Mini' is selected
- `Tools` -> `USB CDC On Boot` -> `Enabled`
- and then compile and flash the sketch again

[image]

closed 2024-03-10 by ricardoquesada

craig-am, 2024-03-18:
I am trying to run the example on an Adafruit Huzzah32 and I get this compiler error about PINS_COUNT not being define anywhere
\Users\ck\Documents\Arduino\libraries\Bluepad32_for_NINA-W10_boards\src\utility\spi_drv.cpp" -o "C:\Users\ck\AppData\Local\Temp\VMBuilds\Bluepad32_ESP32_Gamecontroller_Connection_Test\esp32_featheresp32\Debug\Bluepad32_for_NINA-W10_boards\utility\spi_drv.cpp.o"
spi_drv.cpp: In static member function static void SpiDrv::begin()
 
spi_drv.cpp: 91:20: error: 'PINS_COUNT' was not declared in this scope
   if (SLAVERESET > PINS_COUNT) {
   ^~~~~~~~~~