ricardoquesada/bluepad32 #66: BLE Controller for the ESP32S3

BrainBoardz, 2024-02-07:
Hi There: 

I'm trying to get a wireless gaming controller to work with the ESP32S3 via Bluepad32. It's turning out to be a PITA. The units listed with BLE support are either out of production or are very expensive (a handful of XBox models). I'm wondering if anyone can recommend a model that will work with the S3 that is readily available (e.g., on Amazon). I'm looking for a basic controller, so a lack of fancy features  an issue. Thanks for any assistance.

ricardoquesada, 2024-02-07:
try to get a used Stadia contoller from eBay.

https://bluepad32.readthedocs.io/en/latest/supported_gamepads/

BrainBoardz, 2024-02-07:
Thanks for the assistance. Unfortunately buying a used  Stadia controller  isn't really a viable long-term solution in my scenario, as I am looking for something that is readily available (non-used) from retail sources (Amazon etc.). It's really unfortunate that the S3 does not support Bluetooth Classic!

ricardoquesada, 2024-02-07:
try this one... not the best quality controller, but works and it is BLE:

https://www.aliexpress.us/item/2251832602044731.html?spm=a2g0o.order_list.order_list_main.5.51801802pNXIqZ&gatewayAdapt=glo2usa

ricardoquesada, 2024-02-07:
only mode that works is "home + x"

See row #97: 

https://docs.google.com/spreadsheets/d/10Ev5ycCAJGgVBQ9sc6y-U0cPcbdUCy1aqvxLBfc50_Y/edit?usp=sharing

closed 2024-02-07 by ricardoquesada

BrainBoardz, 2024-02-07:
Hi Ricardo :

Wow! That's really helpful. I'll check out that controller.  So any
controller with a Home+X designation should work? So even the GameSir TS3
potentially?

*Jeremy Littler*

*LEAD Emerging Technologies*

The Creative School

Toronto Metropolitan University

416-979-5000 ext. 552222

RCC 353

ricardoquesada, 2024-02-07:
sorry, row #93... not row#97.... the "Terios T3 (Generic clone)" should work with "Home+X"

BrainBoardz, 2024-02-29:
Hi Ricardo:

The Terios T3 clone worked perfectly! Thanks for all the help with that.
BTW I did try a Gamesir T3S and it did not work. Is this due to the
library not supporting that particular device with the ESP32S3 or you don't
have one to test
or it just isn't going to ever be compatible?  I'd happily send you mine if
that would help in getting it supported.The TS3 is a much nicer quality
controller and it would be nice if there was an option other than the T3
that is higher quality. Still,
I am very happy about the T3 working either way. That is awesome.

*Jeremy Littler*

*LEAD Emerging Technologies*

The Creative School

Toronto Metropolitan University

416-979-5000 ext. 552222

RCC 353

ricardoquesada, 2024-02-29:
Is the terios t3s BLE or Classic?
If it is classic it won't work on the ESP32-S3

See:

https://bluepad32.readthedocs.io/en/latest/supported_gamepads/#bluetooth-gamepads-and-their-protocol

BrainBoardz, 2024-03-01:
Hi Ricardo:

I'll do some digging and find out. It was actually this unit:

https://www.amazon.ca/dp/B09ZQWR5RD?ref=ppx_yo2ov_dt_b_product_details&th=1

It looked promising but it would not connect.

*Jeremy Littler*

*LEAD Emerging Technologies*

The Creative School

Toronto Metropolitan University

416-979-5000 ext. 552222

RCC 353

i R

ricardoquesada, 2024-03-01:
GameSir T3S should connect. See rows 101-103 of https://docs.google.com/spreadsheets/d/10Ev5ycCAJGgVBQ9sc6y-U0cPcbdUCy1aqvxLBfc50_Y/edit#gid=0

Use any of these 3 modes:
iOS (Home+B)
PC (Home+X)
Switch (Home+Y)

ricardoquesada, 2024-03-01:
My bad. Gamesir is BT classic only. So it won't work with ESP32-S3

BrainBoardz, 2024-03-01:
Hi Ricardo:

That's unfortunate, but also good to know. It looks like the Terios T3 is
the only option then. Good to know that it is readily available.  I did see
this one as well.  The specs posted do indicate
BTLE 4.0 and Android modes are supported. Am I correct that BluePad32 would
not necessarily work with this unit either as the BD Addr needs to be in
your library to be supported?

https://www.amazon.ca/GameSir-Controller-Wireless-Configurable-Buttons/dp/B07HG51ZYK?th=1

Either way it was a hoot seeing the joystick data from the T3 being
displayed.  The pairing is so seamless. You have created a really essential
library for the S3!

*Jeremy Littler*

*LEAD Emerging Technologies*

The Creative School

Toronto Metropolitan University

416-979-5000 ext. 552222

RCC 353

ricardoquesada, 2024-03-03:
The GameSir T4 might work without changes to Bluepad32... or not. Depends on how GameSir T4 identifies itself.

BrainBoardz, 2024-03-04:
Hi Richardo:

Thanks.  I will order one and see if it works.

bye for now,

*Jeremy Littler*

*LEAD Emerging Technologies*

The Creative School

Toronto Metropolitan University

416-979-5000 ext. 552222

RCC 353

BrainBoardz, 2024-03-06:
Hi Ricardo:

Wow, it is a very narrow list (i.e., one off-the-shelf  unit afaik). I had
high hopes, but the GameSir T4 Pro also did not work with BluePad on the
S3.  If you do come across any controller other than the Terios T3 that
does work with the S3 I would love
to hear about it.

bye for now,

*Jeremy Littler*

*LEAD Emerging Technologies*

The Creative School

Toronto Metropolitan University

416-979-5000 ext. 552222

RCC 353