ricardoquesada/bluepad32 #68: Support for "TikTok Ring Remote Controller"

M4rkoV2, 2024-02-19:
I have [THIS](https://www.aliexpress.com/item/1005005783027995.html?spm=a2g0o.productlist.main.1.19275fe09JOm2M&algo_pvid=df554b4f-dcb3-4f6e-b1e6-a0731070a1d2&algo_exp_id=df554b4f-dcb3-4f6e-b1e6-a0731070a1d2-0&pdp_npi=4%40dis%21EUR%2114.07%214.69%21%21%21106.73%2135.54%21%402101e9ec17083299935043854ef16f%2112000034340327563%21sea%21SI%21726186692%21&curPageLogUid=oTbAEbainb5w&utparam-url=scene%3Asearch%7Cquery_from%3A) bluetooth remote controller with 5 buttons that I would like connect with bluepad32.

I was able to connect it fast, but output is same (idx=0) for all keys. Can support for this controller be added?

```
Firmware: Bluepad32 for Arduino v4.0-beta1
BD Addr: AC: B:FB:25:9F:7E
CALLBACK: Controller is connected, index=0
Controller model: Keyboard, VID=0x05ac, PID=0x022c
idx=0
idx=0
idx=0
idx=0
```

ricardoquesada, 2024-02-19:
For controllers that are not supported and that I don't have in my invetory is:

- you send me the cost of the controller + shipping
- I purchase it myself
- I'll do my best to add support it, but I don't guarantee anything

(details here:  https://bit.ly/bluepad32-help )

For this particular case, the cost of the controller + shipping is $6 (US dollars).
If you are Ok with it, let me know and I'll send you my paypal account.

M4rkoV2, 2024-02-21:
I'm ok with it. Send me paypal. 

Is it possible to Serial Print raw info of the button?

ricardoquesada, 2024-02-21:
Paypal: https://www.paypal.com/paypalme/RicardoQuesada

From Arduino IDE, you cannot print the raw packet. 

But if you use "Arduino + ESP-IDF" you can add a `printf_hexdump(report, report_len);` somewhere here to see the raw data:
https://github.com/ricardoquesada/bluepad32/blob/main/src/components/bluepad32/parser/uni_hid_parser.c#L14

ricardoquesada, 2024-03-02:
What's interesting is that this device reports itself as a keyboard... wondering if Bluepad32 should honor that and treat it as keyboard, or as a gamepad.

Since it has a sort of  Dpad, I'd guess that Bluepad32 should override that and treat it as gamepad.

ricardoquesada, 2024-03-02:
Events generated by

array up:
```
Keyboard: Unsupported page: 0x000d, usage: 0x0042, value=0x1
Keyboard: Unsupported page: 0x000d, usage: 0x0032, value=0x1
Keyboard: Unsupported page: 0x000d, usage: 0x0051, value=0x6
Keyboard: Unsupported page: 0x0001, usage: 0x0030, value=0x370
Keyboard: Unsupported page: 0x0001, usage: 0x0031, value=0x3f4
Keyboard: Unsupported page: 0x000d, usage: 0x0054, value=0x1
Keyboard: Unsupported page: 0x000d, usage: 0x0042, value=0x1
Keyboard: Unsupported page: 0x000d, usage: 0x0032, value=0x1
Keyboard: Unsupported page: 0x000d, usage: 0x0051, value=0x6
Keyboard: Unsupported page: 0x0001, usage: 0x0030, value=0x370
Keyboard: Unsupported page: 0x0001, usage: 0x0031, value=0x44c
Keyboard: Unsupported page: 0x000d, usage: 0x0054, value=0x1
Keyboard: Unsupported page: 0x000d, usage: 0x0042, value=0x1
Keyboard: Unsupported page: 0x000d, usage: 0x0032, value=0x1
Keyboard: Unsupported page: 0x000d, usage: 0x0051, value=0x6
Keyboard: Unsupported page: 0x0001, usage: 0x0030, value=0x370
Keyboard: Unsupported page: 0x0001, usage: 0x0031, value=0x578
Keyboard: Unsupported page: 0x000d, usage: 0x0054, value=0x1
Keyboard: Unsupported page: 0x000d, usage: 0x0042, value=0x1
Keyboard: Unsupported page: 0x000d, usage: 0x0032, value=0x1
Keyboard: Unsupported page: 0x000d, usage: 0x0051, value=0x6
Keyboard: Unsupported page: 0x0001, usage: 0x0030, value=0x370
Keyboard: Unsupported page: 0x0001, usage: 0x0031, value=0x6a4
Keyboard: Unsupported page: 0x000d, usage: 0x0054, value=0x1
Keyboard: Unsupported page: 0x000d, usage: 0x0042, value=0x1
Keyboard: Unsupported page: 0x000d, usage: 0x0032, value=0x1
Keyboard: Unsupported page: 0x000d, usage: 0x0051, value=0x6
Keyboard: Unsupported page: 0x0001, usage: 0x0030, value=0x370
Keyboard: Unsupported page: 0x0001, usage: 0x0031, value=0x7d0
Keyboard: Unsupported page: 0x000d, usage: 0x0054, value=0x1
Keyboard: Unsupported page: 0x000d, usage: 0x0042, value=0x1
Keyboard: Unsupported page: 0x000d, usage: 0x0032, value=0x1
Keyboard: Unsupported page: 0x000d, usage: 0x0051, value=0x6
Keyboard: Unsupported page: 0x0001, usage: 0x0030, value=0x370
Keyboard: Unsupported page: 0x0001, usage: 0x0031, value=0x8fc
Keyboard: Unsupported page: 0x000d, usage: 0x0054, value=0x1
Keyboard: Unsupported page: 0x000d, usage: 0x0042, value=0x1
Keyboard: Unsupported page: 0x000d, usage: 0x0032, value=0x1
Keyboard: Unsupported page: 0x000d, usage: 0x0051, value=0x6
Keyboard: Unsupported page: 0x0001, usage: 0x0030, value=0x370
Keyboard: Unsupported page: 0x0001, usage: 0x0031, value=0xa28
Keyboard: Unsupported page: 0x000d, usage: 0x0054, value=0x1
Keyboard: Unsupported page: 0x000d, usage: 0x0042, value=0x1
Keyboard: Unsupported page: 0x000d, usage: 0x0032, value=0x1
Keyboard: Unsupported page: 0x000d, usage: 0x0051, value=0x6
Keyboard: Unsupported page: 0x0001, usage: 0x0030, value=0x370
Keyboard: Unsupported page: 0x0001, usage: 0x0031, value=0xb54
Keyboard: Unsupported page: 0x000d, usage: 0x0054, value=0x1
Keyboard: Unsupported page: 0x000d, usage: 0x0042, value=0x1
Keyboard: Unsupported page: 0x000d, usage: 0x0032, value=0x1
Keyboard: Unsupported page: 0x000d, usage: 0x0051, value=0x6
Keyboard: Unsupported page: 0x0001, usage: 0x0030, value=0x370
Keyboard: Unsupported page: 0x0001, usage: 0x0031, value=0xc80
Keyboard: Unsupported page: 0x000d, usage: 0x0054, value=0x1
Keyboard: Unsupported page: 0x000d, usage: 0x0042, value=0x0
Keyboard: Unsupported page: 0x000d, usage: 0x0032, value=0x0
Keyboard: Unsupported page: 0x000d, usage: 0x0051, value=0x6
Keyboard: Unsupported page: 0x0001, usage: 0x0030, value=0x370
Keyboard: Unsupported page: 0x0001, usage: 0x0031, value=0xdac
Keyboard: Unsupported page: 0x000d, usage: 0x0054, value=0x0
```

Arrow down:
```
Keyboard: Unsupported page: 0x000d, usage: 0x0042, value=0x1
Keyboard: Unsupported page: 0x000d, usage: 0x0032, value=0x1
[... 225 lines cut ...]
Keyboard: page:12, usage:234, value:0
Keyboard: page:12, usage:233, value:0
Keyboard: page:12, usage:48, value:1
Keyboard: page:12, usage:430, value:0
Keyboard: page:12, usage:547, value:0
Keyboard: page:12, usage:234, value:0
Keyboard: page:12, usage:233, value:0
Keyboard: page:12, usage:48, value:0
Keyboard: page:12, usage:430, value:0
Keyboard: page:12, usage:547, value:0
```

long center:
```
Keyboard: page:12, usage:234, value:1
Keyboard: page:12, usage:233, value:0
Keyboard: page:12, usage:48, value:0
Keyboard: page:12, usage:430, value:0
Keyboard: page:12, usage:547, value:0
Keyboard: page:12, usage:234, value:0
Keyboard: page:12, usage:233, value:0
Keyboard: page:12, usage:48, value:0
Keyboard: page:12, usage:430, value:0
Keyboard: page:12, usage:547, value:0
```

ricardoquesada, 2024-03-03:
Fixed.

The JX-05 controller is not a good controller to control generic devices.
It injects multiple packets per button to simulate mouse events on Android (iOS too I guess).

So, it does NOT translate ok as a regular d-pad.

That being said, since the JX-05 impersonates itself as a keyboard, I'm converting the "mouse" packets into cursor key events.

So:

- left button: left arrow key
- right button: right arrow key
- up button: up arrow key
- down button: down arrow key
- center button: space key

also, since it injects multiple packets per button, latency is not that great.

In any case, support for it is complete, even though I wouldn't recommend using this controller... works for very basic use cases, and that's it.
-

closed 2024-03-03 by ricardoquesada

M4rkoV2, 2024-03-17:
Thank you. Do I need to update Arduino library for this to work for me? because i did it and I still only get "idx=0" event

ricardoquesada, 2024-03-17:
@M4rkoV2 you should be able to see the new events with latest arduino... but remember it is recognized as a keyboard.

Your code should be here:
https://github.com/ricardoquesada/esp-idf-arduino-bluepad32-template/blob/main/main/sketch.cpp#L100

I guess I need to add code that shows how to parse it when it is a keyboard

danimihalca, 2025-10-06:
Hi @M4rkoV2, did you manage to transform it into a proper dpad? I have something similar which has a ST17H66B chip, and was curious if it's the same and how did you suceed flashing it. Thanks!