ricardoquesada/bluepad32 #70: Unable to connect to esp32 in AP mode

antonix883, 2024-02-19:
### What happened?

Today I noticed that using the bluepad32 library boards, the wifi in Access Point mode does not allow connection to any device. With the same code, compiling using standard esp32 boards, there are no problems. Has anyone else encountered the same issue? The controller however connects without problems and works. The board also connects in Station mode.

### Bluepad32 Version

3.10.3

### Bluepad32 version custom

Example: Using Git develop branch commit hash #xxxxxxx

### Bluepad32 Platform

Arduino IDE

### Platform version

Arduino IDE 1.8.4

### Controller

DS4 AceGamer

### ESP32 chip

ESP32

### ESP32 board

ESP32-Wroom 32

### OS

Windows

ricardoquesada, 2024-02-24:
please attach your sketch. In theory BT and WiFi should co-exists

antonix883, 2024-02-24:
I simply used the WiFiAccessPoint.ino example.
Using the expressif esp32 board works fine. Using the same bluepad32 board the connection is rejected. The wifi works because in station mode the board regularly connects to the home network using bluepad32. So it must be a problem that only affects the access point mode.

ricardoquesada, 2024-02-24:
Would you mind attaching the `WIFIAccessPoint.ino` example ?

antonix883, 2024-02-24:
Surely. I thought it was enough to get it from the library, like I did.
https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WiFiAccessPoint/WiFiAccessPoint.ino

jsolderitsch, 2024-02-24:
I can confirm this issue. I am attaching a sketch here:
```
/*
  Configuration AP example
  The example shows a very simple Configuration Access Point.

  created in August 2019
  by Juraj Andrassy https://github.com/jandrassy

*/
#include <WiFi.h>

static void decodeURL(char* url) {

    // Replace (in place) any %<hex><hex> sequences with the appropriate 8-bit character.
    // Source: https://www.programcreek.com/cpp/?CodeExample=url+decode (Example 6)
    
    char* cursor = url;
    while (*cursor) {
        if ((cursor[0] == '%') && cursor[1] && isxdigit(cursor[1]) && cursor[2] && isxdigit(cursor[2])) {
            // We saw a % followed by 2 hex digits, so we copy the literal hex value into the URL, then advance the cursor past it:
            char hex[3];
            hex[0] = cursor[1];
            hex[1] = cursor[2];
            hex[2] = '\0';
            *url++ = (char)strtol(hex, NULL, 16);
            cursor += 3;
        }
        else {
            // Common case: This is a normal character or a bogus % expression, so just copy it
            *url++ = *cursor++;
        }
    }
    *url++ = 0; // null terminate in case last character was decoded      
}

void configAP() {

  WiFiServer configWebServer(80);

  WiFi.mode(WIFI_AP_STA);  // starts the default AP (factory default or setup as persistent)

  Serial.print("Connect your computer to the WiFi network ");
  Serial.print("to SSID of your ESP32");  // no getter for SoftAP SSID
  Serial.println();
  IPAddress ip = WiFi.softAPIP();
  Serial.print("and enter http://");
  Serial.print(ip);
  Serial.println(" in a Web browser");

  configWebServer.begin();

  while (true) {

    WiFiClient client = configWebServer.available();
    if (client) {
      char line[64];
      int l = client.readBytesUntil('\n', line, sizeof(line));
      line[l] = 0;
      Serial.println("Inside of Client. 1");
      Serial.println(line);
      client.find((char*)"\r\n\r\n");
      if (strncmp_P(line, PSTR("POST"), strlen("POST")) == 0) {
        l = client.readBytes(line, sizeof(line));
        line[l] = 0;
        Serial.println("Inside of Client. 2");
        Serial.println(line);
        decodeURL(line);
        Serial.println(line);

        // parse the parameters sent by the html form
        const char* delims = "=&";
        strtok(line, delims);
        Serial.println("Inside of Client. 3");
        Serial.println(line);
        const char* ssid = strtok(NULL, delims);
        strtok(NULL, delims);
        Serial.println("Inside of Client. 4");
        Serial.println(ssid);
        const char* pass = strtok(NULL, delims);
        // const char* pass = "MontgomeryBurns!";
        Serial.println("Inside of Client. 5");
        Serial.println(pass);

        // send a response before attemting to connect to the WiFi network
        // because it will reset the SoftAP and disconnect the client station
        client.println(F("HTTP/1.1 200 OK"));
        client.println(F("Connection: close"));
        client.println(F("Refresh: 10"));  // send a request after 10 seconds
        client.println();
        client.println(F("<html><body><h3>Configuration AP</h3><br>connecting...</body></html>"));
        client.stop();

        Serial.println();
        Serial.print("Attempting to connect to WPA SSID: ");
        Serial.println(ssid);
        WiFi.persistent(true);
        WiFi.setAutoConnect(true);
        WiFi.begin(ssid, pass);
        WiFi.waitForConnectResult(10000);

        // configuration continues with the next request

      } else {

        client.println(F("HTTP/1.1 200 OK"));
        client.println(F("Connection: close"));
        client.println();
        client.println(F("<html><body><h3>Configuration AP</h3><br>"));

        int status = WiFi.status();
        if (status == WL_CONNECTED) {
          client.println(F("Connection successful. Ending AP."));
        } else {
          client.println(F("<form action='/' method='POST'>WiFi connection failed. Enter valid parameters, please.<br><br>"));
          client.println(F("SSID:<br><input type='text' name='i'><br>"));
          client.println(F("Password:<br><input type='password' name='p'><br><br>"));
          client.println(F("<input type='submit' value='Submit'></form>"));
        }
        client.println(F("</body></html>"));
        client.stop();

        if (status == WL_CONNECTED) {
          delay(1000);  // to let the SDK finish the communication
          Serial.println("Connection successful. Ending AP.");
          configWebServer.stop();
          WiFi.mode(WIFI_STA);
          return;
        }
      }
    }
  }
}

void setup() {

  Serial.begin(115200);
  delay(500);

  // WiFi.begin(); // use SSID and password stored by SDK. commented out to test the Configuration AP

  // waiting for connection to remembered  Wifi network
  Serial.println("Waiting for connection to WiFi");
  WiFi.waitForConnectResult(10000); // 10 seconds?

  if (WiFi.status() != WL_CONNECTED) {
    Serial.println();
    Serial.println("Could not connect to WiFi. Starting configuration AP...");
    configAP();
  } else {
    Serial.println("WiFi connected");
    Serial.println(WiFi.localIP());
  }
}

void loop() {
}

```

jsolderitsch, 2024-02-24:
This sketch works with my ESP32 when I don't use the bluepad32 board type

antonix883, 2024-02-24:
The problem only occurs in access point mode, whether using WiFi.mode(WIFI_AP_STA) or WiFi.mode(WIFI_AP). Using WiFi.mode(WIFI_STA) connects to network without problems. I tried from an Android smartphone and a Windows PC.

jsolderitsch, 2024-02-24:
And in my use case, the WIFI_STA mode is not applicable as I am trying to gather a local SSID and password via a local AP. Then I could connect in WIFI_STA mode after.

SirFridge, 2024-02-24:
I have the exact same problem with my code. I want to drive a little tank around that has a camera built in that I view via access point. I can connect it to a network but the camera will not load in browser.

ricardoquesada, 2024-02-25:
interesting... perhaps I forgot to enable a feature in the `menuconfig`

jsolderitsch, 2024-02-25:
> Surely. I thought it was enough to get it from the library, like I did. https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WiFiAccessPoint/WiFiAccessPoint.ino

I can confirm that this example fails for me also when the board type is one with bluepad32 enabled.

Works find with an ordinary ESP32 board type.

ricardoquesada, 2024-02-26:
Watchdog is enabled in Bluepad32 version... so your code should never have a `while(true)` without yielding the CPU, including `loop()`.

E.g: from  https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WiFiAccessPoint/WiFiAccessPoint.ino, you should add in line  98 something like

```c++
    // close the connection:
    client.stop();
    Serial.println("Client Disconnected.");
  }
  vTaskDelay(1);   // <-- ADD THIS LIE
}
```

reopen if that doesn't work

jsolderitsch, 2024-02-26:
This does not work for me. So re-opening.

jsolderitsch, 2024-02-26:
My sample code does NOT even have any contents in ```loop{}```

jsolderitsch, 2024-02-26:
There is a ```while(1)``` loop in ```setup{}``` and I added a ```vTaskDelay(1)``` there and that didn't help either.

jsolderitsch, 2024-02-26:
[image]
This alert pops up again and again without the access point being joined

ricardoquesada, 2024-02-28:
I confirm that with the attached sketch from @jsolderitsch using the template project in `develop` branch  ( https://github.com/ricardoquesada/esp-idf-arduino-bluepad32-template ) it works Ok.

(just rename `Serial` with `Console` if you want to try it).

After the connection, I see these messages on the console, that confirms that the connection was Ok:

```
bp32> I (103046) wifi:new:<1,0>, old:<1,0>, ap:<1,1>, sta:<255,255>, prof:1
I (103049) wifi:station: bc:d0:74:51:93:a2 join, AID=1, bgn, 20
I (106147) wifi:new:<1,0>, old:<1,0>, ap:<1,1>, sta:<255,255>, prof:1
I (106150) wifi:station: bc:d0:74:51:93:a2 join, AID=1, bgn, 20
I (118612) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.2
I (126222) wifi:<ba-add>idx:2 (ifx:1, bc:d0:74:51:93:a2), tid:0, ssn:4, winSize:64
I (143792) wifi:<ba-del>idx
I (143800) wifi:<ba-add>idx:2 (ifx:1, bc:d0:74:51:93:a2), tid:0, ssn:363, winSize:64
I (143876) wifi:<ba-del>idx
I (143881) wifi:<ba-add>idx:2 (ifx:1, bc:d0:74:51:93:a2), tid:0, ssn:375, winSize:64
I (143899) wifi:<ba-del>idx
I (143963) wifi:<ba-add>idx:2 (ifx:1, bc:d0:74:51:93:a2), tid:0, ssn:396, winSize:64
```

I don't know why it doesn't work from the Arduino IDE... I'll look at it next week

jsolderitsch, 2024-02-28:
Thanks for checking. I will check this out real soon. I would like the Arduino IDE to be happy with this as well as my class is planning on using an extension of this example with gamepads in an assignment in a couple of weeks. 
Which example do you mean? The first one I included or the modified version of the original issue poster's sketch?
And do I need a ```vTaskDelay()``` in the ```while (client.connected())```  loop too?

ricardoquesada, 2024-02-28:
@jsolderitsch this one: https://github.com/ricardoquesada/bluepad32/issues/70#issuecomment-1964381650

jsolderitsch, 2024-02-28:
> @jsolderitsch this one: [#70 (comment)](https://github.com/ricardoquesada/bluepad32/issues/70#issuecomment-1964381650)

Great, will use this one.

jsolderitsch, 2024-03-05:
I tested the example out with the latest pull of the develop branch and I am sorry to say it still doesn't work. In the console I see:
```
ere We Go
Configuring access point...
I (1676) wifi:wifi driver task: 3ffea058, prio:23, stack:6656, core=0
I (1701) wifi:wifi firmware version: c7dbced
I (1703) wifi:wifi certification version: v7.0
I (1704) wifi:config NVS flash: enabled
I (1705) wifi:config nano formating: disabled
I (1717) wifi:Init data frame dynamic rx buffer num: 32
I (1718) wifi:Init management frame dynamic rx buffer num: 32
I (1719) wifi:Init management short buffer num: 32
I (1731) wifi:Init dynamic tx buffer num: 32bp32>
I (1732) wifi:Init static rx buffer size: 1600
I (1740) wifi:Init static rx buffer num: 4
I (1741) wifi:Init dynamic rx buffer num: 32
I (1744) wifi_init: rx ba win: 6
I (1745) wifi_init: tcpip mbox: 32
I (1756) wifi_init: udp mbox: 6
I (1757) wifi_init: tcp mbox: 6
I (1758) wifi_init: tcp tx win: 5744
I (1769) wifi_init: tcp rx win: 5744
I (1769) wifi_init: tcp mss: 1440
I (1770) wifi_init: WiFi IRAM OP enabled
I (1781) wifi_init: WiFi RX IRAM OP enabled
I (1786) wifi:mode : softAP (0c:dc:7e:cc:4a:95)
I (1792) wifi:Total power save buffer number: 16
I (1793) wifi:Init max length of beacon: 752/752
I (1795) wifi:Init max length of beacon: 752/752
AP IP address: ��A?����A?��8:�?
Server started
I (31333) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<255,255>, prof:1
I (31337) wifi:station: f8:ff:c2:15:2c:a2 join, AID=1, bgn, 20
I (35443) wifi:station: f8:ff:c2:15:2c:a2 leave, AID = 1, bss_flags is 658530, bss:0x3ffee6b0
I (35446) wifi:new:<1,0>, old:<1,0>, ap:<1,1>, sta:<255,255>, prof:1
```
and I get the same repeated dialog to choose the named access point ESP32Bluepad
[image]

jsolderitsch, 2024-03-05:
Code for sketch.cpp:
```
/*
  WiFiAccessPoint.ino creates a WiFi access point and provides a web server on it.

  Steps:
  1. Connect to the access point "yourAp"
  2. Point your web browser to http://192.168.4.1/H to turn the LED on or http://192.168.4.1/L to turn it off
     OR
     Run raw TCP "GET /H" and "GET /L" on PuTTY terminal with 192.168.4.1 as IP address and 80 as port

  Created for arduino-esp32 on 04 July, 2018
  by Elochukwu Ifediora (fedy0)
*/

#include <WiFi.h>
#include <WiFiClient.h>
#include <WiFiAP.h>
#include <Bluepad32.h>

#ifndef LED_BUILTIN
#define LED_BUILTIN 26   // Set the GPIO pin where you connected your test LED or comment this line out if your dev board has a built-in LED
#endif

// Set these to your desired credentials.
const char *ssid = "ESP32Bluepad";
const char *password = "password";

WiFiServer server(80);

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
  Console.println("Here We Go");

  Console.println("Configuring access point...");

  // You can remove the password parameter if you want the AP to be open.
  // a valid password must have more than 7 characters
  if (!WiFi.softAP(ssid, password)) {
    // log_e("Soft AP creation failed.");
    while(1) {
       vTaskDelay(1);
    };
  }
  IPAddress myIP = WiFi.softAPIP();
  Console.printf("AP IP address: %s\n",myIP);

  Console.println("Server started");
}

void loop() {
  WiFiClient client = server.accept();   // listen for incoming clients

  if (client) {                             // if you get a client,
    Console.println("New Client.");           // print a message out the serial port
    String currentLine = "";                // make a String to hold incoming data from the client
    while (client.connected()) {            // loop while the client's connected
      if (client.available()) {             // if there's bytes to read from the client,
        char c = client.read();             // read a byte, then
        // Serial.write(c);                    // print it out the serial monitor
        if (c == '\n') {                    // if the byte is a newline character

          // if the current line is blank, you got two newline characters in a row.
          // that's the end of the client HTTP request, so send a response:
          if (currentLine.length() == 0) {
            // HTTP headers always start with a response code (e.g. HTTP/1.1 200 OK)
            // and a content-type so the client knows what's coming, then a blank line:
            client.println("HTTP/1.1 200 OK");
            client.println("Content-type:text/html");
            client.println();

            // the content of the HTTP response follows the header:
            client.print("Click <a href=\"/H\">here</a> to turn ON the LED.<br>");
            client.print("Click <a href=\"/L\">here</a> to turn OFF the LED.<br>");

            // The HTTP response ends with another blank line:
            client.println();
            // break out of the while loop:
            break;
          } else {    // if you got a newline, then clear currentLine:
            currentLine = "";
          }
        } else if (c != '\r') {  // if you got anything else but a carriage return character,
          currentLine += c;      // add it to the end of the currentLine
        }

        // Check to see if the client request was "GET /H" or "GET /L":
        if (currentLine.endsWith("GET /H")) {
          digitalWrite(LED_BUILTIN, HIGH);               // GET /H turns the LED on
        }
        if (currentLine.endsWith("GET /L")) {
          digitalWrite(LED_BUILTIN, LOW);                // GET /L turns the LED off
        }
      }
    }
    // close the connection:
    client.stop();
    Console.println("Client Disconnected.");
  }
  vTaskDelay(1);
}
```

jsolderitsch, 2024-03-05:
This is with an AdaFruit ESP32 Feather. I had to use menuconfig to set a large partition size.

jsolderitsch, 2024-03-05:
I reused an existing my_project folder rather than starting from scratch. I replaced the sketch.cpp with the contents I pasted above and refreshed git with the command sequence:
```
git pull
git checkout develop
git submodule update --init
``` 
and then did the idf.py commands like I have always been doing.
That leads to the non-working behavior I see now.
Must I begin with a fresh my_project clone?

jsolderitsch, 2024-03-05:
Further up in the output from the idf.py flash command I see:
```
I (0) cpu_start: App cpu up.
I (596) cpu_start: Pro cpu start user code
I (596) cpu_start: cpu freq: 160000000
I (596) cpu_start: Application information:
I (600) cpu_start: Project name:     bluepad32_arduino_app_template
I (607) cpu_start: App version:      4.0-beta1-21-g3b0d794-dirty
I (614) cpu_start: Compile time:     Mar  5 2024 11:57:26
I (620) cpu_start: ELF file SHA256:  3a117d5e7cacf847...
I (626) cpu_start: ESP-IDF:          v4.4.6-198-g5c2142eedb
I (632) cpu_start: Min chip rev:     v0.0
I (637) cpu_start: Max chip rev:     v3.99
I (642) cpu_start: Chip rev:         v3.0
I (647) heap_init: Initializing. RAM available for dynamic allocation:
I (654) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (660) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (666) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (672) heap_init: At 3FFD7478 len 00008B88 (34 KiB): DRAM
I (678) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (684) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (691) heap_init: At 4009F934 len 000006CC (1 KiB): IRAM
I (698) spi_flash: detected chip: generic
I (702) spi_flash: flash io: dio
I (707) esp_core_dump_flash: Init core dump to flash
I (712) esp_core_dump_flash: Found partition 'coredump' @ 187000 65536 bytes
E (719) esp_core_dump_flash: Incorrect size of core dump image: -240834719
I (727) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
***** initBluepad32() ****
```
Is this a serious error that must be resolved? If so, how?

jsolderitsch, 2024-03-26:
Gently asking if there might be time to investigate this now that the 4.0 release is available?

ricardoquesada, 2024-03-26:
not clear what's the issue.
I've just tested the exact sketch from https://github.com/ricardoquesada/bluepad32/issues/70#issuecomment-1979100042 and it seems to be working (??).
This is the output:

```
ax connected gamepads: 4
BR/EDR support: enabled
BLE support: enabled
Device ID SDP service record size: 64
Gap security level: 2
Periodic Inquiry: max=5, min=4, len=3
I (825) BTDM_INIT: BT controller compile version [0f0c5a2]
I (832) system_api: Base MAC address is not set
I (836) system_api: read default base MAC address from EFUSE
I (842) BTDM_INIT: Bluetooth MAC: 24:0a:c4:00:9f:98
I (850) phy_init: phy_version 4791,2c4672b,Dec 20 2023,16:06:06
Bluetooth Allowlist: Disabled
I (1423) console: Command history disabled

Type 'help' to get the list of commands.
Use UP/DOWN arrows to navigate through command history.
Press TAB when typing command name to auto-complete.
Could not find property 12
Could not find property 11
BTstack up and running at 24:0A:C4:00:9F:98
                                                                                                                                                                                                                                                                                                                               HCI not ready, cannot send packet, will again try later. Current state idx=1
BR/EDR scan -> 1
BLE scan -> 1
I (1622) gpio: GPIO[26]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
Here We Go
Configuring access point...
I (1643) wifi:wifi driver task: 3ffea1a8, prio:23, stack:6656, core=0
I (1654) wifi:wifi firmware version: 1fd20f4
I (1655) wifi:wifi certification version: v7.0
I (1656) wifi:config NVS flash: enabled
I (1657) wifi:config nano formating: disabled
I (1669) wifi:Init data frame dynamic rx buffer num: 32bp32>
I (1673) wifi:Init static rx mgmt buffer num: 5
I (1678) wifi:Init management short buffer num: 32
I (1679) wifi:Init dynamic tx buffer num: 32
I (1680) wifi:Init static rx buffer size: 1600
I (1692) wifi:Init static rx buffer num: 4
I (1693) wifi:Init dynamic rx buffer num: 32
I (1695) wifi_init: rx ba win: 6
I (1696) wifi_init: tcpip mbox: 32
I (1707) wifi_init: udp mbox: 6
I (1708) wifi_init: tcp mbox: 6
I (1709) wifi_init: tcp tx win: 5760
I (1720) wifi_init: tcp rx win: 5760
I (1721) wifi_init: tcp mss: 1440
I (1722) wifi_init: WiFi IRAM OP enabled
I (1733) wifi_init: WiFi RX IRAM OP enabled
I (1738) wifi:mode : softAP (24:0a:c4:00:9f:97)
I (1743) wifi:Total power save buffer number: 16
I (1745) wifi:Init max length of beacon: 752/752
I (1747) wifi:Init max length of beacon: 752/752
AP IP address: XA?XA?;?
Server started
I (25439) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<255,255>, prof:1
I (25443) wifi:station: 86:38:93:68:54:72 join, AID=1, bgn, 20
I (28946) wifi:<ba-add>idx:2 (ifx:1, 86:38:93:68:54:72), tid:0, ssn:0, winSize:64
I (46746) wifi:station: 86:38:93:68:54:72 leave, AID = 1, bss_flags is 658531, bss:0x3ffeed7c
I (46750) wifi:new:<1,0>, old:<1,0>, ap:<1,1>, sta:<255,255>, prof:1
I (46751) wifi:<ba-del>idx
I (51247) wifi:new:<1,0>, old:<1,0>, ap:<1,1>, sta:<255,255>, prof:1
I (51250) wifi:station: 86:38:93:68:54:72 join, AID=1, bgn, 20
I (52565) wifi:<ba-add>idx:2 (ifx:1, 86:38:93:68:54:72), tid:0, ssn:3, winSize:64
I (52578) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.2
I (100144) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.2
```

I used an Android phone to connect to it. And it got the 192.168.4.2 IP address.
After a while it disconnects, but I think it has to do with the "no internet access"... but who knows.

My logs has more info that yours... so not sure why it didn't work for you.

There is some kind of "balance between WiFi and Bluetooth" settings.

It would be great if you can play with it and tell me which one works for you.

E.g:

Go to the template project and from there do:

```
cd my_template_project
idf.py menuconfig
```

and I wouldn't know which option you have to change, but IIRC search for "component config" -> "Wi-Fi" and play with the different options.

And do:

```
idf.py build
idf.py flash monitor
```

jsolderitsch, 2024-03-27:
What version of idf.py are you running? I tried this example back in the Arduino IDE, after converting the console prints back to serial ones, and I get the same failure to complete the password challenge. Used the iPhone this time rather than my Mac WiFi. So I will set up a new template project and try again. I will also try with a Dev Kit version of the ESP32 instead of my AdaFruit Feather.

ricardoquesada, 2024-03-27:
I tested it using v4.4.7 using SparkFun ESP32 Thing.

If you can play a bit with the different WiFi options I mentioned on the previous comment.

But guess is that this is not related to Bluepad32... but to how to use both Bluetooth and WiFi at the some time on ESP32... search online to see whether there are some hints.

Perhaps new need to disable `enableNewBluetoothConnections(false)` while you are doing the WiFi connection (???).

https://github.com/ricardoquesada/esp-idf-arduino-bluepad32-template/blob/main/components/bluepad32_arduino/include/ArduinoBluepad32.h#L54

antonix883, 2024-03-27:
`#include <WiFi.h>

// Replace with your network credentials
const char* ssid     = "ESP32-Access-Point";
const char* password = "123456789";

// Set web server port number to 80
WiFiServer server(80);

// Variable to store the HTTP request
String header;

// Auxiliar variables to store the current output state
String output26State = "off";
String output27State = "off";

// Assign output variables to GPIO pins
const int output26 = 26;
const int output27 = 27;

void setup() {
  Serial.begin(115200);
  // Initialize the output variables as outputs
  pinMode(output26, OUTPUT);
  pinMode(output27, OUTPUT);
  // Set outputs to LOW
  digitalWrite(output26, LOW);
  digitalWrite(output27, LOW);

  // Connect to Wi-Fi network with SSID and password
  Serial.print("Setting AP (Access Point)…");
  // Remove the password parameter, if you want the AP (Access Point) to be open
  WiFi.softAP(ssid, password);

  IPAddress IP = WiFi.softAPIP();
  Serial.print("AP IP address: ");
  Serial.println(IP);
  
  server.begin();
}

void loop(){
  WiFiClient client = server.available();   // Listen for incoming clients

  if (client) {                             // If a new client connects,
    Serial.println("New Client.");          // print a message out in the serial port
    String currentLine = "";                // make a String to hold incoming data from the client
    while (client.connected()) {            // loop while the client's connected
      if (client.available()) {             // if there's bytes to read from the client,
        char c = client.read();             // read a byte, then
        Serial.write(c);                    // print it out the serial monitor
        header += c;
        if (c == '\n') {                    // if the byte is a newline character
          // if the current line is blank, you got two newline characters in a row.
          // that's the end of the client HTTP request, so send a response:
          if (currentLine.length() == 0) {
            // HTTP headers always start with a response code (e.g. HTTP/1.1 200 OK)
            // and a content-type so the client knows what's coming, then a blank line:
            client.println("HTTP/1.1 200 OK");
            client.println("Content-type:text/html");
            client.println("Connection: close");
            client.println();
            
            // turns the GPIOs on and off
            if (header.indexOf("GET /26/on") >= 0) {
              Serial.println("GPIO 26 on");
              output26State = "on";
              digitalWrite(output26, HIGH);
            } else if (header.indexOf("GET /26/off") >= 0) {
              Serial.println("GPIO 26 off");
              output26State = "off";
              digitalWrite(output26, LOW);
            } else if (header.indexOf("GET /27/on") >= 0) {
              Serial.println("GPIO 27 on");
              output27State = "on";
              digitalWrite(output27, HIGH);
            } else if (header.indexOf("GET /27/off") >= 0) {
              Serial.println("GPIO 27 off");
              output27State = "off";
              digitalWrite(output27, LOW);
            }
            
            // Display the HTML web page
            client.println("<!DOCTYPE html><html>");
            client.println("<head><meta name=\"viewport\" content=\"width=device-width, initial-scale=1\">");
            client.println("<link rel=\"icon\" href=\"data:,\">");
            // CSS to style the on/off buttons 
            // Feel free to change the background-color and font-size attributes to fit your preferences
            client.println("<style>html { font-family: Helvetica; display: inline-block; margin: 0px auto; text-align: center;}");
            client.println(".button { background-color: #4CAF50; border: none; color: white; padding: 16px 40px;");
            client.println("text-decoration: none; font-size: 30px; margin: 2px; cursor: pointer;}");
            client.println(".button2 {background-color: #555555;}</style></head>");
            
            // Web Page Heading
            client.println("<body><h1>ESP32 Web Server</h1>");
            
            // Display current state, and ON/OFF buttons for GPIO 26  
            client.println("<p>GPIO 26 - State " + output26State + "</p>");
            // If the output26State is off, it displays the ON button       
            if (output26State=="off") {
              client.println("<p><a href=\"/26/on\"><button class=\"button\">ON</button></a></p>");
            } else {
              client.println("<p><a href=\"/26/off\"><button class=\"button button2\">OFF</button></a></p>");
            } 
               
            // Display current state, and ON/OFF buttons for GPIO 27  
            client.println("<p>GPIO 27 - State " + output27State + "</p>");
            // If the output27State is off, it displays the ON button       
            if (output27State=="off") {
              client.println("<p><a href=\"/27/on\"><button class=\"button\">ON</button></a></p>");
            } else {
              client.println("<p><a href=\"/27/off\"><button class=\"button button2\">OFF</button></a></p>");
            }
            client.println("</body></html>");
            
            // The HTTP response ends with another blank line
            client.println();
            // Break out of the while loop
            break;
          } else { // if you got a newline, then clear currentLine
            currentLine = "";
          }
        } else if (c != '\r') {  // if you got anything else but a carriage return character,
          currentLine += c;      // add it to the end of the currentLine
        }
      }
    }
    // Clear the header variable
    header = "";
    // Close the connection
    client.stop();
    Serial.println("Client disconnected.");
    Serial.println("");
  }
}`

Can anyone connect to access point using this code? Because at present, with Arduino Ide it is still not possible to connect...

ricardoquesada, 2024-03-27:
@antonix883 I can try later today/tomorrow.

In the meantime, could you add this:

```
void setup(void) {
...
BP32.enableNewBluetoothConnections(false);    // <--- ADD THIS LINE IN SETUP
...
```

to see whether that helps?

antonix883, 2024-03-28:
> @antonix883 I can try later today/tomorrow.
> 
> In the meantime, could you add this:
> 
> ```
> void setup(void) {
> ...
> BP32.enableNewBluetoothConnections(false);    // <--- ADD THIS LINE IN SETUP
> ...
> ```
> 
> to see whether that helps?

"'BP32' was not declared in this scope"

The problem is that the connection in Access Point mode does not occur only if the code is compiled with the "ESP32 Arduino + Bluepad32" library cards. If the same code is compiled with "ESP32 Arduino" the connection is successful. This is regardless of whether the features of the Bluepad32 library are used or not. I hope I was clear.

jsolderitsch, 2024-03-28:
I tried this approach to my example, adding in the ```BP32.enableNewBluetoothConnections(false);``` to my example and the WifI access still does not work!! Inside the Arduino IDE.

antonix883, 2024-03-28:
> @antonix883 add at the top:
> 
> ```c++
> #include <Bluepad32.h>
> ```
> 
> and try again with `BP32.enableNewBluetoothConnections(false); ` please.
> 
> I understand what you are saying... my guess is that Bluetooth is scanning while WiFi is broadcasting its AP... and that might be the issue

Of course I tried to include the library just to try. It did not work. Besides, if the code doesn't use the library, I don't think the problem could be Bluetooth. Also because it works in station mode. So the problem only affects wifi in access point mode.