ricardoquesada/bluepad32 #71: esp-idf using Rust (esp-rs) | Generate FFI Bindings

lnxdxtf, 2024-02-19:
I'm trying to use this library for an embedded project in Rust with ESP-IDF (esp-rs), but I'm stuck on how to generate the necessary FFI bindings to use it.

I need guidance or documentation on generating FFI bindings for Rust and integrating them with ESP-IDF, allowing me to use this library in my project.

Additional context
Integrating this library into my Rust-based ESP-IDF project is crucial, and any help with FFI bindings would be greatly appreciated.

ricardoquesada, 2024-02-19:
you are kind on your own here.

I wouldn't know how to generate them

lnxdxtf, 2024-02-21:
🥲I will try to found a way 
Like in this video <a href="https://www.youtube.com/watch?v=zSWkrpu8KBA&t=67s">Don't Worry, C and Rust Can FINALLY Coexist (Here's How) | Embedded Rust and C on the RP2040 Pi Pico</a>
And share here