ricardoquesada/bluepad32 #72: BluePad32 and RP2040 only working when connected to PC and arduino IDE open

JclayArt, 2024-02-20:
### What happened?

I followed your instructions and Flashed Bluepad32 v4.0-beta1 onto an Arduino Nano RP2040 Connect and verified the install. While the RP2040 is connected to the PC and ArduinoIDE is open everything works as expected (the example code buttons all work to change the color and make the controller vibrate), I sync my PS4 controller to it and It will connect without any issues and run the example code as long as Arduino IDE is open. However when I try and power the Arduino RP2040 from an external source or Close ArduinoIDE the controller will no longer connect/loses connection.

Do you have any thoughts as to what could be causing this to only function when connected through USB and ArduinoIDE open? Maybe some sort of library issue that I'm not understanding?

I'm running the example controller code, maybe this is user error and I'm missing something here? Very excited to use your awesome Library to control a Nerf Dart Robot I've been building, very much hoping this is solvable.

Thanks for your time and efforts!
-Joshua Clay

### Bluepad32 Version

latest from develop branch

### Bluepad32 version custom

Example: Using Git develop branch commit hash #xxxxxxx

### Bluepad32 Platform

Arduino + NINA

### Platform version

Arduino IDE 2.3.1

### Controller

PlayStation4 controller

### ESP32 chip

ESP32

### ESP32 board

Arduino Nano RP2040 Connect

### OS

None

### Relevant sketch

```shell
// Copyright 2021 - 2023, Ricardo Quesada
// SPDX-License-Identifier: Apache 2.0 or LGPL-2.1-or-later

/*
 * This example shows how to use the Controller API.
 *
 * Supported on boards with NINA W10x. In particular, these boards:
 *  - Arduino MKR WiFi 1010,
 *  - UNO WiFi Rev.2,
 *  - Nano RP2040 Connect,
 *  - Nano 33 IoT,
 *  - Arduino MKR Vidor 4000
 */
#include <Bluepad32.h>

ControllerPtr myControllers[BP32_MAX_CONTROLLERS];

// Arduino setup function. Runs in CPU 1
void setup() {
  // Initialize serial
  Serial.begin(9600);
  while (!Serial) {
    // Wait for serial port to connect.
    // You don't have to do this in your game. This is only for debugging
[... 192 lines cut ...]
// Arduino loop function. Runs in CPU 1
void loop() {
  // This call fetches all the controller info from the NINA (ESP32) module.
  // Call this function in your main loop.
  // The controllers' pointer (the ones received in the callbacks) gets updated
  // automatically.
  BP32.update();

  // It is safe to always do this before using the controller API.
  // This guarantees that the controller is valid and connected.
  for (int i = 0; i < BP32_MAX_CONTROLLERS; i++) {
    ControllerPtr myController = myControllers[i];

    if (myController && myController->isConnected()) {
      if (myController->isGamepad())
        processGamepad(myController);
      else if (myController->isMouse())
        processMouse(myController);
      else if (myController->isBalanceBoard())
        processBalanceBoard(myController);
    }
  }
  delay(150);
}
```

JclayArt, 2024-02-21:
Ran through the installation steps again on a second Arduino Nano RP2040 and have the same results. Everything works as expected while ArduinoIDE is open, once I close ArduinoIDE the controller stops responding and eventually powers off and will not reconnect unless I open ArduinoIDE.

closed 2024-02-21 by ricardoquesada

JclayArt, 2024-02-21:
Oh wow that worked perfectly, thank you for taking the time to respond to me so quickly! Is there anyway I could buy you dinner or something as a thank you? I could zelle or Venmo you or if there's another platform you'd prefer I could look into it. I'll definitely credit you for the awesome Bluepad32 library and support.

-Joshua Clay
www.joshuaclayart.com

remove this part from the sketch:

JclayArt, 2024-04-15:
> thanks. donations are welcome. This page contains different forms to help the project:
> https://bluepad32.readthedocs.io/en/latest/contributing/
>
> thanks

Hello there, I'm wondering if you'd possible be able to give me an example of how to use axisX and axisY within an Arduino RP2040 sketch similiar to how you give an example for using the B button:

if (gamepad->b()) {
 }

I tried using it like this but haven't had any luck and I'm not sure what I'm doing wrong:

if (gamepad->axisX() > 35) {
}

if (gamepad->axisX() < -35) {
}

Any help would be greatly appreciated!!! Love being able to use a PS4 controller with the RP2040, so cool!

Thanks for everything you do!
-Joshua Clay
www.joshuaclayart.com<http://www.joshuaclayart.com>

JclayArt, 2024-04-15:
Hello there, I'm wondering if you'd possible be able to give me an example of how to use axisX and axisY within an Arduino RP2040 sketch similiar to how you give an example for using the B button:  

if (gamepad->b()) { 
 }

I tried using it like this but haven't had any luck and I'm not sure what I'm doing wrong:

if (gamepad->axisX() > 35) {
}

if (gamepad->axisX() < -35) {
}

Any help would be greatly appreciated!!! Love being able to use a PS4 controller with the RP2040, so cool!

Thanks for everything you do!
-Joshua Clay
[www.joshuaclayart.com](http://www.joshuaclayart.com/)

ricardoquesada, 2024-04-15:
that should work. what issues are you having ? which controller are you using ?

JclayArt, 2024-04-15:
Ok interesting, well its good to know I'm on the correct path, thank you!

I'm using a PS4 controller and the issue I'm having is when the axisX portion is included (same as above) in the code the controller will connect however no button presses work, when I comment out the axisX portion everything works again leading me to believe I was somehow calling the axisX portion incorrectly. Everything compiles, it just stops working when I put that portion of code in.

Essentially when axisX is >35 I'm mapping the joystick value to a servos MID and MAX range and when the axisX  value is <-35 I'm mapping the joystick value to the servos MID and MIN range. using 35 and -35 for some deadspace/wiggle room within the joystick axisX value.

ricardoquesada, 2024-04-15:
weird... but what you can do is to print the value of `axisX()`. You should see a value between -512 to 511