ricardoquesada/bluepad32 #73: Adaptive trigger support for dualsense

selimrecep, 2024-02-24:
**Is your feature request related to a problem? Please describe.**
As far as I see this library doesn't support adaptive triggers of dualsense, it would be nice if we could control them

**Describe the solution you'd like**
Implementing adaptive triggers using https://controllers.fandom.com/wiki/Sony_DualSense#FFB_Trigger_Effect_Factories 

**Describe alternatives you've considered**
I am probably going to modify library to implement this, not sure how neat it would fit the project though.

**Additional context**
https://www.youtube.com/watch?v=7gJLVQy-bvM if you don't know what adaptive triggers are

ricardoquesada, 2024-02-24:
two things should be done:

a) add the functionality in the driver
b) expose that functionality to the users

For a), go ahead and implement it.

For b), I think a more "holistic" approach is needed. E.g.: Rumble API needs to be updated. Something similar to what Android is doing. And after that, think a bit more about this feature

selimrecep, 2024-02-24:
Yes, I am going to try an implementation that will fit well to library's internal structure, actually this is first time I tried this library so can't make any promises, but bluepad32's structure seems flexible to add functionality so worth giving a shot.

selimrecep, 2024-02-24:
Not sure if the implementation is clear enough for the library but I don't think it is a good idea to integrate adaptive trigger functionality into "Rumble API" just for sake of single controller having the unique feature imo.

ricardoquesada, 2024-02-25:
yes, I agree. "rumble" and "trigger" should be different... only that when I think of the public API it is important to think of them in an "holistic" way.

I looked at your pull-request and it looks good. I made a few small comments.
Probably I will make a few changes after your pull-request gets merged. 
thanks

selimrecep, 2024-02-25:
I created new one (https://github.com/ricardoquesada/bluepad32/pull/75) and closed the old one(https://github.com/ricardoquesada/bluepad32/pull/74)

closed 2024-02-25 by selimrecep