ricardoquesada/bluepad32 #78: bluepad32 arduino library should be open source

mkornmue, 2024-03-06:
Is it possible to get the Arduino Package as an open source package?
Cause now every c code is integrated as Lib. 

It would be nice so that I can adapt a few things for my usage.

Please give advice if there is another possibility how i can get Bluepad32 as full open source Lib into Arduino IDE, thx!

ricardoquesada, 2024-03-06:
Bluepad32 / BTstack are pre-compiled for Arduino IDE... but you can generate your own build with your custom chanages.

https://github.com/ricardoquesada/esp32-arduino-lib-builder/tree/release/v4.4

(checkout the `release/v4.4` branch).

Read the README.md... and try to follow those steps...

In any case, it might be better to offer the lib outside it, but it is not that trivial

closed 2024-03-06 by ricardoquesada

mkornmue, 2024-03-06:
That sounds good! I will try that. But i am still having the issue that the parser for the dualsense is not reporting battery percent level. Can you add that for future release?

ricardoquesada, 2024-03-06:
could you file a bug for it ?ty