ricardoquesada/bluepad32 #79: Dualsense PS5 controller bluetooth range?

mkornmue, 2024-03-11:
Does anybody find out what the maximum range is where PS5 is connected to ESP32 without external antenna and with external antenna?

I want to know how far it can be driven with 2.4Ghz. I have a RC project where I want to consider Bluepad32 but for this I also need to know what the range is ca.~ 

If somebody can give me this info that would be awsome.

ricardoquesada, 2024-03-11:
I wouldn't know... but look at this workaround:

https://github.com/ricardoquesada/bluepad32/issues/42#issuecomment-1933270864

ricardoquesada, 2024-03-12:
DualSense is Bluetooth Classic only, not BLE.

If you are looking for BLE controllers, look here, and search for "BLE": https://bluepad32.readthedocs.io/en/latest/supported_gamepads/

closed 2024-03-12 by ricardoquesada