ricardoquesada/bluepad32 #81: Timeout selection need for disconnected controller

mkornmue, 2024-03-16:
Is there a way to select the timeout when no data is available from controller?

Now it appears that the timeout when the controller is not in range takes about 1 to 5 seconds or something?!
I want to adjust this timeout to get a disconnected after 100ms at least. 

Best way would be that within the Arduino setup function where the callbacks for on Connected and on Disconnected also the timeout can be specified. 

Is there a way to do that?

ricardoquesada, 2024-03-16:
The alternative is to use this: https://github.com/ricardoquesada/bluepad32/issues/42#issuecomment-1933270864

If you don't receive data in certain amount of time, then assume it is disconnected.

(closing it as duplicate... use the bug linked above if you need further clarification)

closed 2024-03-16 by ricardoquesada

mkornmue, 2024-03-16:
Thx. Thats solves my problem.