ricardoquesada/bluepad32 #82: PS4 esp32

Genesiscode99, 2024-03-17:
I’m on Arduino ide I got my ps4 controller to connect to my ESP32 and the sketch uploads with no errors. Wiring and everything is good too but The joy sticks is not controlling the servo motors. [image] [image] [image]

ricardoquesada, 2024-03-17:
why do you think this is a Bluepad32 issue ? (perhaps it is, perhaps it is not).

In any case, DO NOT attach photos. Attach the code.

Genesiscode99, 2024-03-17:
#include <Bluepad32.h>
#include <Adafruit_PWMServoDriver.h>
#include <Wire.h> // Include the Wire library for I2C communication

// Define the number of servo motors
#define NUM_SERVOS 16

// Define the servo parameters
#define SERVO_MIN_ANGLE 0
#define SERVO_MAX_ANGLE 180

// Create an array of servo objects
Adafruit_PWMServoDriver pwm[NUM_SERVOS];

// Map joystick values to servo angle
int mapJoystickToServo(int value) {
    // Map joystick values (-511 - 512) to servo angle (180 - 0)
    // Reverse the mapping to move from right to left
    return map(value, -511, 512, SERVO_MAX_ANGLE, SERVO_MIN_ANGLE);
}

// Define SDA and SCL pins
#define SDA_PIN 21  // GPIO pin for SDA
#define SCL_PIN 22  // GPIO pin for SCL

void processGamepad(ControllerPtr ctl) {
    // Read joystick's horizontal axis
    int joystickX = ctl->axisX();

    // Map joystick value to servo angle
    int servoAngle = mapJoystickToServo(joystickX);

    // Set servo positions for all servo motors
    for (int i = 0; i < NUM_SERVOS; i++) {
        pwm[i].setPWM(0, 0, servoAngle); // Adjust pin number if needed
    }
}

void onConnectedController(ControllerPtr ctl) {
    // Print controller connected message
    Serial.printf("Controller connected: %s\n", ctl->getModelName().c_str());
}

void onDisconnectedController(ControllerPtr ctl) {
    // Print controller disconnected message
    Serial.printf("Controller disconnected: %s\n", ctl->getModelName().c_str());
}

void setup() {
    // Initialize serial communication
    Serial.begin(115200);

    // Initialize I2C communication
    Wire.begin(SDA_PIN, SCL_PIN); // Initialize I2C with specified SDA and SCL pins

    // Initialize Bluepad32
    BP32.setup(&onConnectedController, &onDisconnectedController);

    // Initialize all servo drivers
    for (int i = 0; i < NUM_SERVOS; i++) {
        pwm[i].begin();
        pwm[i].setPWMFreq(60);  // Set the PWM frequency to 60Hz
    }
}

void loop() {
    // Update Bluepad32
    BP32.update();

    // Add a small delay to prevent overwhelming the system
    delay(20);
} 

I mean I don't actually know if it's bluepad32 I'm probably just missing something I don't know.

ricardoquesada, 2024-03-17:
I guess you never reported a bug before... so welcome to the world of reporting bugs.

 - use triple backticks to post code: I tell what's happening: https://docs.github.com/en/get-started/writing-on-github/working-with-advanced-formatting/creating-and-highlighting-code-blocks
 - I'm not going to debug your code for you... you have to debug it yourself. And present some "proof".  E.g: This library doesn't work ok if... whatever.
 -  and when someone tells you to read documentation, read it. This is the 3rd time I'm telling you to read this link: http://bit.ly/bluepad32-help  (read every piece of it). E.g: I mentioned there how to paste code

dwilliss, 2024-10-11:
You have a function called `processGamepad`, but nothing ever calls it.
Use the sample program as a starting point change the existing `processGamepad` to yours, but leave the existing `processControllers` and make sure it gets called from `loop()`