ricardoquesada/bluepad32 #84: Arduino Nano ESP32 / Lolin S3 mini: Serial output not working

lp524, 2024-03-20:
Hello Im still having trouble getting my Arduino Nano ESP32 to work with my xbox controller. after making the changes in issue #65 I still get the following error when trying to upload.
No DFU capable USB device available
Failed uploading: uploading error: exit status 74
It also seems to think the arduino is a LOLIN S3 still
If you could help that would be great.

ricardoquesada, 2024-03-20:
Sorry, but I cannot help you with that. I wouldn't know...
have you Googled that error message ? I guess you'll find some kind of answer on the Arduino forums.

closed 2024-03-20 by ricardoquesada

lp524, 2024-03-20:
> Sorry, but I cannot help you with that. I wouldn't know... have you Googled that error message ? I guess you'll find some kind of answer on the Arduino forums.

Ok thank you but I believe it is an error in your code recognizing the Adruino Nano ESP 32

ricardoquesada, 2024-03-20:
https://letmegooglethat.com/?q=arduino+nano+esp32+%22No+DFU+capable+USB+device+available+Failed+uploading%3A+uploading+error%3A+exit+status+74%22&l=1