ricardoquesada/bluepad32 #85: [Bug]: 8bitdo controllers go into constant rumble with new play_dual_rumble API

microbmen, 2024-03-22:
### What happened?

rumble comes on and doesn't go off.  tested on a 8bitdo N30 Pro 2 controller.

### Bluepad32 Version

I don't know

### Bluepad32 version custom

4.0 RC0

### Bluepad32 Platform

None

### Platform version

Arduinio IDE , latest.

### Controller

8bitdo N30 pro 2 ...  (N30 pro (1) doesn't even rumble, so no bug there... or  is it?) 

### Microcontroller

ESP32

### Microcontroller board

ESP32-S3-WROOM1 DevKit from Lolin. This is the store URL: https://example.com

### OS

None

### Relevant sketch

```shell
ctl->playDualRumble(0, /*duration in ms */ 50, force, force);
```

ricardoquesada, 2024-03-22:
in which mode (switch, xbox, etc...) are you using the 8bitdo gamepad ?  does that happen in all modes ?

this is a regression of: https://gitlab.com/ricardoquesada/unijoysticle2/-/issues/10

but I thought it was triggered because I was enabling both the "triggers" and the "motors" and now I'm only enabling the motors.

microbmen, 2024-03-22:
Oh interesting.  I tried it in other modes and it works perfect.  It appears to only exhibit this behavior in xbox mode.  

Which is interesting, (not related) using the UHS 2.0 library for XBOX bluetooth the 8bitdo works perfect in xbox mode... but not the other modes.  complete opposite of this library.

ricardoquesada, 2024-03-22:
mmm... what is UHS 2.0 ? I guess I should take a look at it.

Yeah... so, it is a regression then... I noticed that 8BitDo is not implementing correctly the Xbox protocol... I know how to work-around it.

Although unfortunately my 8BitDo SN30 has a dead battery and I can only put it in "Android" mode.
I have like 5 others 8BitDo, but they don't seem to support rumble (???)... In any case I've just ordered a Pro2... 

Please, could also add the FW version of your 8BitDo controller ? ty

microbmen, 2024-03-22:
UHS is **U**SB **H**ost **S**hield (https://github.com/felis/USB_Host_Shield_2.0/tree/master)  its a library for connecting devices via USB to a number of Arduino chips.  There are libraries in here for Bluetooth when using a USB Bluetooth dongle plugged into the host shield.  Look at the XBOXONESBT example.  - which you can connect an 8bitdo via xbox mode.  
(UHS supports left and right rumble btw, if looking at the code helps) 

I just checked my firmware and it was 6.22.  I upgraded it to 6.23, and same result.  Only in xbox mode, so this is an edge case and easily mitigated by changing modes.

ricardoquesada, 2024-03-22:
thanks for the info

ricardoquesada, 2024-03-23:
fixed.

closed 2024-03-23 by ricardoquesada

microbmen, 2024-03-25:
weird, now with 4.0 I get constant rumble with xbox 5.x firmware.  4.0 rc0 worked perfect.    it was a quick update test, ill try more later. just wanted you to know.

ricardoquesada, 2024-03-25:
what do you mean by "constant rumble" ?

I tested with original Xbox controllers with FW 4.8 and FW 5.17 and FW 5.20... and was working ok... 

I also tested with 8BitDo Pro2 and SN30 (managed to have it working by disasembling it), and was working ok.

Perhaps I missed some testes.... how are you testing it ? Please paste the API call that you are calling

microbmen, 2024-03-25:
ill have to test more later, but seemed to only be the FW5.20 ,  the older firmware was fine. 

tested with original xbox (3 button) fw 5.20

```
uint8_t bforce = map(ctl->brake(), 0, 1023, 0, 255);
ctl->playDualRumble(0, 50, 0, bforce);

uint8_t tforce = map(ctl->throttle(), 0, 1023, 0, 255);
ctl->playDualRumble(0, 50, tforce, 0);
```

happens for either trigger... 
also if i set the force to be some arbitrary flat value, like 200
ie . `ctl->playDualRumble(0, 50, 200, 0);`

seemed to be less consistent then the 8bitdo, i.e.  a few trigger presses and it would stop.   

ill do more testing when i get home later.

ricardoquesada, 2024-03-25:
ah, I see.

The new "fix" is that the duration is no longer controlled by the controller for two reasons:

- that's the bug that 8BitDo has... they are not calculating correctly the "duration" in the request
- It was causing some issues when cancelling the previous request in case "delayedStart" was issue

so now Bluepad32 controls the duration by "starting" and "stoping" the rumbling.

And if you only use 50 milliseconds as duration it might be possible that you don't even feel it.

Try using something bigger like 150 ms... or even 200.

that should work... I guess (???)

microbmen, 2024-03-26:
Tried some things, and it doesn't work.  Only the Xbox 5.20 firmware.  Other Xbox works fine.  
changing duration to higher values doesn't help.  I went to 500ms, still hangs it up.  Tried the other direction too... down to 10ms.  

4.0 RC0 works nicely.  Not sure if its worth it to do this for the 8bitdo Xbox mode issue.  Its an easy fix to change modes on the 8bitdo to android mode.

ricardoquesada, 2024-03-26:
just to understand, what do you mean by "still hangs it up" ? thanks

microbmen, 2024-03-26:
I mean that the rumble doesn't stop.

ricardoquesada, 2024-03-26:
weird... I cannot reproduce it.

I'm a bit confused. Could you describe the steps to reproduce it... as detailed as you can.

- Controller that reproduces the issue. Controller that does not
- sketch that you are using (unless it is 100% identical of `Controller.ino` no need to copy. if you changed a byte, please copy & paste the entire sketch)
-  how to reproduce it: which buttons are you pressing
- does it stop if  you press the buttons again 
- what platform are you using ? Arduino IDE ?
- which board are you using?

etc...

I tried different combinations and I couldn't reproduce it... that's why I need a detailed stops-to-repro.

microbmen, 2024-03-26:
[Controller.txt](https://github.com/ricardoquesada/bluepad32/files/14759692/Controller.txt)

microbmen, 2024-03-26:
Try with this script above.

Controller that reproduces the issue. Controller that does not. Which FW is the controller using.
`I tried a few xbox controllers from 1708, to 1914 ... all my controllers with 5.20 did this behavior.   Controllers on  FW 3.1 did not do this behavior.` 

sketch that you are using (unless it is 100% identical of Controller.ino no need to copy. if you changed a byte, please attach the entire sketch)
`attached`

how to reproduce it: which buttons are you pressing.
`ok , this is tricky part.  doing a simple button press wont cause it.  its intermittent and makes no sense that I can see.  You have to 'mess' with the buttons.  Using the script i attached pull the triggers,  both at the same time, one after the other.  within a few seconds of pulling trigger you should get the rumble to go continuously.  Let go of the triggers and they will keep rumbling.`

the frequency of pressing... are you pressing it more than once ?
`yes as stated above.  sometimes pressing the button once will cause it, but that's very infrequent,  rotating how your press the buttons will induce the behavior every time.`

does it stop if you press the buttons again ?
`yes, you can keep 'messing' with the buttons causing the rumble to change and it will then stop the consistent rumble,  pressing again might start the constant rumble again. `

what platform are you using ? Arduino IDE ?
`Yes,  Arduino IDE. `

which board are you using?
'this is a ESP32-WROOM-32 $10 board from amazon'

Hope this helps to get you to reproduce.

microbmen, 2024-03-26:
I just tested on another board... a LOLIN C3 Mini (seeed studio mini esp32-c3) and same exact behavior.

ricardoquesada, 2024-03-26:
ty. got it... by any chance, did you see any error on the console ?

something like this error:

https://github.com/ricardoquesada/bluepad32/blob/main/src/components/bluepad32/parser/uni_hid_parser_xboxone.c#L655

I think you might be triggering that... I need to add a "retry" there.

microbmen, 2024-03-26:
No not that error,  but occasionally will see, 

`Xbox: Failed to send rumble report, error=0xc
idx=0, dpad: 0x00, buttons: 0x0000, axis L:   49,    7, axis R:    2,   -3, brake:    0, throttle:    0, misc: 0x00, gyro x:     0 y:     0 z:     0, accel x:     0 y:     0 z:     0, battery: 0
idx=0, dpad: 0x00, buttons: 0x0000, axis L:   47,   15, axis R:    2,   -3, brake:    0, throttle:    0, misc: 0x00, gyro x:     0 y:     0 z:     0, accel x:     0 y:     0 z:     0, battery: 0
idx=0, dpad: 0x00, buttons: 0x0000, axis L:   47,   10, axis R:    2,   -3, brake:    0, throttle:    0, misc: 0x00, gyro x:     0 y:     0 z:     0, accel x:     0 y:     0 z:     0, battery: 0
idx=0, dpad: 0x00, buttons: 0x0000, axis L:   47,    4, axis R:    2,   -3, brake:    0, throttle:    0, misc: 0x00, gyro x:     0 y:     0 z:     0, accel x:     0 y:     0 z:     0, battery: 0
idx=0, dpad: 0x00, buttons: 0x0000, axis L:   47,    1, axis R:    2,   -3, brake:    0, throttle:    0, misc: 0x00, gyro x:     0 y:     0 z:     0, accel x:     0 y:     0 z:     0, battery: 0
idx=0, dpad: 0x00, buttons: 0x0000, axis L:   47,    1, axis R:    2,   -3, brake:    0, throttle:    0, misc: 0x00, gyro x:     0 y:     0 z:     0, accel x:     0 y:     0 z:     0, battery: 0
`

microbmen, 2024-03-26:
now that im playing more with it, i think that error appears every time the rumbling get stuck (goes into constant rumble)

`Xbox: Failed to send rumble report, error=0xc`

ricardoquesada, 2024-03-26:
got it. I think I know how to fix it.

microbmen, 2024-03-26:
I was curious if that error message appeared in BP 4.0-rc0.   

With XBOX FW 5.20 that same error message appears `Xbox: Failed to send rumble report, error=0xc` , however it doesnt disrupt the rumble., rumble works perfect, no constant rumble.  

With XBOX FW 3.1 a different error appears that doesn't appear with BP 4.0 `ERROR: circular buffer full. Cannot queue report` - still rumble works perfect ... not sure why that error occurs in BP 4.0-rc0 but its gone in BP 4.0.  

thought you might like to know this.

ricardoquesada, 2024-03-26:
ty. yes, FW 3.1 uses BR/EDR and I already implemented a "retry" solution for it.

but I haven't done it for BLE (FW 5.x).

ricardoquesada, 2024-03-27:
re-fixed

microbmen, 2024-03-27:
ill await 4.0.1 or whatever :-)  thank you.