ricardoquesada/bluepad32 #87: Filter gamepad connection

miguel-jar, 2024-03-28:
Hello. I was wondering how I can filter a gamepad conection. I want bluepad32 to connect just in a specific gamepad . Let's suppose that  I have many gamepads powerd on, I want bluepad32 to connect just in my gamepad. I'll use this control in an enviroment with many bluetooth devices (including other gamepads), so I wanna make sure bluepad32 will connet just in my gamepad. I was thinking about using the control's MAC address to do this filter, but I couldn't' find a way to do that. Any ideas? 

I have a generic gamepad that is simillar to a PS3 controller, but is not a original one. I don't have any problems connecting my gamepad using bluepad32, I just wanna do this filter.

PS: I've already tried using esp32-ps3 library (by jvpernis) to do this connect, but  it' didn't work. I assuming that it's because my controller isn't an original PS3 controller.

ricardoquesada, 2024-03-28:
try this:
https://bluepad32.readthedocs.io/en/latest/FAQ/#how-to-pair-just-one-controller-to-one-particular-board

if that doesn't work, reopen bug detailing what failed

miguel-jar, 2024-03-28:
> try this: https://bluepad32.readthedocs.io/en/latest/FAQ/#how-to-pair-just-one-controller-to-one-particular-board
> 
> if that doesn't work, reopen bug detailing what failed

Okay, I'll try this tutorial and see what I get. Thank you so much for your response.