ricardoquesada/bluepad32 #88: [Bug]: erro ao baixar pacote de placas

thacyano15, 2024-04-05:
### What happened?

Não consigo adicionar o pacote de placas no Arduíno IDE, dá erro no arquivo Json.

### Bluepad32 Version

I don't know

### Bluepad32 version custom

Example: Using Git develop branch commit hash #xxxxxxx

### Bluepad32 Platform

Arduino IDE

### Platform version

E.g: Arduino IDE 2.1.0, or ESP-IDF v4.4.4, or CircuitPython 8.0.0, etc.

### Controller

issues

### Microcontroller

ESP32

### Microcontroller board

ESP32-S3-WROOM1 DevKit from Lolin. This is the store URL: https://example.com

### OS

None

ricardoquesada, 2024-04-05:
I don't understand the description of the bug.

Impossible to fix, if I don't understand it.

Please read: http://bit.ly/bluepad32-help

thacyano15, 2024-04-05:
Estou tentando utilizar o Arduíno IDE para programar o ESP 32.
segui o passo a passo que tem no link, consigui baixar o pacote de placas ESP 32 fornecido pela Espressif. 
Porém o pacote da placa Bluepad32 fornecido por você não consigo instalar na IDE, apresenta uma mensagem dizendo que o link Json está com problemas.

ricardoquesada, 2024-04-05:
any chance that you translate the message to English ? ty.

thacyano15, 2024-04-05:
I'm sorry for that.

I'm trying to use the Arduino IDE to program the ESP 32.
 I followed the step-by-step instructions in the link and managed to download the ESP 32 board package provided by Espressif.
 However, the Bluepad32 board package provided by you cannot be installed in the IDE, it displays a message saying that the Json link has problems.

thacyano15, 2024-04-06:
can you help me?

thacyano15, 2024-04-08:
I'm sorry I didn't get back.
I solved it, the problem was an error in my notebook's DNS.