ricardoquesada/bluepad32 #89: [Bug]: Failed to disconnect HIDS client for hids_cid=0, status=2

Imlerih, 2024-04-08:
### What happened?

Compiled standard "Controoler" exaple skatch for esp32 devkitc v4. Connecting to gamepad in PC/Android mode. After sucsessful connection with gamepad periodic gettting error "Failed to disconnect HIDS client for hids_cid=0, status=2" which freezes connection for several seconds. The connection restarts then, but the error occures again.
Tested the gamepad with PC and android and it worked fine.

### Bluepad32 Version

I don't know

### Bluepad32 version custom

Example: Using Git develop branch commit hash #xxxxxxx

### Bluepad32 Platform

None

### Platform version

Arduino IDE 1.8.19

### Controller

Dong Guan SW570 Wireless Controller

https://manuals.plus/dong-guan/sw570-wireless-controller-manual

### Microcontroller

ESP32

### Microcontroller board

ESP32 Devkitc V4

### OS

None

### Relevant log output

```shell
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0
Bluepad32 (C) 2016-2024 Ricardo Quesada and contributors.
Version: v4.0.2
BTstack: Copyright (C) 2017 BlueKitchen GmbH.
BTstack up and running at A8:42:E3:AB:21:9E
[   654][D]Firmware: Bluepad32 for Arduino v4.0.2
BD Addr: A8:42:E3:AB:21:9E
67 B7 86 A7 8A 53 BE D4 A5 B6 7A 07 49 8A 28 66 
05 01 09 06 A1 01 85 01 75 01 95 08 05 07 19 E0 29 E7 15 00 25 01 81 02 95 01 75 08 81 03 95 05 75 01 05 08 19 01 29 05 91 02 95 01 75 03 91 03 95 06 75 08 15 00 26 FF 00 05 07 19 00 29 FF 81 00 C0 05 0C 09 01 A1 01 85 02 15 00 25 01 75 01 95 0A 09 EA 09 CD 09 B6 09 B5 09 E9 09 B8 09 95 0A 23 02 0A 24 02 09 40 81 02 95 01 75 06 81 03 C0 05 01 09 02 A1 01 85 06 09 01 A1 00 05 09 19 01 29 03 15 00 25 01 75 01 95 03 81 02 75 05 95 01 81 01 05 01 09 30 09 31 15 81 25 7F 75 08 95 02 81 06 09 38 15 81 25 7F 75 08 95 01 81 06 C0 C0 05 01 09 05 A1 01 85 07 09 01 A1 00 09 30 09 31 09 32 09 35 15 00 26 FF 00 75 08 95 04 81 02 C0 09 39 15 00 25 07 35 00 46 3B 01 65 14 75 04 95 01 81 42 75 04 95 01 81 01 05 09 19 01 29 10 15 00 25 01 75 01 95 10 81 02 05 02 15 00 26 FF 00 09 C4 09 C5 95 02 75 08 81 02 75 08 95 01 81 01 C0 
CALLBACK: Controller is connected, index=0
Controller model: Android, VID=0x1949, PID=0x0402
idx=0, dpad: 0x00, buttons: 0x0000, axis L:    0,    0, axis R:    0,    0, brake:    0, throttle:    0, misc: 0x00, gyro x:     0 y:     0 z:     0, accel x:     0 y:     0 z:     0
idx=0, dpad: 0x00, buttons: 0x0000, axis L:    0,    0, axis R:    0,    0, brake:    0, throttle:    0, misc: 0x00, gyro x:     0 y:     0 z:     0, accel x:     0 y:     0 z:     0
idx=0, dpad: 0x00, buttons: 0x0000, axis L:    0,    0, axis R:    0,    0, brake:    0, throttle:    0, misc: 0x00, gyro x:     0 y:     0 z:     0, accel x:     0 y:     0 z:     0
idx=0, dpad: 0x00, buttons: 0x0000, axis L:    0,    0, axis R:    0,    0, brake:    0, throttle:    0, misc: 0x00, gyro x:     0 y:     0 z:     0, accel x:     0 y:     0 z:     0
idx=0, dpad: 0x00, buttons: 0x0000, axis L:    0,    0, axis R:    0,    0, brake:    0, throttle:    0, misc: 0x00, gyro x:     0 y:     0 z:     0, accel x:     0 y:     0 z:     0
idx=0, dpad: 0x00, buttons: 0x0000, axis L:    0,    0, axis R:    0,    0, brake:    0, throttle:    0, misc: 0x00, gyro x:     0 y:     0 z:     0, accel x:     0 y:     0 z:     0
idx=0, dpad: 0x00, buttons: 0x0000, axis L:    0,    0, axis R:    0,    0, brake:    0, throttle:    0, misc: 0x00, gyro x:     0 y:     0 z:     0, accel x:     0 y:     0 z:     0
idx=0, dpad: 0x00, buttons: 0x0000, axis L:    0,    0, axis R:    0,    0, brake:    0, throttle:    0, misc: 0x00, gyro x:     0 y:     0 z:     0, accel x:     0 y:     0 z:     0
idx=0, dpad: 0x00, buttons: 0x0000, axis L:    0,    0, axis R:    0,    0, brake:    0, throttle:    0, misc: 0x00, gyro x:     0 y:     0 z:     0, accel x:     0 y:     0 z:     0
idx=0, dpad: 0x00, buttons: 0x0000, axis L:    0,    0, axis R:    0,    0, brake:    0, throttle:    0, misc: 0x00, gyro x:     0 y:     0 z:     0, accel x:     0 y:     0 z:     0
[... 436 lines cut ...]
    // When enabled, controllers like DualSense and DualShock4 generate two connected devices:
    // - First one: the gamepad
    // - Second one, which is a "virtual device", is a mouse.
    // By default, it is disabled.
    BP32.enableVirtualDevice(false);
}

// Arduino loop function. Runs in CPU 1.
void loop() {
    // This call fetches all the controllers' data.
    // Call this function in your main loop.
    bool dataUpdated = BP32.update();
    if (dataUpdated)
        processControllers();

    // The main loop must have some kind of "yield to lower priority task" event.
    // Otherwise, the watchdog will get triggered.
    // If your main loop doesn't have one, just add a simple `vTaskDelay(1)`.
    // Detailed info here:
    // https://stackoverflow.com/questions/66278271/task-watchdog-got-triggered-the-tasks-did-not-reset-the-watchdog-in-time

    //vTaskDelay(1);
    delay(150);
}
```

ricardoquesada, 2024-04-08:
make sure that your controller has battery...

other than that, I wouldn't know.

gopro2027, 2025-09-01:
Yea this seems to be an issue when you have a classic bluetooth controller connected and then try to pair a new ble controller. Almost the exact instant that i disconnected my classic controller, the ble controller would then pair up just fine.