ricardoquesada/bluepad32 #90: [Bug]: 4.0.2 reports weird 99.99.99 version after flashing

mneuhaus, 2024-04-08:
### What happened?

When i flash the latest release on to my Nina Chip and try to verify the version number with arduino-fwuploaded i get 99.99.99 back.

### Bluepad32 Version

4.0-beta2

### Bluepad32 version custom

latest 4.0.2 release

### Bluepad32 Platform

Arduino + NINA

### Platform version

not relevant

### Controller

not relevant

### Microcontroller

ESP32

### Microcontroller board

Arduino Nano RP2040 Connect

### OS

None

### Relevant log output

```shell
(base) ➜  bluepad32-nina-v4.0.2 ./arduino-fwuploader firmware flash -b $BOARD -a $ADDRESS -i bluepad32-nina-full.bin
rp2040load 1.0.6 - compiled with go1.16.2
Loading into Flash: [==============================]  100%

Flashing progress: 99%
Upload completed!%                                                                                                                                                                                                     (base) ➜  bluepad32-nina-v4.0.2 ./arduino-fwuploader firmware get-version -b $BOARD -a $ADDRESS                   
rp2040load 1.0.6 - compiled with go1.16.2
Loading into Flash: [==============================]  100%

FIRMWARE-VERSION: 99.99.99
Firmware version installed: 99.99.99%                                                                                                                                                                                  (base) ➜  bluepad32-nina-v4.0.2 ./arduino-fwuploader firmware get-version -b $BOARD -a $ADDRESS
(base) ➜  bluepad32-nina-v4.0.2 ./arduino-fwuploader firmware flash -b $BOARD -a $ADDRESS -i bluepad32-nina-full-v3.10.3.bin 
rp2040load 1.0.6 - compiled with go1.16.2
Loading into Flash: [==============================]  100%

Flashing progress: 99%
Upload completed!%                                                                                                                                                                                                     (base) ➜  bluepad32-nina-v4.0.2 ./arduino-fwuploader firmware get-version -b $BOARD -a $ADDRESS                             
rp2040load 1.0.6 - compiled with go1.16.2
Loading into Flash: [==============================]  100%

FIRMWARE-VERSION: Bluepad32 for NINA v3.10.3
Firmware version installed: Bluepad32 for NINA v3.10.3%                                                                                                                                                                (base) ➜  bluepad32-nina-v4.0.2 
```
```

ricardoquesada, 2024-04-08:
what are the values of `$BOARD` and `$ADDRESS` ?

ricardoquesada, 2024-04-09:
mmm... weird. I'll double check it later today/tomorrow

ricardoquesada, 2024-04-10:
I confirm that I can reproduce it

ricardoquesada, 2024-04-10:
regression introduced between v4.0-beta1 and v4.0-beta2.

Probably the changes in console/uart triggered this regression

closed 2024-04-10 by ricardoquesada