ricardoquesada/bluepad32 #91: [Bug]: More than three Xboxone handles will cause problems

yythackery, 2024-04-09:
### What happened?

When I am using ESP32 as my platform and simultaneously connecting the fourth Xboxone controller, the program crashes as shown below. And when connecting to the third handle, it shows that the connection is normal, but there is no data given to the callback function my_platform_on_controller_data when operating the third handle.

### Bluepad32 Version

latest from develop branch

### Bluepad32 version custom

Example: Using Git develop branch commit hash #xxxxxxx

### Bluepad32 Platform

ESP-IDF

### Platform version

ESP-IDF v5.2.1

### Controller

Xbox Wireless (model 1914, 3 buttons)

### Microcontroller

ESP32

### Microcontroller board

ESP32-WROOM-32

### OS

Windows

### Relevant log output

```shell
Bluepad32 (C) 2016-2024 Ricardo Quesada and contributors.
Version: v4.0.2
BTstack: Copyright (C) 2017 BlueKitchen GmbH.
Platform: custom
custom: init()
Max connected gamepads: 4
BR/EDR support: enabled
BLE support: enabled
Device ID SDP service record size: 64
Gap security level: 2
Periodic Inquiry: max=5, min=4, len=3
I (602) BTDM_INIT: BT controller compile version [0f0c5a2]
I (612) BTDM_INIT: Bluetooth MAC: d0:ef:76:45:f9:8a
I (622) phy_init: phy_version 4791,2c4672b,Dec 20 2023,16:06:06
Bluetooth Allowlist: Disabled
I (1172) console: Command history disabled

Type 'help' to get the list of commands.
Use UP/DOWN arrows to navigate through command history.
Press TAB when typing command name to auto-complete.
Could not find property 12
Could not find property 11
BTstack up and running at D0:EF:76:45:F9:8A
HCI not ready, cannot send packet, will again try later. Current state idx=1
custom: on_init_complete()
BR/EDR scan -> 1
BLE scan -> 1
custom: Bluetooth enabled: 1
Deleting stored BR/EDR link keys:
.
Deleting stored BLE link keys:
[... 339 lines cut ...]
        uint8_t green = (ins->gamepad_seat & 0x02) ? 0xff : 0;
        uint8_t blue = (ins->gamepad_seat & 0x04) ? 0xff : 0;
        d->report_parser.set_lightbar_color(d, red, green, blue);
    }
}

//
// Entry Point
//
struct uni_platform* get_my_platform(void) {
    static struct uni_platform plat = {
        .name = "custom",
        .init = my_platform_init,
        .on_init_complete = my_platform_on_init_complete,
        .on_device_connected = my_platform_on_device_connected,
        .on_device_disconnected = my_platform_on_device_disconnected,
        .on_device_ready = my_platform_on_device_ready,
        .on_oob_event = my_platform_on_oob_event,
        .on_controller_data = my_platform_on_controller_data,
        .get_property = my_platform_get_property,
    };

    return &plat;
}
```

mitchellcairns, 2024-04-09:
I'm not able to connect my Xbox One BLE gamepad whatsoever :/

ricardoquesada, 2024-04-09:
I can reproduce it as well.

It happens with BLE controllers.
I cannot pair more than 2 BLE controllers... I don't receive data from it... or crashes.

But works Ok with non-BLE (BR/EDR) controllers.

yythackery, 2024-04-13:
After my testing, it should have been due to insufficient resources in the ESP32 hardware, so I eventually replaced it with another chip.

ricardoquesada, 2024-04-13:
what do you mean "insufficient resources" ? did you replace it with another esp32 chip ? which one ?
is it working now?

ricardoquesada, 2024-04-13:
Pico W had the same issue.
Fixed here: https://github.com/ricardoquesada/bluepad32/commit/db6d69a1c42bf5f6dc2401ce002c9289a3bd4b49

yythackery, 2024-04-14:
> what do you mean "insufficient resources" ? did you replace it with another esp32 chip ? which one ? is it working now?

I want to choose another Bluetooth chip to implement it, but I haven't chosen it yet.

yythackery, 2024-04-14:
> resources

> what do you mean "insufficient resources" ? did you replace it with another esp32 chip ? which one ? is it working now?

It should be tested that when connecting to the third one, the data can enter the uni_hid_device_process_controller function, but cannot enter uni_get_platform() ->on_controller_data (d,&d ->controller); This call is very strange. Then an exception is reported when connecting the fourth x1 handle. In summary, it seems that there is an issue with the Bluetooth firmware of ESP32.

ricardoquesada, 2024-04-14:
I'll investigate it a bit further... but in the meantime use Pico W (latest `develop` branch) which works Ok with 4 BLE controllers

yythackery, 2024-04-15:
> I'll investigate it a bit further... but in the meantime use Pico W (latest `develop` branch) which works Ok with 4 BLE controllers

Ok. Tanks.

ricardoquesada, 2024-04-17:
@yythackery wondering why did you close it? did you find how to connect 4 BLE devices in ESP32 ?

ricardoquesada, 2024-04-17:
(I'm reopening so that we can keep with the investigation)

yythackery, 2024-04-20:
> @yythackery wondering why did you close it? did you find how to connect 4 BLE devices in ESP32 ?

 I have chosen another chip.

ricardoquesada, 2024-04-20:
@yythackery great. which one?

yythackery, 2024-05-01:
> @yythackery great. which one?

esp32 s2

ricardoquesada, 2024-05-01:
@yythackery  esp32-s2 doesn't have bluetooth. you mean esp32 s3 ? thanks.

yythackery, 2024-05-10:
> @yythackery esp32-s2 doesn't have bluetooth. you mean esp32 s3 ? thanks.

yes，its s3

solhuebner, 2025-03-14:
> > > [@yythackery](https://github.com/yythackery) esp32-s2 doesn't have bluetooth. you mean esp32 s3 ? thanks.esp32-s2 没有蓝牙，您是指 esp32 s3 吗？
> > 
> > 
> > yes，its s3  是，它是 S3
> 
> ESP32-S3 only has BLE, no BR/EDR, How could it connect XBOX controller?

Depends on the controller version and model: https://bluepad32.readthedocs.io/en/latest/supported_gamepads/