ricardoquesada/bluepad32 #92: [Bug]: Xbox One controller hangs on pairing

mitchellcairns, 2024-04-09:
### What happened?

The Xbox One gamepad is using BLE and it's on the latest firmware provided by the Xbox Accessories app. I can never get the connection to complete and reports started up, it hangs at the 'Connection encrypted'. I'm basing my project off of the existing example code, which works great for Switch Pro controller input.

### Bluepad32 Version

latest from develop branch

### Bluepad32 version custom

Example: Using Git develop branch commit hash #xxxxxxx

### Bluepad32 Platform

Pico SDK

### Platform version

Pico SDK v1.5.1

### Controller

Xbox One (3 button type)

### Microcontroller

Pico W

### Microcontroller board

Pico-W (official)

### OS

Windows

### Relevant log output

```shell
Version: 7.95.61 (abcd531 CY) CRC: 4528a809 Date: Wed 2023-01-11 10:29:38 PST Ucode Ver: 1043.2169 FWID 01-7afb0879
cyw43 loaded ok, mac 28:cd:c1:0b:2a:c6
Bluepad32 (C) 2016-2024 Ricardo Quesada and contributors.
Version: v4.0.3
BTstack: Copyright (C) 2017 BlueKitchen GmbH.
Platform: My Platform
my_platform: init()
Max connected gamepads: 4
BR/EDR support: enabled
BLE support: enabled
Device ID SDP service record size: 64
Gap security level: 2
Periodic Inquiry: max=5, min=4, len=3
BT FW download, version = CYW4343A2_001.003.016.0031.0000_Generic_SDIO_37MHz_wlbga_BU_dl_signed
No TLV for bp.bt.allowlist, returning default value '(null)'
Bluetooth Allowlist: Disabled
HCI not ready, cannot send packet, will again try later. Current state idx=0
HCI not ready, cannot send packet, will again try later. Current state idx=1
my_platform: on_init_complete()
BR/EDR scan -> 1
BLE scan -> 1
Deleting stored BR/EDR link keys:
.
Deleting stored BLE link keys:
.
BLUEPAD: ready to fill reportsFound, connect to device with public address 0C:35:26:78:C9:F0 ...
Creating device: 0C:35:26:78:C9:F0 (idx=0)
Device '' identified as Gamepad
BLE scan -> 0
Using con_handle: 0x40
SM_EVENT_IDENTITY_RESOLVING_STARTED
Identity resolving failed for 0C:35:26:78:C9:F0

SM_EVENT_PAIRING_STARTED
Just works requested
Connection encrypted: 1
```

ricardoquesada, 2024-04-09:
did you patch BTstack ?

https://github.com/ricardoquesada/bluepad32/tree/main/external/patches

Also, confirm that it doesn't work with the unmodified example.

ricardoquesada, 2024-04-09:
also confirm that the controller has batteries.. I saw some connection issues when the Xbox controller was low on battery

mitchellcairns, 2024-04-09:
> also confirm that the controller has batteries.. I saw some connection issues when the Xbox controller was low on battery

Batteries are fully charged

mitchellcairns, 2024-04-09:
> did you patch BTstack ?
> 
> https://github.com/ricardoquesada/bluepad32/tree/main/external/patches
> 
> Also, confirm that it doesn't work with the unmodified example.

With the patch, still the same issue.

ricardoquesada, 2024-04-09:
from discord discussion, it is working now. closing it.

closed 2024-04-09 by ricardoquesada

Invictaz, 2024-04-09:
@mitchellcairns can you post the solution for others to use?

ricardoquesada, 2024-04-09:
@Invictaz discussion was here:

https://discord.com/channels/775177861665521725/1227319715668688959/1227319724736516097

mitchellcairns, 2024-04-10:
> @mitchellcairns can you post the solution for others to use?

I completely re-set my project, using the example as the base and it compiles now. Another thing I didn't realize is that the PIO1 is used by the bluetooth stack, so I had to move some code to use PIO0.