ricardoquesada/bluepad32 #95: ESP32-C3 does not found Switch / 8Bitdo Controller but XBOX Wireless Controller

nauen, 2024-04-11:
Hey Guys! Hey Ricardo!

First thank you very much for this library! It met exactly what I searched for using an bluetooth controller for an ESP32-C3 Board.
I'm figuring the behaviors with the library right now.

With ESP32-C3 boards that I'm testing It's only possible to connect and use with the XBOX Controller.
I was also experience that it's only working with esp32_bluepad32/LOLIN C3 Mini or esp32_bluepad32/XIAO_ESP32C3. But not with esp32_bluepad32/ESP32C3 DEV Module.

A WEMOS ESP32 DEV Board runs just fine. All Controllers connect directly and Data is receiving. This is really stunning to see how fast and reliable it works!

The behavior is independent between ESP32 lib versions and bluepad32 lib versions as I see so far.

Testsetup:

- Switch Pro Controller 
- Switch Jaycon
- 8Bitdo Wireless Controller
- XBOX Wireless Controller

And three ESP32 Boards:
- an WEMOS ESP32 DEV Board [link ](https://hackaday.com/2017/05/03/hands-on-hot-new-wemos-esp-32-breakout/)
- and Super Mini ESP32-C3 Board [aliexpress](https://de.aliexpress.com/item/1005006165377284.html?spm=a2g0o.order_list.order_list_main.115.38b75c5fFUpcGp&gatewayAdapt=glo2deu)
- and an ESP32-C3 Custom Board

Software Setup: Android IDE 2.3.0, ESP32 2.0.14, bluepad32 4.0.2 / 3.10.2

Does someone have an idea about it? Is there a way to get some more DEBUG informations?

>Arduino IDE 2.3.0
2.0.14 ESP32 lib
Xbox Controller Wireless Test with ESP32C3 super mini board:
Configured: LOLIN C3 Mini
Bluepad32 (C) 2016-2024 Ricardo Quesada and contributors.
Firmware: Bluepad32 for Arduino v4.0.2
BD Addr: F4:12:FA:26: F:EA
Device Information service found
CALLBACK: Controller is connected, index=0
idx=0, dpad: 0x00, buttons: 0x0000, axis L:  -10,    0, axis R:  -14,   -3, brake:    0, throttle:    0, misc: 0x01, gyro x:     0 y:     0 z:     0, accel x:     0 y:     0 z:     0
idx=0, dpad: 0x00, buttons: 0x0000, axis L:  -10,    0, axis R:  -14,   -3, brake:    0, throttle:    0, misc: 0x01, gyro x:     0 y:     0 z:     0, accel x:     0 y:     0 z:     0
idx=0, dpad: 0x00, buttons: 0x0000, axis L:  -10,    0, axis R:  -14,   -3, brake:    0, throttle:    0, misc: 0x01, gyro x:     0 y:     0 z:     0, accel x:     0 y:     0 z:     0
idx=0, dpad: 0x00, buttons: 0x0000, axis L:  -10,    0, axis R:  -14,   -3, brake:    0, throttle:    0, misc: 0x01, gyro x:     0 y:     0 z:     0, accel x:     0 y:     0 z:     0
idx=0, dpad: 0x00, buttons: 0x0000, axis L:  -10,    0, axis R:  -14,   -3, brake:    0, throttle:    0, misc: 0x01, gyro x:     0 y:     0 z:     0, accel x:     0 y:     0 z:     0
idx=0, dpad: 0x00, buttons: 0x0000, axis L:  -10,    0, axis R:  -14,   -3, brake:    0, throttle:    0, misc: 0x01, gyro x:     0 y:     0 z:     0, accel x:     0 y:     0 z:     0
idx=0, dpad: 0x00, buttons: 0x0000, axis L:  -10,    0, axis R:  -14,   -3, brake:    0, throttle:    0, misc: 0x01, gyro x:     0 y:     0 z:     0, accel x:     0 y:     0 z:     0
CALLBACK: Controller disconnected from index=0

>Xbox Controller Wireless Test with ESP32C3 super mini board:
Configured: ESP32C3 Dev Module
Bluepad32 (C) 2016-2024 Ricardo Quesada and contributors.
Failed to disconnect HIDS client for hids_cid=0, status=2
Failed to disconnect HIDS client for hids_cid=0, status=2

>Xbox Controller Wireless Test with ESP32C3 super mini board:
Configured: LOLIN C3 Mini
BTstack up and running at F4:12:FA:26:0F:EA
Firmware: Bluepad32 for Arduino v3.10.2
BD Addr: F4:12:FA:26: F:EA
CALLBACK: Controller is connected, index=0
Controller model: XBox One, VID=0x045e, PID=0x0b13
Data not available yet
idx=0, dpad: 0x00, buttons: 0x0000, axis L:   20,  -12, axis R:  -13,   -3, brake:    0, throttle:    0, misc: 0x00, gyro x:     0 y:     0 z:     0, accel x:     0 y:     0 z:     0
idx=0, dpad: 0x00, buttons: 0x0000, axis L:   20,  -12, axis R:  -13,   -3, brake:    0, throttle:    0, misc: 0x00, gyro x:     0 y:     0 z:     0, accel x:     0 y:     0 z:     0
idx=0, dpad: 0x00, buttons: 0x0000, axis L:   20,  -12, axis R:  -13,   -3, brake:    0, throttle:    0, misc: 0x00, gyro x:     0 y:     0 z:     0, accel x:     0 y:     0 z:     0
idx=0, dpad: 0x00, buttons: 0x0000, axis L:    4,   -5, axis R:  -13,   -3, brake:    0, throttle:    0, misc: 0x01, gyro x:     0 y:     0 z:     0, accel x:     0 y:     0 z:     0
idx=0, dpad: 0x00, buttons: 0x0000, axis L:    4,   -5, axis R:  -13,   -3, brake:    0, throttle:    0, misc: 0x01, gyro x:     0 y:     0 z:     0, accel x:     0 y:     0 z:     0
idx=0, dpad: 0x00, buttons: 0x0000, axis L:    4,   -5, axis R:  -13,   -3, brake:    0, throttle:    0, misc: 0x01, gyro x:     0 y:     0 z:     0, accel x:     0 y:     0 z:     0
idx=0, dpad: 0x00, buttons: 0x0000, axis L:    4,   -5, axis R:  -13,   -3, brake:    0, throttle:    0, misc: 0x01, gyro x:     0 y:     0 z:     0, accel x:     0 y:     0 z:     0
idx=0, dpad: 0x00, buttons: 0x0000, axis L:    4,   -5, axis R:  -13,   -3, brake:    0, throttle:    0, misc: 0x01, gyro x:     0 y:     0 z:     0, accel x:     0 y:     0 z:     0
idx=0, dpad: 0x00, buttons: 0x0000, axis L:    4,   -5, axis R:  -13,   -3, brake:    0, throttle:    0, misc: 0x01, gyro x:     0 y:     0 z:     0, accel x:     0 y:     0 z:     0
idx=0, dpad: 0x00, buttons: 0x0000, axis L:    4,   -5, axis R:  -13,   -3, brake:    0, throttle:    0, misc: 0x01, gyro x:     0 y:     0 z:     0, accel x:     0 y:     0 z:     0
CALLBACK: Controller disconnected from index=0
[image]
[image]

ricardoquesada, 2024-04-12:
In terms of gamepad support, ESP32-C3 is different from ESP32. See:
https://bluepad32.readthedocs.io/en/latest/supported_gamepads/#bluetooth-gamepads-and-their-protocol

In terms of ESP32-C3 vs ESP32-C3... weird. But just in case, look at:
https://github.com/ricardoquesada/bluepad32/issues/65#issuecomment-1987046804

nauen, 2024-04-12:
Hey Ricardo!
thank you very much for your feedback. I was blind, not seeing the beautiful table in the docs. That clearing things up!
So then, the ESP32-C3 Boards are behaving as expected, thats very nice!

Then is only one point left weird:

ESP32-C3 vs ESP32-C3 is meant the board configuration in Arduino IDE:

ESP32C3 DEV Module vs. LOLIN C3 Mini etc.

The point here is, that I already activated Boot CDC for the LOLIN Board configuration but missed it for ESP32C3 DEV Module.
So there is only the issue, if you miss to activate the CDC function you could get lost.

The behavior without it, is very confusing:
Without USB CDC on Boot:
[image]
The Serial.print (marked area) gives some Output and Errors. Controller is connected but you are not seeing valid output.

With USB CDC on Boot:
[image]
No Errormessage, everything works like expected.

ricardoquesada, 2024-04-12:
correct. that's the "bug".
You need to enable USB CDC on ESP32-C3 (mentioned here: https://bluepad32.readthedocs.io/en/latest/plat_arduino/ ).

So, if there is no other issue, I'm closing it.
Please re-open if I missed something.

closed 2024-04-12 by ricardoquesada