ricardoquesada/bluepad32 #100: ESP32 Pico D4 support

nevsie, 2024-04-27:
I have successfully used your code on various boards, and am highly appreciative. 

I use it for controlling small combat robots.

I was looking to build a custom board that had inbuilt power conversion, motor drivers, etc. And was looking at chip options.  

I am aware it will need BLE and BT Classic, for controller support. The ESP32 Pico D4 SIP seems an ideal small package ...
But then I wondered if there would be any code issues with your library as it's not explicitly on the list of supported items. 

I'm hoping the blanket ESP32 support means it will be fine. But if you knew better or different... I thought I'd ask!

Thanks N

ricardoquesada, 2024-04-27:
ESP32 Pico is supported as well. It is based on ESP32.

closed 2024-04-27 by ricardoquesada

nevsie, 2024-04-27:
Thanks. I assumed that was the case. Then while checking pin outs, I thought I should check!
Legend thanks.