ricardoquesada/bluepad32 #106: [Bug]: Reconnecting to already paired controller after startup does not work reliably

bbsan2k, 2024-06-07:
### What happened?

When trying to reconnect after startup using an 8BitDo SF30Pro, the connection does not reliably establish. This also happens with other Controllers.
This is happening on 3.10.3, Arduino, 2 GamePads enabled in config.

### Bluepad32 Version

3.10.3

### Bluepad32 Platform

Arduino ESP-IDF

### Controller

8BitDo SF30 Pro

### Microcontroller

ESP32

### Microcontroller board

ESP32-S3-WROOM1 DevKit from Lolin. This is the store URL: https://example.com

### OS

None

ricardoquesada, 2024-06-08:
How are you reconnecting it ?
Are you just doing the re-pair again ? or what button are you pressing ?

And what does it mean "not reliable establish" ?

And when you have the chance, attach the logs

john3dc, 2024-06-11:
hello ricardo. i have the same problem as bbsan2k on my gbahd. the problem is not the 8bitdo but also other controllers. if i pair a controller with the esp then it takes up to 2 seconds. after that i can use the controller normally and it runs stable. but if i then restart the controller and try to connect then it either takes a very long time or is not found at all. so i currently have to pair the controller again most of the time.

ricardoquesada, 2024-06-11:
@john3dc any chance that you can provide the info I mentioned here: https://github.com/ricardoquesada/bluepad32/issues/106#issuecomment-2156217289 ? ty

john3dc, 2024-06-16:
hello ricardo. sorry for the delay. we had to activate the log in the firmware first.

pairing is allways stable:
#######################
`Device found: E4:17:D8:33:0D:79 with COD: 0x002508, pageScan 1, clock offset 0x47de, rssi 229 dBm, name '8Bitdo SNES30 GamePad'
Creating device: E4:17:D8:33:0D:79 (idx=0)
uni_bt_process_fsm, bd addr:E4:17:D8:33:0D:79,  state: 4, incoming:0
uni_bt_process_fsm: Starting L2CAP connection
--> HCI_EVENT_CONNECTION_COMPLETE
L2CAP_EVENT_CHANNEL_OPENED (channel=0x0041)
PSM: 0x0011, local CID=0x0041, remote CID=0x0041, handle=0x0081, incoming=0, local MTU=1691, remote MTU=1000, addr=E4:17:D8:33:0D:79
HID Control opened, cid 0x41
uni_bt_process_fsm, bd addr:E4:17:D8:33:0D:79,  state: 10, incoming:0
uni_bt_process_fsm: Create L2CAP interrupt connection
--> HCI_EVENT_LINK_KEY_REQUEST:
L2CAP_EVENT_CHANNEL_OPENED (channel=0x0042)
PSM: 0x0013, local CID=0x0042, remote CID=0x0042, handle=0x0081, incoming=0, local MTU=1691, remote MTU=1000, addr=E4:17:D8:33:0D:79
HID Interrupt opened, cid 0x42
Device E4:17:D8:33:0D:79 is connected
uni_bt_process_fsm, bd addr:E4:17:D8:33:0D:79,  state: 12, incoming:0
uni_bt_process_fsm: starting SDP query
-----------> sdp_query_start()
Starting SDP VID/PID query for E4:17:D8:33:0D:79
SSP User Confirmation Request with numeric value '883433'
SSP User Confirmation Auto accept
Vendor ID: 0x2dc8 - Product ID: 0x2840
Device detected as 8BITDO: 0x34
uni_bt_process_fsm, bd addr:E4:17:D8:33:0D:79,  state: 6, incoming:0
uni_bt_process_fsm: querying HID descriptor
Starting SDP HID-descriptor query for E4:17:D8:33:0D:79
--> HCI_EVENT_AUTHENTICATION_COMPLETE_EVENT: status=0, handle=0x0081
SDP HID Descriptor (86):
05 01 09 05 A1 01 85 03 05 01 15 00 25 07 46 3B 01 95 01 75 04 65 14 09 39 81 42 75 01 95 04 81 01 15 00 26 FF 00 09 30 09 31 09 32 09 35 95 04 75 08 81 02 05 02 15 00 26 FF 00 09 C4 09 C5 95 02 75 08 81 02 05 09 19 01 29 10 15 00 25 01 75 01 95 10 81 02 C0 
<----------- sdp_query_end()
uni_bt_process_fsm, bd addr:E4:17:D8:33:0D:79,  state: 8, incoming:0
uni_bt_process_fsm: Device is ready
Device setup (E4:17:D8:33:0D:79) is complete
CALLBACK: 8BitDo is connected!
Model changed to 8BitDo
Current Controls: 0x8000`

Controlled Disconnect:
###########
```
Connected devices:
idx=0:
Current Controls: 0x8100
	btaddr: E4:17:D8:33:0D:79
	bt: handle=129 (ACL), ctrl_cid=0x0041, intr_cid=0x0042, cod=0x00002508, flags=0x00003f00, incoming=0
	model: vid=0x2dc8, pid=0x2840, model='8BitDo', name='8Bitdo SNES30 GamePad'
	battery: 0 / 255, type=gamepad

L2CAP_EVENT_CHANNEL_CLOSED: 0x0042 (channel=0x0042)
Disconnecting device: E4:17:D8:33:0D:79
Deleting device: E4:17:D8:33:0D:79
L2CAP_EVENT_CHANNEL_CLOSED: 0x0041 (channel=0x0041)
CCouldn't not find hid_device for cid = 0x0041
--> HCI_EVENT_DISCONNECTION_COMPLETE
ALLBACK: Controller is disconnected!
Current Controls: 0x0
```
mostly the reconnect fail:
##########
```
--> HCI_EVENT_CONNECTION_REQUEST: link_type = 1 <--
Creating device: E4:17:D8:33:0D:79 (idx=0)
on_hci_connection_request from: address = E4:17:D8:33:0D:79, cod=0x2508
--> HCI_EVENT_ROLE_CHANGE
--> HCI_EVENT_CONNECTION_COMPLETE
--> HCI_EVENT_LINK_KEY_REQUEST:
[... 39 lines cut ...]
HID Control opened, cid 0x46
uni_bt_process_fsm, bd addr:E4:17:D8:33:0D:79,  state: 10, incoming:0
uni_bt_process_fsm: Create L2CAP interrupt connection
L2CAP_EVENT_CHANNEL_OPENED (channel=0x0047)
PSM: 0x0013, local CID=0x0047, remote CID=0x0043, handle=0x0081, incoming=0, local MTU=1691, remote MTU=1000, addr=E4:17:D8:33:0D:79
HID Interrupt opened, cid 0x47
Device E4:17:D8:33:0D:79 is connected
uni_bt_process_fsm, bd addr:E4:17:D8:33:0D:79,  state: 12, incoming:0
uni_bt_process_fsm: starting SDP query
-----------> sdp_query_start()
Starting SDP VID/PID query for E4:17:D8:33:0D:79
Vendor ID: 0x2dc8 - Product ID: 0x2840
Device detected as 8BITDO: 0x34
uni_bt_process_fsm, bd addr:E4:17:D8:33:0D:79,  state: 6, incoming:0
uni_bt_process_fsm: querying HID descriptor
Starting SDP HID-descriptor query for E4:17:D8:33:0D:79
SDP HID Descriptor (86):
05 01 09 05 A1 01 85 03 05 01 15 00 25 07 46 3B 01 95 01 75 04 65 14 09 39 81 42 75 01 95 04 81 01 15 00 26 FF 00 09 30 09 31 09 32 09 35 95 04 75 08 81 02 05 02 15 00 26 FF 00 09 C4 09 C5 95 02 75 08 81 02 05 09 19 01 29 10 15 00 25 01 75 01 95 10 81 02 C0 
<----------- sdp_query_end()
uni_bt_process_fsm, bd addr:E4:17:D8:33:0D:79,  state: 8, incoming:0
uni_bt_process_fsm: Device is ready
Device setup (E4:17:D8:33:0D:79) is complete
CALLBACK: 8BitDo is connected!
Current Controls: 0x8000
```

ricardoquesada, 2024-06-21:
I didn't know that 8BitDo controllers support "reconnect" by just pressing Start.

I don't have the SNES30 Gamepad, but I'll try to reproduce it with my other 8Bitdo controllers

nsjoubert, 2024-12-01:
Hi, I have the same issue with the Xbox Wireless controller (3 button). Pairing works 100% of the time. However, when I press the Xbox button to turn the controller on, it automatically reconnects, If I hold in the Xbox button to turn off the controller, it disconnects fine and reconnects when turned on again. The problem is now that the controller turns itself off when not in use after a while to save battery and in this case the controller disconnects, but then I get:

When the controller disconnects due to battery saver:
uni_hid_device_disconnect: Unknown GAP connection type: 0

On trying to reconnect by turning the controller on:
uni_bt_le_on_connection_complete: Device not found for addr: XX:XX:XX:XX:XX:XX

I am using it in Arduino IDE with a custom made ESP32 board, but similar to a Wemos D1 mini setup.

ricardoquesada, 2025-01-02:
from @john3dc logs, I see `Removing key for device: E4:17:D8:33:0D:79.` which means that the ESP32 is removing the bonding key... meaning that re-connect will fail.

I had to add that hack to improve connections on some devices. I can try removing it and see what happens.

In any case, make sure that your sketch DOES NOT have this line:
https://github.com/ricardoquesada/esp-idf-arduino-bluepad32-template/blob/1c4cb01c261961fe6dce3d323b09ed774f9511c8/examples/ota/main/sketch.cpp#L103

this will prevent re-connections after a reboot